libimobiledevice/idevicerestore #356: ERROR: Device did not disconnect. Possibly invalid iBSS. Reset device and try again.

zhou0, 2020-08-06:
Found device in DFU mode
Identified device as n94ap, iPhone4,1
The following firmwares are currently being signed for iPhone4,1:
  [1] 9.3.6 (build 13G37)
  [2] 9.3.5 (build 13G36)
Select the firmware you want to restore: 1
Selected firmware 9.3.6 (build 13G37)
Verifying 'iPhone4,1_9.3.6_13G37_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 9.3.6
Product Build: 13G37 Major: 13
Device supports Image4: false
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone4,1_9.3.6_13G37_Restore/058-48374-040.dmg'
Found ECID 2089276808695
Getting ApNonce in dfu mode... 98 1b 92 af 1f cf 8a 39 2f d5 bb 9c 26 6d a1 9f 10 36 57 6b 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.n94.RELEASE.dfu...
Personalizing IMG3 component iBSS...
reconstructed size: 80410
Sending iBSS (80410 bytes)...
[==================================================] 100.0%
ERROR: Device did not disconnect. Possibly invalid iBSS. Reset device and try again.
ERROR: Unable to place device into recovery mode from DFU mode

demhademha, 2020-08-17:
Maybe try manually placing the device in **recovery mode**

Cryptiiiic, 2021-06-25:
I possibly think this is an actual bug with subscribe callback not updating device mode, but we could also all be using it wrong somehow. Maybe Nikias can shed some light. I don't know much about this project.

Cryptiiiic, 2021-06-25:
Ok I ended up fixing this bug but for mac only, the main issue stems from libusb, for whatever reason libusb isn't updating disconnect/reconnect status. When I switched to IOKit compiled libirecovery on mac, the issue was completely fixed. So I might open an issue on the libirecovery repo about libusb problems. @zhou0 @Tegridy Are you guys on mac, linux, or windows.

nikias, 2021-07-01:
@Cryptiiiic so you say on macOS with IOKit it just works? We need to understand if it's the callback implementation or if it's a libusb issue.

Cryptiiiic, 2021-07-03:
@nikias i will test with libusb debug log enabled on Mac and Linux when I get time, yes iokit libirecovery fixed the issues I was experiencing. Will diagnose the issue further soon.

TheAwesomeGuy11, 2022-01-12:
I know this is an older issue but how do you use get futurerestore to use IOKit with libirecovery.

rivciks, 2023-01-17:
How is the status now? I am also getting the error message "Device did not disconnect. Possibly invalid iBSS" when using the idevicerestore.

beni20, 2023-05-10:
happy to test the fix or do a bit of coding. What callback needs to be updated?