libimobiledevice/idevicerestore #357: ERROR: Unable to connect to FDR client (-2)

zhou0, 2020-08-10:
Found device in Recovery mode
Identified device as n94ap, iPhone4,1
The following firmwares are currently being signed for iPhone4,1:
  [1] 9.3.6 (build 13G37)
  [2] 9.3.5 (build 13G36)
Select the firmware you want to restore: 1
Selected firmware 9.3.6 (build 13G37)
Verifying 'iPhone4,1_9.3.6_13G37_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 9.3.6
Product Build: 13G37 Major: 13
INFO: device serial number is DX5MVCHQFML6
Device supports Image4: false
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone4,1_9.3.6_13G37_Restore/058-48374-040.dmg'
Found ECID 2089276808695
Getting ApNonce in recovery mode... 80 37 4f 0c 88 91 ea 5d 72 a4 9f e3 56 b4 2c 6e 77 a8 60 ec 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Sending APTicket (2732 bytes)
Extracting iBEC.n94.RELEASE.dfu...
Not personalizing component iBEC...
Sending iBEC (278924 bytes)...
Getting ApNonce in recovery mode... 80 37 4f 0c 88 91 ea 5d 72 a4 9f e3 56 b4 2c 6e 77 a8 60 ec 
Sending APTicket (2732 bytes)
Recovery Mode Environment:
iBoot build-version=iBoot-2817.60.2
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.s5l8940x.img3...
Not personalizing component RestoreLogo...
Sending RestoreLogo (6796 bytes)...
ramdisk-size=0x4000000
Extracting 058-49412-040.dmg...
Not personalizing component RestoreRamDisk...
Sending RestoreRamDisk (21602572 bytes)...
Extracting DeviceTree.n94ap.img3...
Not personalizing component RestoreDeviceTree...
Sending RestoreDeviceTree (83532 bytes)...
Extracting kernelcache.release.n94...
Not personalizing component RestoreKernelCache...
Sending RestoreKernelCache (10718924 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 13
Device ffffffffffffffffffffffffffffffff00000004 has successfully entered restore mode
Hardware Information:
BoardID: 8
ChipID: 35136
UniqueChipID: 2089276808695
ProductionMode: true
Panic log is available:
panic(cpu 0 caller 0xa0091891): "IOP panic: fmiss_get_next_operation_status: WMR_PANIC: "Timeout" drivers/apple/h2fmi/fmiss_ppn.c:1320  "
Debugger message: panic
OS version: 13G37
Kernel version: Darwin Kernel Version 15.6.0: Fri Aug 19 10:37:54 PDT 2016; root:xnu-3248.61.1~1/RELEASE_ARM_S5L8940X
Paniclog version: 3
Kernel slide:     0x000000001f000000
Kernel text base: 0x9f001000
  Boot    : 0x5f30d421 0x00000000
  Sleep   : 0x00000000 0x00000000
  Wake    : 0x00000000 0x00000000
  Calendar: 0x5f30d42d 0x000116c9

Panicked task 0x8005b440: 8179 pages, 123 threads: pid 0: kernel_task
panicked thread: 0x80180ba0, backtrace: 0x92013e88
		0x9f0c9b63
		0x9f0c9e39
		0x9f020f63
		0xa0091891
		0xa00909a3
		0x9f352031
		0x9f046ecb
		0x9f0c755c

Task 0x8005b440: 8179 pages, 123 threads: pid 0: kernel_task
Task 0x8005b7a0: 675 pages, 5 threads: pid 1: launchd
Task 0x8005bb00: 146 pages, 1 threads: pid 2: fsck
Task 0x8005be60: 186 pages, 1 threads: pid 3: fsck_hfs

Starting FDR listener thread
ERROR: Unable to connect to FDR client (-2)
ERROR: Failed to start FDR Ctrl channel
ERROR: Unable to start the restore process
ERROR: Unable to restore device

kenkit, 2020-12-30:
When this happens not even itunes works again. 
I have atleast 4 devices that are now bricked because of this.
I suspect it has something to do with jailbreaking

SmiVan, 2021-03-30:
Seems to be a similar issue to #316, which was closed without explanation.
Might also be related to #333, #174 and #66, which all show the same `ERROR: Unable to connect to FDR client (-2)`.

At least some of these seem to share the property of being run on previously-jailbroken devices.

V3ntus, 2021-09-03:
So weird. I don't know what I did differently. While stuck at "Starting FDR listener thread", I unplugged and replugged my iPad. The restore failed, iPad returned to restore mode. Then I ran it again, and even though it gave me a "Unable to connect to FDR client", it still went with the restore. Although, I've tried this so many times on my Debian install. This time, I did it on Windows with this: https://github.com/Devjam81/libimobile2019

vaslabs, 2022-05-20:
basically had the same issue and added
```c
info("Wait for your device to restart and then press enter");
getchar();
```
Just to control the flow below https://github.com/libimobiledevice/idevicerestore/blob/master/src/idevicerestore.c#L1371

I waited for a few seconds and then pressed enter

and worked with the combination of `--erase --latest`

Fijxu, 2022-12-28:
Same error here, trying to restore an iPhone 3G model A1241, everything goes well until the iPhone goes trough the recovery mode, AFAIK the device has never been jailbroken before but it doesn't boot.
Tried with the 2 currently signed firmwares, but the same error appears.

> basically had the same issue and added
> 
> ```c
> info("Wait for your device to restart and then press enter");
> getchar();
> ```

I did this too, adding it to the line 1335 and 1336 if the latest release .bz2. But i doesn't worked.
I also tried using the latest commit at this date (7b89019fbb81ce64e1c1d259c5746544c8e8eeea), it doesn't work too because of the SHSH thing that i don't know that it is

```
All required components found in IPSW
Using cached filesystem from 'iPhone1,2_4.2.1_8C148_Restore/038-0015-002.dmg'
checking for local shsh
no local file shsh/3460839113548-iPhone1,2-4.2.1.shsh
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... 
ERROR: Unable to find required ApNonce in parameters
ERROR: Unable to add img3 tags to TSS request
ERROR: Unable to get SHSH blobs for this device
```

Tried using manually downloaded IPSW, with the `--latest` argument and with `--latest --erase` together arguments too. It also fails.

**Plus** i also have to connect the iPhone to iTunes (from a windows machine) first and then connect it to my Linux computer for some reason; Because when i boot into DFU mode normally, the iPhone is recognized as other DFU type (i think) and `idevicerestore` don't recognize it, so i need to connect the iPhone to iTunes first, and then to the Linux machine so `idevicerestore` can recognize it.

Entering DFU mode normally dmesg says:
```
[19600.632478] usb 1-10: new high-speed USB device number 42 using xhci_hcd
[19600.759451] usb 1-10: New USB device found, idVendor=05ac, idProduct=1222, bcdDevice= 0.01
[19600.759454] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19600.759456] usb 1-10: Product: USB DFU Device
[19600.759457] usb 1-10: Manufacturer: Apple Computer, Inc.
[19600.759458] usb 1-10: SerialNumber: [REDACTED] I should censor this?
```
Connecting the iPhone to iTunes and then to the Linux machine: 
```
[19782.579522] usb 1-10: new high-speed USB device number 43 using xhci_hcd
[19782.707252] usb 1-10: New USB device found, idVendor=05ac, idProduct=1227, bcdDevice= 0.00
[19782.707256] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19782.707257] usb 1-10: Product: Apple Mobile Device (DFU Mode)
[19782.707258] usb 1-10: Manufacturer: Apple Inc.
[19782.707259] usb 1-10: SerialNumber: CPID:8900 CPRV:30 CPFM:03 SCEP:05 BDID:04 ECID:0000************ IBFL:00 SRTG:[iBoot-596.24]
```
And with that is now able to be used with `idevicerestore`

Please fix this lol. If the devs need more information, just contact me