libimobiledevice/idevicerestore #358: ASR: Could not restore - No space left on device

eplatt, 2020-08-16:
Hi there,

Long story short, my wife's iPad has a botched OS and the update is failing because of not enough space. When I use iPad OS 13.6.1 I get to about 96~% before it fails and run out of space whereas 13.6 I can get to about 98%. I was hoping to install 13.5.1 to hopefully have enough space but it seems this is no longer signed by Apple.

I know I should have had backups, but I a wanted to see if there is any possible way to free up / use less space during the restore to see if I can get this to go through without wiping the user data. Perhaps there are some logs that can be purged before attempting the filesystem copy? Wondering if anyone has some ideas.

Right now I can only get into recovery/DFU so looking for some suggestions.

The last log snippit before it fails is here:

ASR RESTORE PROGRESS: 92%
ramrod_display_set_granular_progress_forced: 45.000000
ASR RESTORE PROGRESS: 94%
ramrod_display_set_granular_progress_forced: 46.000000
ASR RESTORE PROGRESS: 96%
ramrod_display_set_granular_progress_forced: 47.000000
ASR RESTORE PROGRESS: 98%
ramrod_display_set_granular_progress_forced: 48.000000
ASR STATUS: fail
ASR: Could not restore - No space left on device
ASR STATUS: fail
restore_apfs_image : failed to restore an APFS image, error = 14
[21:50:20.0028-GMT]{3>6} CHECKPOINT FAILURE:(FAILURE:14) (null):[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 196
restore-step-user-progress = 48

I think 13.5.1 would work and let me get back into the device to take a backup but it is no longer signed by Apple. I don't think I am able to get the keys required to bypass Apple unless I was able to get into the device, but if this is an option I'd really appreciate some guidance.

Thanks a lot!

demhademha, 2020-08-17:
Well, if your ipad    supported **checkm8 then you could use a ssh ramdisk to gain access to the device. Then you could remove some unneeded stuff and continue with the update 
I'm not saying this will work, but you can try. .

eplatt, 2020-08-17:
Well, unfortunately after explicitly telling Apple NOT to wipe my device - they ended up wiping my device. I'm trying to see if I can leverage the ssh ramdisk option to extract the usr disk image and maybe run scalpel or something on it. Right now the iPad is effectively factory reset on 13.6.1

demhademha, 2020-08-17:
To be brutally honest, I **don't think you'll get your data back** no matter what you do now that the device is erased. 
I am not in the forensics industry,  maybe see an expert as see if they can help you. Anyways, I don't think it is possible to get your data back personally.

eplatt, 2020-08-17:
I agree, it's not like recovering a regular flash drive or something. That said, if there is even 0.01% chance of recovering anything it will be immediately after restoring and before writing to the drive further. I feel like I will probably have this in the back of my mind forever if I don't at least try. I'll close off this issue since my problem is now not at all related to this utility or project. Thanks again.

closed 2020-08-17 by eplatt

nikias, 2020-08-17:
@eplatt sorry to tell you that but the erase destroyed all the user data, because of disk encryption. The keys are wiped from effaceable storage during erase restore, and thus the data becomes unrecoverable. Furthermore, the restore created a new (encrypted) filesystem, overwriting the old one so unfortunately your will not get your data back 😢

eplatt, 2020-08-17:
@nikias I agree with you. Thanks for everyone time.