libimobiledevice/idevicerestore #359: iPhone 5s restore stuck at "ERROR: Unable to send RestoreRamDisk component: Unable to upload data to device"

cshapeshifter, 2020-08-18:
Hi. First of all, thanks for this tool.

I'm trying to restore a bootlooping iPhone 5S. I can boot into recovery and into DFU. If I try to restore from recovery, I get stuck very early
```
...
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (731057 bytes)...
ERROR: Unable to send iBEC component: Unable to upload data to device
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC
```
I found some advice to try restoring from DFU. When I tried that, I got stuck a bit later, at:
```
...
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (731057 bytes)...
[==================================================] 100.0%
Waiting for device to disconnect...
irecv_event_cb: device 000006348bd45d30 (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
ERROR: Device did not reconnect in recovery mode. Possibly invalid iBEC. Reset device and try again.
ERROR: Unable to place device into recovery mode from DFU mode
```
But I found that idevicerestore is a bit too impatient with the device and just needs to wait a bit longer for it to reboot into recovery. 10 seconds is not enough. The following patch works around this problem:
```
diff --unified --recursive --text a/src/dfu.c b/src/dfu.c
--- a/src/dfu.c	2020-08-18 20:12:22.194457462 +0200
+++ b/src/dfu.c	2020-08-18 20:12:56.321137480 +0200
@@ -446,7 +446,7 @@
 		return -1;
 	}
 	debug("Waiting for device to reconnect in recovery mode...\n");
-	cond_wait_timeout(&client->device_event_cond, &client->device_event_mutex, 10000);
+	cond_wait_timeout(&client->device_event_cond, &client->device_event_mutex, 60000);
 	if (client->mode != &idevicerestore_modes[MODE_RECOVERY] || (client->flags & FLAG_QUIT)) {
 		mutex_unlock(&client->device_event_mutex);
 		if (!(client->flags & FLAG_QUIT)) {
```
But now I'm stuck again, and I can't figure out how to move along:
```
...
Received SHSH blobs
opening device 05ac:1227...
found device with ECID 000006348bd45d30
Setting to configuration 1
Setting to interface 0:0
DEBUG: tss_response_get_path_by_entry: No entry 'iBSS' in TSS response
NOTE: No path for component iBSS in TSS, will fetch from build_identity
Extracting iBSS.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Tag found
Sending iBSS (174065 bytes)...
[==================================================] 100.0%
Waiting for device to disconnect...
irecv_event_cb: device 000006348bd45d30 (udid: N/A) disconnected
Waiting for device to reconnect...
irecv_event_cb: device 000006348bd45d30 (udid: N/A) connected in DFU mode
opening device 05ac:1227...
found device with ECID 000006348bd45d30
Setting to configuration 1
Setting to interface 0:0
Nonce: 22 2f 34 0f b6 95 18 34 16 10 44 24 c3 77 ff 28 53 be d2 53 
Setting to configuration 1
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (731057 bytes)...
[==================================================] 100.0%
Waiting for device to disconnect...
irecv_event_cb: device 000006348bd45d30 (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 000006348bd45d30 (udid: N/A) connected in Recovery mode
opening device 05ac:1281...
found device with ECID 000006348bd45d30
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is DNPMN3DWFFG9
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreLogo' in TSS response
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (11857 bytes)...
DEBUG: RestoreTrustCache is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreTrustCache' in TSS response
NOTE: No path for component RestoreTrustCache in TSS, will fetch from build_identity
Extracting 038-60285-004.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Tag found
Sending RestoreTrustCache (9336 bytes)...
ramdisk-size=0x10000000
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 038-60285-004.dmg...
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (93851886 bytes)...
ERROR: Unable to send RestoreRamDisk component: Unable to upload data to device
ERROR: Unable to send RestoreRamDisk to device.
ERROR: Unable to send Ramdisk
ERROR: Unable to place device into restore mode
```
This is Archlinux, all relevant packages built via AUR from git today:
```
[2020-08-18T18:48:09+0200] [ALPM] upgraded libirecovery-git (2:0.1.1.r159.g2bd64e6-1 -> 2:1.0.0.r3.g6e96529-1)
[2020-08-18T18:52:02+0200] [ALPM] removed libplist (2.2.0-1)
[2020-08-18T18:52:02+0200] [ALPM] installed libplist-git (1:2.2.0.r2.g93dbf11-1)
[2020-08-18T18:53:19+0200] [ALPM] installed libusbmuxd-git (2.0.2.r1.gc7d7d1a-1)
[2020-08-18T20:14:51+0200] [ALPM] installed idevicerestore-git (1:r714.8ebee55-1)
```
Any help appreciated!

cshapeshifter, 2020-08-23:
So I found that my cable was somehow faulty. Now I have a new, genuine cable and the above problem is not occuring anymore. But I'm stuck again, shortly after:
```
Received SHSH blobs
Extracting iBSS.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBSS...
Sending iBSS (174065 bytes)...
[==================================================] 100.0%
Nonce: 36 e3 cc 94 ee 48 a2 e9 05 80 af 1f af 74 7a 75 8f 20 36 07 
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (731057 bytes)...
[==================================================] 100.0%
INFO: device serial number is DNPMN3DWFFG9
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (11857 bytes)...
Extracting 038-60285-004.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9336 bytes)...
ramdisk-size=0x10000000
Extracting 038-60285-004.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (93851886 bytes)...
Extracting DeviceTree.n53ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (109009 bytes)...
Extracting kernelcache.release.iphone6...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (13336432 bytes)...
ERROR: Failed to place device in restore mode
ERROR: Unable to place device into restore mode
```
I should note that restoring from recovery still fails early and I only got this far by starting the restore from DFU, like explained above.

douglasg, 2020-09-04:
I have the same problem under windows with different iOS devices, iPad Pro 13.7, iPhone 6  ...

`Entering recovery mode...
INFO: device serial number is C6KPC1R2G5MP
NOTE: device is not in DFU mode, assuming recovery mode.
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (12334 bytes)...
ERROR: Unable to send RestoreLogo component: Unable to find device
ERROR: Unable to send RestoreLogo to device.
ERROR: Unable to send AppleLogo
ERROR: Unable to place device into restore mode`

And when i try it in recovery mode i get a loop with the output connect / disconnect:

`irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected
irecv_event_cb: device 00126d8118681826 (udid: N/A) connected in Recovery mode
irecv_event_cb: device 00126d8118681826 (udid: N/A) disconnected`

cshapeshifter, 2020-09-04:
@douglasg I'm not sure this is the same problem, since I was (apparently) able to send RestoreLogo and all the other things, while in your case, it appears that the process fails earlier.

In any case, I'm now assuming that my device has a hardware malfunction. The (fresh) battery does not charge, the device only turns on when connected with a cable. If I'm not going into DFU or Recovery mode, the screen shows the black apple log on white background for a few seconds, then briefly flashes a red screen and reboots, over and over again. It seems that many devices with these symptoms have a hardware defect, usually a solder point that came undone on a chip. I'm not able to resolder an entire IC, but I might, for better or worse, try to just "bake" the components and hope for a re-solder that lasts long enough for a flash and data retrieval.

One thing you could try: open the device and disconnect some of the internal cables (to the front-facing camera, digitizer or screen) and see if it helps. It doesn, for some people, if the defect is in one of those cables or components.

Johnny314, 2020-10-22:
Hi i am also getting this problem in Windows 

C:\msys64\mingw32\bin>idevicerestore -e D:\3utools\Firmware\iPhone_4.0_32bit_10.3.3_14G60_Restore.ipsw
Found device in Recovery mode
Identified device as n48ap, iPhone5,3
Extracting BuildManifest from IPSW
Product Version: 10.3.3
Product Build: 14G60 Major: 14
ERROR: Device is in an invalid state
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'D:\3utools\Firmware\iPhone_4.0_32bit_10.3.3_14G60_Restore/058-74968-062.dmg'
ERROR: Device is in an invalid state
ERROR: Unable to find device ECID

C:\msys64\mingw32\bin>

Johnny314, 2020-10-24:
i am also getting this issue but  3utools able to restore device smoothly 
anyone have idea?