libimobiledevice/idevicerestore #362: IOS14,Unhandled progress operation 80 (80),Unknown data request 'PersonalizedData' received

barluc, 2020-09-01:
[FisOS.log](https://github.com/libimobiledevice/idevicerestore/files/5153529/FisOS.log)

barluc, 2020-09-01:
...
verify fs ...[89] 
verify fs ...[89] 
verify fs ...[89] 
verify fs ...[89] 
verify fs ...[90] 
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
FDR 0058B790 timeout waiting for command
FDR 0058B790 waiting for message...
Checking filesystems (15)
Checking filesystems (15)
Mounting filesystems (16)
Mounting filesystems (16)
Mounting filesystems (16)
Mounting filesystems (16)
Unknown operation (80)
Unhandled progress operation 80 (80)
FDR 0058B790 timeout waiting for command
FDR 0058B790 waiting for message...
Unknown data request 'PersonalizedData' received
idevicerestore_\common.c:printing 430 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Arguments</key>
	<dict>
		<key>ImageList</key>
		<true/>
		<key>ImageType</key>
		<string>IsiBootEANFirmware</string>
	</dict>
	<key>DataType</key>
	<string>PersonalizedData</string>
	<key>MsgType</key>
	<string>DataRequestMsg</string>
</dict>
</plist>

nikias, 2020-09-01:
https://github.com/libimobiledevice/idevicerestore/blob/master/src/restore.c#L2650-L2655
Latest code handles this. Make sure to use latest code

barluc, 2020-09-01:
> https://github.com/libimobiledevice/idevicerestore/blob/master/src/restore.c#L2650-L2655
> Latest code handles this. Make sure to use latest code.

-------------------
After Update the latest code, it's  restore success.
thanks!

closed 2020-09-01 by barluc