libimobiledevice/idevicerestore #364: ASR failing on iPadOS14

cheesycod, 2020-09-11:


cheesycod, 2020-09-11:
[liveuser@tmeow Downloads]$ sudo idevicerestore -e iPad_64bit_TouchID_14.0_18A5373a_Restore.ipsw 
Found device in Recovery mode
Identified device as j81ap, iPad5,3
Extracting BuildManifest from IPSW
Product Version: 14.0
Product Build: 18A5373a Major: 18
INFO: device serial number is DLXQD75SG5VJ
Device supports Image4: true
Variant: Developer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPad_64bit_TouchID_14.0_18A5373a_Restore/038-44187-100.dmg'
Found ECID 4830739247531046
Getting ApNonce in recovery mode... 4e 87 b1 f6 09 d5 66 ba 1f 4d 4f 9c 51 7a 6d db 25 dd 90 89 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 59 40 80 5d 19 35 eb b9 56 c4 64 40 68 1e 5d 59 3d d4 27 99 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.ipad5b.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (772708 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6723.0.48
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~ipad.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (27607 bytes)...
Extracting 038-44206-142.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (10843 bytes)...
ramdisk-size=0x20000000
Extracting 038-44206-142.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (102880163 bytes)...
Extracting DeviceTree.j81ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (27986 bytes)...
Extracting kernelcache.release.ipad5b...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (18435882 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device 27366c82a3e380dbcf6c2a0a6614acbc352c8b70 has successfully entered restore mode
Hardware Information:
BoardID: 6
ChipID: 28673
UniqueChipID: 4830739247531046
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
[... 946 lines cut ...]
restore-step-uptime = 16
restore-step-user-progress = 0
[05:26:55.0284-GMT]{4>8} CHECKPOINT END: RESTORED:[0x1613] cleanup_recovery_os_volume
restore-step-ids = {0x11030677:43;0x1103065B:51}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 16
restore-step-user-progress = 0
[05:26:55.0285-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:43;0x1103065B:51;0x11030647:54}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 16
restore-step-user-progress = 0
[05:26:55.0285-GMT]{4>8} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:43;0x1103065B:51}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 16
restore-step-user-progress = 0
[05:26:55.0286-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030677:43;0x1103065B:51;0x11030648:55}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 16
restore-step-user-progress = 0

ERROR: Unable to restore device
[liveuser@tmeow Downloads]$

cheesycod, 2020-09-11:
This only happens with iPadOS 14 and does not happen on iPadOS 13

cheesycod, 2020-09-11:
This is on a iPad Air 2

cheesycod, 2020-09-11:
This bug also only happens on idevicerestore. 3utools using the same ipsw and cable via virtualbox works perfectly fine

nikias, 2020-09-11:
Can you make sure libusbmuxd is latest?

cheesycod, 2020-09-11:
libusbmuxd-2.0.2-3.fc33.x86_64
So yes

nikias, 2020-09-11:
Can you try to increase the timeout here to something like 5000 or 10000 and recompile, then try again: https://github.com/libimobiledevice/libusbmuxd/blob/master/common/socket.c#L855

a-gunter, 2020-09-16:
Had a similar issue on iOS 14 on iPhone 11 Pro. ASR kept failing at sending the filesystem consistently around the 57% mark. Tried different cables and different devices to no avail, and then increasing the timeout to 10000 did the trick.

nikias, 2020-09-16:
@a-gunter I also had it fail on an iPad Air 2 right now. With an increased timeout it did work. I pushed commit https://github.com/libimobiledevice/libusbmuxd/commit/73cb5c182e36e7aafcff8a8932b21c6ab5cae5a6 to libusbmuxd with the increased timeout. But the actual problem is the error handling that I need to take care of in `usbmuxd_send` and `idevice_connection_send`, and then also inside idevicerestore. But the current workaround will help for most cases now :)