libimobiledevice/idevicerestore #368: Restore locked iPhone 5

dedesite, 2020-10-02:
Hi,
I just get an old iPhone 5 from my son and it seems that he try too many times to find his password (he didn't use the phone for years)... So the phone is know "locked" displaying a message like (sorry it's in french so my translation is maybe not accurate) :
"iPhone is unactivated, please connect to iThunes".

The problem is that I don't have MacOs or Windows install on any of my computers...

I though idevicerestore was a way to do a full restore via the linux command line but when I launch `idevicerestore`, I get this message : 
> ERROR: Unable to discover device mode. Please make sure a device is attached.

When I try `ideviceinfo`, i got : 
> ERROR: No device found!

I tried to restart usbmux daemon this way : 
` systemctl restart usbmuxd.service`

But didn't work.

Operating system : Xubuntu 18.04
FYI, I manually install from source `libusbmuxd`, `libplist`, `librecovery` and `libimobiledevice` because they were out of date on my distro.

Is there a solution for my problem ?

Thanks,
Andréas

dedesite, 2020-10-02:
I tried with Ubuntu 20.04 which have an higher usbmuxd version and it give more informations.

It says that the phone is password protected (error -17).

So I guess I got my answer, it's not possible to use idevicerestore with a locked phone...

nikias, 2020-10-02:
Salut :)
No worries, you can use it. But `idevicerestore` doesn't reflect this state properly. You need to put the device into recovery mode by using the `ideviceenterrecovery` tool, just pass the device UDID: `ideviceenterrecovery UDID`. You can find the UDID with `idevice_id -l`. Alternatively you can use the home and power button combo on the device to force it into recovery mode (just google for it). When it's in recovery mode, you can restore it with `idevicerestore`. Please note that you will have to use `-e` to perform an erase restore, effectively destroying all user data on the device. Otherwise it will come back and still be in "locked" state. But in any case, the only way to get the device functioning again is to wipe user data.

dedesite, 2020-10-03:
Thank you @nikias for your reply, it works just like a charm on Ubuntu 20.04. It didn't work on 18.04 because of the too old `usbmuxd` version I suppose.

I feel a bit dump not having find the information about restore mode myself. It's even on official Apple website : 
https://support.apple.com/en-us/HT204306

This issue can be close :+1:

closed 2020-10-03 by dedesite