libimobiledevice/idevicerestore #370: Device doesnt accept Baseband - D10 (iPhone 7)

Exploite-d, 2020-10-14:
Log : 
Entering recovery mode...
INFO: device serial number is xxxxxxx
Extracting iBEC.d10.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (522500 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6723.40.120.0.2
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (13408 bytes)...
Extracting 038-77076-035.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (11701 bytes)...
ramdisk-size=0x20000000
Extracting 038-77076-035.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (103240445 bytes)...
Extracting DeviceTree.d101ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (36416 bytes)...
Extracting kernelcache.release.iphone9...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (14877907 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device ffffffffffffffffffffffffffffffff00000009 has successfully entered restore mode
Hardware Information:
BoardID: 12
ChipID: 32784
UniqueChipID: 8519257147419942
ProductionMode: true
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.d10.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~iphone.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1334~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.d101ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting liquiddetect@1334~iphone-lightning.im4p...
Personalizing IMG4 component Liquid...
Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.d10.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.d101.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.d101.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
About to send FDR Trust data...
Sending FDR Trust data now...
Done sending FDR Trust Data
Checking for uncollected logs (44)
Checking for uncollected logs (44)
ERROR: Unable to receive message from FDR 0x7ff71e62fd90 (-2). 0/2 bytes
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Creating partition map (11)
Creating filesystem (12)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[==================================================] 100.0%
Done sending filesystem
Verifying restore (14)
[==================================================] 100.0%
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
Checking filesystems (15)
Checking filesystems (15)
Mounting filesystems (16)
Mounting filesystems (16)
Mounting filesystems (16)
Mounting filesystems (16)
Unknown operation (80)
Sending IsiBootEANFirmware image list
Sending IsiBootNonEssentialFirmware image list
Flashing firmware (18)
[==================================================] 100.0%
Unknown operation (80)
Sending IsEarlyAccessFirmware image list
Sending IsiBootEANFirmware image list
Sending IsiBootNonEssentialFirmware image list
Requesting FUD data (36)
Found IsFUDFirmware component AOP
Found IsFUDFirmware component Homer
Found IsFUDFirmware component RestoreTrustCache
Found IsFUDFirmware component StaticTrustCache
Sending IsFUDFirmware image list
Extracting aopfw-t8010aop.im4p...
Personalizing IMG4 component AOP...
Sending IsFUDFirmware for AOP...
Extracting homer_D101.im4p...
Personalizing IMG4 component Homer...
Sending IsFUDFirmware for Homer...
Extracting 038-77076-035.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending IsFUDFirmware for RestoreTrustCache...
Extracting 038-76885-024.dmg.trustcache...
Personalizing IMG4 component StaticTrustCache...
Sending IsFUDFirmware for StaticTrustCache...
Updating gas gauge software (47)
Updating gas gauge software (47)
Updating Stockholm (55)
Requesting FUD data (36)
Found IsFUDFirmware component AOP
Found IsFUDFirmware component Homer
Found IsFUDFirmware component RestoreTrustCache
Found IsFUDFirmware component StaticTrustCache
Sending IsFUDFirmware image list
Updating baseband (19)
About to send BasebandData...
Sending Baseband TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs
WARNING: size mismatch when parsing MBN file. Continuing anyway.
WARNING: size mismatch when parsing MBN file. Continuing anyway.
Sending BasebandData now...
Done sending BasebandData
Updating Baseband in progress...
About to send BasebandData...
Sending Baseband TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received Baseband SHSH blobs
WARNING: size mismatch when parsing MBN file. Continuing anyway.
WARNING: size mismatch when parsing MBN file. Continuing anyway.
Sending BasebandData now...
Done sending BasebandData
ERROR: device didn't accept BasebandData
ERROR: Unable to successfully restore device
Updating baseband (19)
Cleaning up...
DONE

However I tested same ipsw with same phone on macOS Big Sur and it restored fine with Finder. I opened issue because idevicerestore is my favourite program for restoring firmwares and would like this to have this issue fixed.
Thanks !

nikias, 2020-10-14:
Do you run idevicerestore on macOS?
When you restore again, does it show any "previous restore log"?

Exploite-d, 2020-10-14:
> Do you run idevicerestore on macOS?
> When you restore again, does it show any "previous restore log"?

Yes , i run it on macOS (10.15)
yes it showed previous restore log when I restored again. Unfortunately I closed terminal so I don’t have this log anymore :/

Exploite-d, 2020-10-15:
> Do you run idevicerestore on macOS?
> When you restore again, does it show any "previous restore log"?

i restored once more today I have restore log:
[18:04:52.0184-GMT]{4>6} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[18:04:52.0185-GMT]{4>6} CHECKPOINT WARNING:(FAILURE:-1) Failed to consume AP nonce [0]D(ramrod_clear_ap_nonce: IOServiceGetMatchingService for class AppleMobileApNonce failed)
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0185-GMT]{4>6} CHECKPOINT NOTICE: NVRAM access is not currently available
[18:04:52.0185-GMT]{4>6} CHECKPOINT PROGRESS: START (unknown) -> (presumed_initial_engine_no_return)
[18:04:52.0185-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x0400] umask
[18:04:52.0185-GMT]{4>6} CHECKPOINT END: MAIN:[0x0400] umask
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0185-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x0402] setvbuf
[18:04:52.0185-GMT]{4>6} CHECKPOINT END: MAIN:[0x0402] setvbuf
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0186-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x0403] kernel_logger_thread
[18:04:52.0186-GMT]{4>6} CHECKPOINT END: MAIN:[0x0403] kernel_logger_thread
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0186-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x0406] set_progress_0
found display: primary
display-boot-rotation = 0
display-scale = 2
display-rotation = 0
found applelogo at /usr/share/progressui/applelogo@2x.tga
display: 750 x 1334
Progress Bar Y offset at 112, display class 8
powering on display
ramrod_display_set_granular_progress_forced: 0.000000
[18:04:52.0201-GMT]{4>6} CHECKPOINT END: MAIN:[0x0406] set_progress_0
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0201-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x0407] start_gasgauge_thread
[18:04:52.0202-GMT]{4>6} CHECKPOINT END: MAIN:[0x0407] start_gasgauge_thread
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0202-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x0408] listen_for_log_client
[18:04:52.0203-GMT]{4>6} CHECKPOINT END: MAIN:[0x0408] listen_for_log_client
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0203-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x040D] create_listen_socket
[18:04:52.0203-GMT]{4>6} CHECKPOINT END: MAIN:[0x040D] create_listen_socket
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[18:04:52.0203-GMT]{4>6} CHECKPOINT BEGIN: MAIN:[0x0404] update_root_mount
[18:04:52.0204-GMT]{4>6} CHECKPOINT END: MAIN:[0x0404] update_root_mount
waiting for matching IOKit service: {
    IOProviderClass = IOResources;
    IOResourceMatch = IONVRAM;
}
[... 3154 lines cut ...]
bbupdater: 	END: bootup
bbupdater: 1602785351.060: BBUFSPrepareDestination: both files not present in backup, removing backup files
bbupdater: 1602785351.060: BBUFSPrepareDestination: FSInit: restore_files=0
bbupdater: 	prepare: Start resetting modem
bbupdater: 1602785351.062[1.0]BBULogIOABP::: received 'kABPEnterLowPowerFailed' error. arg0 = 0 arg1 = 0
bbupdater: 1602785351.076[1.0]BBULogIOABP::: received 'kABPLinkDown' error. arg0 = 0x4 arg1 = 0
bbupdater: 	BEGIN: bootup
bbupdater: 		prepare: Start resetting modem
bbupdater: 1602785351.410[1.0]BBULogIOABP::: received 'kABPEnterLowPowerFailed' error. arg0 = 0x1 arg1 = 0
bbupdater: 1602785351.424[1.0]BBULogIOABP::: received 'kABPLinkDown' error. arg0 = 0x4 arg1 = 0
bbupdater: 		Connecting to Loader...
bbupdater: 		OK
bbupdater: 		Getting Version...
bbupdater: 		OK
bbupdater: 		CHIPID=0x60 SNUM=0x93310829410223337895F126, PKHASH=0x54B405F17AA69CF7ECA8AE32BF0F383CE9DC42D4E909EE59513E69D1649D9153 CERTID=0x54b405f1

bbupdater: 		fusingState: 3
bbupdater: 		queryInfoFirstStageInternal ret status is 0
bbupdater: 		Send PSI-RAM (131068 bytes)
bbupdater: 		Sent PSI sucessfully!
bbupdater: 		Preferred interface for DIAG? 0x8
bbupdater: 		In restoreOS?: Bool = 1
bbupdater: 		Is performManCheckFlow?: Bool = 0
bbupdater: 		IsBBFused? Bool = 1
AMFDRDataCreateCombined: sealingManifestData is NULL. kAMF

truonggiang0710, 2020-11-18:
I have the same issue. @Exploite-d How does fix it ?