libimobiledevice/idevicerestore #374: Not able to restore multiple devices at the same time

rajeevs12, 2020-10-20:
Hello ,
we are trying to restore 2 devices at the same time and getting below error on both devices. Can anyone help here what we are doing wrong .We have latest idevicerestore executable and running it on windows 10.
Attached output of both devices.

[iPhone6sOutput.txt](https://github.com/libimobiledevice/idevicerestore/files/5406154/iPhone6sOutput.txt)
[iphone11output.txt](https://github.com/libimobiledevice/idevicerestore/files/5406156/iphone11output.txt)

nikias, 2020-10-26:
Seems like the code fails to get the device ECID when you pass the UDID. I suggest using `-i <ECID>` instead of using UDID until this issue is fixed.

nSneerfulBike, 2020-10-26:
Have you managed to get it working for Windows? For me it doesn't even work with one device

rajeevs12, 2020-10-27:
> Seems like the code fails to get the device ECID when you pass the UDID. I suggest using `-i <ECID>` instead of using UDID until this issue is fixed.

Hi @nikias Thank you this is  working fine now  with <ECID> option. But<ECID> works when device is in recovery mode .What we are doing now first we send restore command to device with UDID and its get failed and turn the device  into  recovery mode. We then send second restore command on device with <ECID>and it works fine. Is there any command to send device into recovery mode directly from normal mode ?

rajeevs12, 2020-10-27:
> Have you managed to get it working for Windows? For me it doesn't even work with one device

Yes it is working fine with single device on windows 10.

nSneerfulBike, 2020-10-27:
> > Have you managed to get it working for Windows? For me it doesn't even work with one device
> 
> Yes it is working fine with single device on windows 10.

Can you tell me what library are you using and where did you get it?

a-gunter, 2020-10-27:
> > Seems like the code fails to get the device ECID when you pass the UDID. I suggest using `-i <ECID>` instead of using UDID until this issue is fixed.
> 
> Hi @nikias Thank you this is working fine now with `-i <ECID>` option. But '`-i <ECID>` works when device is in recovery mode .What we are doing now first we send restore command to device with UDID and its get failed and turn the device into recovery mode. We then send second restore command on device with `-i <ECID>` and it works fine. Is there any command to send device into recovery mode directly from normal mode ?

Use the ideviceenterrecovery tool from libimobiledevice:
`ideviceenterrecovery -u {UDID}`

rajeevs12, 2020-10-28:
> Use the ideviceenterrecovery tool from libimobiledevice:
> `ideviceenterrecovery -u {UDID}`

Many thanks

rajeevs12, 2020-10-28:
> Many thanks

closed 2020-10-28 by rajeevs12

rajeevs12, 2020-10-28:
> > > Have you managed to get it working for Windows? For me it doesn't even work with one device
> > 
> > 
> > Yes it is working fine with single device on windows 10.
> 
> Can you tell me what library are you using and where did you get it?

we get them from https://github.com/libimobiledevice

nSneerfulBike, 2020-10-28:
> > > > Have you managed to get it working for Windows? For me it doesn't even work with one device
> > > 
> > > 
> > > Yes it is working fine with single device on windows 10.
> > 
> > 
> > Can you tell me what library are you using and where did you get it?
> 
> we get them from https://github.com/libimobiledevice

What did you use to compile it? I am really struggling to find a way to compile it on Windows

airwolf2026, 2021-01-14:
> What did you use to compile it? I am really struggling to find a way to compile it on Windows

Using VS Code or Visual studio ,
If you don't want to change a lot warning or errors,
you hava to set a lot predefine marco like '_CRT_SECURE_NO_WARNINGS
_WINSOCK_DEPRECATED_NO_WARNINGS' .