libimobiledevice/idevicerestore #375: iPhone 12: ERROR: can't match element name 'Misc' to baseband firmware file name

skortela, 2020-10-26:
idevicerestore fails for iPhone 12.

[10:44:17.567][4956] : Updating baseband (19)
[10:44:17.567][4956] : About to send BasebandData...
[10:44:18.048][4956] : ERROR: can't match element name 'Misc' to baseband firmware file name.
[10:44:18.048][4956] : ERROR: Unable to send baseband data
[10:44:18.048][4956] : ERROR: Unable to successfully restore device

[failurelog.txt](https://github.com/libimobiledevice/idevicerestore/files/5437972/failurelog.txt)

daxiongaijingxiang, 2020-10-26:
{ "Misc", "multi_image.mbn" },

nikias, 2020-10-26:
@skortela can you try with this patch:
```
diff --git a/src/restore.c b/src/restore.c
index e35047a..4f167ea 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -1282,6 +1282,8 @@ static const char* restore_get_bbfw_fn_for_element(const char* elem)
                // ICE19 firmware files
                { "RestorePSI2", "restorepsi2.bin" },
                { "PSI2", "psi_ram2.bin" },
+               // Mav20 Firmware file
+               { "Misc", "multi_image.mbn" },
                { NULL, NULL }
        };
```

skortela, 2020-10-27:
Thanks,
Restore failed when device was on recovery mode.

[16:18:38.106][8352] : Waiting for device to disconnect...
[16:18:38.271][15872] : irecv_event_cb: device 000a74a11861401e (udid: N/A) disconnected
[16:18:38.271][8352] : Waiting for device to reconnect in recovery mode...
[16:18:40.333][15872] : irecv_event_cb: device 000a74a11861401e (udid: N/A) connected in Recovery mode
[16:18:40.333][8352] : ERROR: could not retrieve device serial number. Can't continue.

And succeeded in normal mode.

[recovery_mode_failed.txt](https://github.com/libimobiledevice/idevicerestore/files/5443082/recovery_mode_failed.txt)

nikias, 2020-10-27:
Ok thanks for testing. I think I will remove the serial number check. Can you make sure the things like FaceID, the LiDAR scanner etc. work? Thanks for testing!

skortela, 2020-10-27:
FaceID, LiDAR and wireless charging works.

a-gunter, 2020-10-27:
Patch works for me as well, attached regular log here for reference.
[iPhone12_Restore.txt](https://github.com/libimobiledevice/idevicerestore/files/5448073/iPhone12_Restore.txt)

The only thing that stuck out to me was a warning about the SE,ChipID. Running with the -d flag got me this:
```
WARNING: Unknown SE,ChipID 0xd2 detected. Restore might fail.
DEBUG: restore_get_se_firmware_data: using SE,UpdatePayload

```
So I guess 0xd2 can map to SE,UpdatePayload.

nikias, 2020-10-27:
@a-gunter thanks for confirming and pointing out the se,chipid warning.
Also in your case it did find the serial number in recovery mode.

skortela, 2020-10-28:
I placed the normal mode device to recovery mode and tried to reproduce the serial missing error, but could not. Now restore succeeded.

Seems that when restore has failed on earlier idevicerestore version, and device left to some recovery mode, in this case serial cannot be read and restore fails.
I don't know can there be different recovery modes and what differences those has, at least there are definitions for 4 recovery modes: 
```
IRECV_K_RECOVERY_MODE_1   = 0x1280,
IRECV_K_RECOVERY_MODE_2   = 0x1281,
IRECV_K_RECOVERY_MODE_3   = 0x1282,
IRECV_K_RECOVERY_MODE_4   = 0x1283,
```

I removed the serial check, so now device enters to restore mode in this case, however there are more serial checks later and restore fails. On second try restore success as device is already on restore mode.

pekn, 2020-11-04:
I have seen this 'no serial number in recovery mode' error as well (it doesn't happen always).. Otherwise, idevicerestore seems to work just fine.

pekn, 2020-11-04:
If you switch iPhone 12 to DFU mode, then this 'no serial' case happens always.