libimobiledevice/idevicerestore #376: let iPhone12 success restore, Do no thinks!

daxiongaijingxiang, 2020-10-26:
In iPhone12 Should be correct：

static const char* restore_get_bbfw_fn_for_element(const char* elem)
{
	struct bbfw_fn_elem_t {
		const char* element;
		const char* fn;
	};

	struct bbfw_fn_elem_t bbfw_fn_elem[] = {
		// ICE3 firmware files
		{ "RamPSI", "psi_ram.fls" },
		{ "FlashPSI", "psi_flash.fls" },
		// Trek firmware files
		{ "eDBL", "dbl.mbn" },
		{ "RestoreDBL", "restoredbl.mbn" },
		// Phoenix/Mav4 firmware files
		{ "DBL", "dbl.mbn" },
		{ "ENANDPRG", "ENPRG.mbn" },	
		// Mav5 firmware files
		{ "RestoreSBL1", "restoresbl1.mbn" },
		{ "SBL1", "sbl1.mbn" },
		// ICE16 firmware files
		{ "RestorePSI", "restorepsi.bin" },
		{ "PSI", "psi_ram.bin" },
		// ICE19 firmware files
		{ "RestorePSI2", "restorepsi2.bin" },
		{ "PSI2", "psi_ram2.bin" },
		{ "Misc", "multi_image.mbn" },
		{ NULL, NULL }
	};

	int i;
	for (i = 0; bbfw_fn_elem[i].element != NULL; i++) {
		if (strcmp(bbfw_fn_elem[i].element, elem) == 0) {
			return bbfw_fn_elem[i].fn;
		}
	}
	return NULL;
}

daxiongaijingxiang, 2020-10-26:
{ "Misc", "multi_image.mbn" },

nikias, 2020-10-26:
Great find, patch here:
```
diff --git a/src/restore.c b/src/restore.c
index e35047a..4f167ea 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -1282,6 +1282,8 @@ static const char* restore_get_bbfw_fn_for_element(const char* elem)
                // ICE19 firmware files
                { "RestorePSI2", "restorepsi2.bin" },
                { "PSI2", "psi_ram2.bin" },
+               // Mav20 Firmware file
+               { "Misc", "multi_image.mbn" },
                { NULL, NULL }
        };
```
We need to figure out if there are other components to add.

DanTheMann15, 2020-10-27:
i do see two vinyl firmwares
`Firmware/vinyl_06n.vnlfw` and `Firmware/vinyl_06n.vnlfw`

and there are also a few new firmware components, such as:

`Firmware/dcp/iphone13dcp.im4p`
`Firmware/rans.t8101.release.im4p`
`Firmware/ansf.t8101.release.im4p`

additionally, you have some new files inside Firmware/all_flash for iBoot
`iBootData.d52g.RELEASE.im4p` <- iPhone13,1
`iBootData.d53g.RELEASE.im4p` <- iPhone13,2
`iBootData.d53p.RELEASE.im4p` <- iPhone13,3
`iBootData.d54p.RELEASE.im4p` <- iPhone13,4

these new files will take some looking-into.

nikias, 2020-10-27:
The above components should be handled already. There's already a success report here: https://github.com/libimobiledevice/idevicerestore/issues/375#issuecomment-717097287

closed 2020-11-09 by daxiongaijingxiang