libimobiledevice/idevicerestore #377: Issue in IPA file installation

rajeevs12, 2020-10-28:
Hello,
We are getting below error while installing .IPA file using ideviceinstaller.exe on windows 10.This is happing mostly after the firmware restoration process we try to install App on device.

ERROR: Could not connect to lockdownd: SSL error (-5)

Thanks in advance.

nikias, 2020-10-28:
This can happen because the pairing record that is stored on your computer is being used on a device that was restored, where the pairing records are deleted. You need to delete the pairing record locally and pair the device again. `idevicerestore` tries to unpair the device but obviously this won't work when the restore was started in recovery mode for example.

rajeevs12, 2020-11-05:
> This can happen because the pairing record that is stored on your computer is being used on a device that was restored, where the pairing records are deleted. You need to delete the pairing record [...]

Thanks @nikias this helped me lot.
But i am getting another error even after device is paired and trusted. Below is the error.
**Could not connect to lockdownd. Exiting.**
Can you please let me know what i am missing here ?