libimobiledevice/idevicerestore #378: Unable to restore device from Windows

nSneerfulBike, 2020-10-29:
I am on Windows 10 Pro 2004 19041.572 with the latest driver updates both for Windows and for Apple. I have got an iPhone 7 Plus with iOS 14.1 and when I enter the following command:
`idevicerestore -e -y -i <ECID> "C:\Users\vincy\Desktop\a\iPhone_5.5_P3_14.1_18A8395_Restore.ipsw"`
I get the following result:
`Found device in Normal mode
Identified device as d111ap, iPhone9,4
Extracting BuildManifest from IPSW
Product Version: 14.1
Product Build: 18A8395 Major: 18
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-74057-065.dmg
Found ECID <ECID>
Getting ApNonce in normal mode... e4 0c 48 fd b0 b7 37 4f 62 1f 10 39 df f5 30 9e 41 1f 00 0d a1 62 c5 96 b3 ec aa 57 44 0d c3 a9
Trying to fetch new SHSH blob
Getting SepNonce in normal mode... 05 7a d6 34 7d 01 f6 14 56 a2 a7 6c 4b 11 00 11 8d 23 0f f4
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Entering recovery mode...
INFO: device serial number is <SERIAL>
NOTE: device is not in DFU mode, assuming recovery mode.
Recovery Mode Environment:
iBoot build-version=iBoot-6723.11.41
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@3x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (20125 bytes)...
ERROR: Unable to send RestoreLogo component: Unable to find device
ERROR: Unable to send RestoreLogo to device.
ERROR: Unable to send AppleLogo
ERROR: Unable to place device into restore mode`

Thinking this is just a bug that doesn't let the program identify the device in recovery correctly, if I re-run the same command I get the following output:
`Found device in Recovery mode
Identified device as d111ap, iPhone9,4
Extracting BuildManifest from IPSW
Product Version: 14.1
Product Build: 18A8395 Major: 18
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-74057-065.dmg
Found ECID <ECID>
Getting ApNonce in dfu mode... e4 0c 48 fd b0 b7 37 4f 62 1f 10 39 df f5 30 9e 41 1f 00 0d a1 62 c5 96 b3 ec aa 57 44 0d c3 a9
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... dc 40 ff ca 23 35 13 6e 0c 03 67 7e 0f 49 80 f1 ef 80 7a 60
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
NOTE: device is not in DFU mode, assuming recovery mode.
ERROR: could not retrieve device serial number. Can't continue.`

I really can't figure out how to fix this problem. The same command (with obviously different path) works on both a Mac and a VM run on VMWare with MacOS run on the same PC I am trying to use this command. Can anybody help me with this?

XanaduAPI, 2020-11-04:
@nSneerfulBike 

// now finally do the magic to put the device into restore mode
	if (client->mode->index == MODE_RECOVERY) {
		if (client->srnm == NULL) {
			error("ERROR: could not retrieve device serial number. Can't continue.\n");
			//不返回以便能刷没有序列号的机子
			//if (delete_fs && filesystem)
			//	unlink(filesystem);
			//return -1;
		}
		if (recovery_enter_restore(client, build_identity) < 0) {
			error("ERROR: Unable to place device into restore mode\n");
			plist_free(buildmanifest);
			if (client->tss)
				plist_free(client->tss);
			if (delete_fs && filesystem)
				unlink(filesystem);
			return -2;
		}
		recovery_client_free(client);
	}

希望这个对你有所帮助
idevicerestore.c

nSneerfulBike, 2020-11-04:
> @nSneerfulBike
>
> // now finally do the magic to put the device into restore mode
> if (client->mode->index == MODE_RECOVERY) {
> if (client->srnm == NULL) {
> error("ERROR: could not retrieve device serial [...]

Well this was a pretty "old" thread, I have managed to solve it tho. Thank you anyway!

tausift0, 2021-02-13:
> Well this was a pretty "old" thread, I have managed to solve it tho. Thank you anyway!

Hi

I am facing the same issue with my iPhone 7. How did you solve it? Please share.

Thanks in advance.

nSneerfulBike, 2021-02-13:
> > Well this was a pretty "old" thread, I have managed to solve it tho. Thank you anyway!
> 
> Hi
> 
> I am facing the same issue with my iPhone 7. How did you solve it? Please share.
> 
> Thanks in advance.

Are you using Windows? I'm assuming so
I've used msys2 (just type in "msys2 download" on Google Search and you'll find it) on 32 bit, which uses pacman and which I've found to be the only solution to the problem. Then, I've git-cloned the libimobiledevice repo and compiled following the instructions. For every error I've proceeded to fix it.
E.g.: it said that I needed usbmuxd, so I cloned usbmuxd in another folder, compiled it according to the instructions and then recompiled libimobiledevice

Please note that in order to compile all of that you need to install the required packages. For instance, you need MinGW-w64, which is downloaded using:
`$ pacman -S mingw-w64-gcc`
`$ pacman -S mingw-w64-i686-gcc`

Usually you just need to search on Google for the required package (for example "mingw-w64 msys2")

Hope this helps

tausift0, 2021-02-13:
> Are you using Windows? I'm assuming so
> I've used msys2 (just type in "msys2 download" on Google Search and you'll find it) on 32 bit, which uses pacman and which I've found to be the only solution [...]

Thanks for replying. Yes I am using Windows and downloaded a precompiled libi for Windows. I am very new to all this. I will try and compile myself following your instructions.

So you mean when you compiled yourself using msys2 then it solved these issues.

Can you upload your compiled libimobile and send me a download link so that I can try. Will that work?

Thanks.

tausift0, 2021-02-13:
I solved the issue by fixing the Apple drivers. I used 3uTools to uninstall all the old drivers and it solved the issue. But now I am stuck again with a new error related to ASR

```
About to send filesystem...
Connected to ASR
Validating the filesystem
ERROR: Unable to receive data from ASR
ERROR: Unable to receive validation packet
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
Got status message
Status: Fail
```

I read that its related to usbmuxd and @nikias put out some fix. I really don't know how to compile so if it will be really helpful if I can get pre-compiled binaries that will have this issue resolved. Can someone please provide me that. I have to update a lot of iPhones and I am trying to write a script to automate the process that's why I need it.

I am currently trying with the pre-compiled libi from here:
https://github.com/libimobiledevice-win32/imobiledevice-net/releases/

tausift0, 2021-02-14:
@nSneerfulBike Thanks to your conversation with @XanaduAPI [here](https://github.com/libimobiledevice/libusbmuxd/issues/100) I was able to solve all the issues and its working perfectly fine now.

Thanks to you all. It was real fun learning how to do it.

nSneerfulBike, 2021-02-14:
> @nSneerfulBike Thanks to your conversation with @XanaduAPI [here](https://github.com/libimobiledevice/libusbmuxd/issues/100) I was able to solve all the issues and its working perfectly fine now.
> 
> Thanks to you all. It was real fun learning how to do it.

Sorry for not having time to reply to you before and I'm glad that my conversation with @XanaduAPI helped you solving your problem. Ask when you need anything!