libimobiledevice/idevicerestore #379: iOS 14.2 Unknown data request 'Personalized Data' received

a-gunter, 2020-11-02:
Testing with iOS 14.2 (GM release) on Mac OS Catalina; this seems to affect all models.

After verifying the restore, there is a new request for 'Personalized Data' which is unhandled and the restore gauge is then stuck at about 50%. Device is stuck until you kick it back into recovery mode

Here's the debug output for an XSM, the relevant section should be at the end: [iOS_14.2_Restore.txt](https://github.com/libimobiledevice/idevicerestore/files/5478243/iOS_14.2_Restore.txt)
)

Will update as I get more info

a-gunter, 2020-11-02:
Nevermind... I ended up cloning a _way_ outdated source haha. 

On a new system and was very confused when I saw that there were already commits to handle PersonalizedData and yet I couldn't find any of the relevant code on my system... whoops. False alarm :)

closed 2020-11-02 by a-gunter

nikias, 2020-11-02:
Was about to reply. One thing I did notice though is that we seem to have a new "Progress Operation (80)". I didn't encounter that before. Will see if there are updated step names.