libimobiledevice/idevicerestore #381: iOS 14.2 in iPhone 12 not show processing restore

NguyenVanVuong, 2020-11-09:
Hello .
I have restore iOS 14.2 in iphone 12, it work but not show processing in restore mode, it black screen.
What happen ?

daxiongaijingxiang, 2020-11-09:
iTnues is OK,you can catch the USB information to analysis of the difference between

NguyenVanVuong, 2020-11-09:
> iTnues is OK,you can catch the USB information to analysis of the difference between

Yes,i have capture but it send restore logo and set picture 4, at this time logo has show but no.I don't know why?

daxiongaijingxiang, 2020-11-10:
What I can confirm is that bootx failed due to USB

daxiongaijingxiang, 2020-11-13:
What I can confirm is that it is standard to send the bootx string to the device, which then appears as a progress bar

nikias, 2020-11-14:
So does it restore, but the screen stays black all the time?

nikias, 2020-11-14:
@a-gunter sorry to tag you here, did you encounter this issue too? Or does it work? Thanks.

NguyenVanVuong, 2020-11-16:
> So does it restore, but the screen stays black all the time?
Hi . @nikias . Yes after recovery screen it stays black all the time.

NguyenVanVuong, 2020-11-16:
Hi @daxiongaijingxiang  it the same , in itunes b_request set 0x01.

a-gunter, 2020-12-11:
I just stumbled upon this now, and yes I can confirm that the screen is black during the actual restore process. After the restore is complete, the final Apple logo + progress bar reappears as the device reboots into normal mode.

Didn't see this initially because I wasn't physically with the phone when I first tried out restoring an iPhone 12, sorry.

daxiongaijingxiang, 2020-12-15:
I have a solution

nikias, 2020-12-21:
@daxiongaijingxiang would you mind sharing some info? I found out it must be related to `RestoreDCP` component, but I don't know how it is being used, I don't have this device so can't dump the USB traffic myself.

a-gunter, 2020-12-29:
I uploaded a usb dump of an iPhone 12 Pro Max restoring to 14.3 (using iTunes on a Mac) [here](https://drive.google.com/drive/folders/1mu85PuxFMKHrhsUiogaCvX7DOmxIZOqK?usp=sharing)

nikias, 2020-12-30:
@a-gunter thanks, really appreciate it! I will have a look as soon as I can.