libimobiledevice/idevicerestore #385: IPhone SE 2 (2020) ASR Error?! / ERROR: Unable to restore device / ASR: Socket unexpectedly closed while trying to read() offset 0x106B20000

yunyhx, 2020-11-22:
Hello im new here sorry for my bad english. I have problem with IPhone SE2 

System Ubuntu 20.04.1 LTS

here is the Problem...
tim@tim-bluechip-BUSINESSline-MT-Pro-G:~$ idevicerestore -e -l
Found device in Recovery mode
Identified device as d79ap, iPhone12,8
The following firmwares are currently being signed for iPhone12,8:
  [1] 14.2 (build 18B92)
Select the firmware you want to restore: 1
Selected firmware 14.2 (build 18B92)
Verifying 'iPhone12,8_14.2_18B92_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 14.2
Product Build: 18B92 Major: 18
INFO: device serial number is FFWCMA40PLJQ
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone12,8_14.2_18B92_Restore/018-01095-004.dmg'
Found ECID 3511042057666606
Getting ApNonce in recovery mode... 50 8f 95 b8 00 7b 7c c1 e7 ee d1 6c 2e 9a ab e9 3a 4d cb ca 95 bc fd 3e 44 f5 3a 5b 6b af 64 6c 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... b4 24 39 0e 3f 15 61 56 90 2a 43 13 f3 c6 03 63 cb 9c fb c6 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d79.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (1562034 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6723.42.3
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (14810 bytes)...
Extracting h12_ane_fw_metis.im4p...
Personalizing IMG4 component ANE...
Sending ANE (2868604 bytes)...
Extracting aopfw-iphone12caop.im4p...
Personalizing IMG4 component AOP...
Sending AOP (1798260 bytes)...
Extracting AppleAVE2FW_H12.im4p...
Personalizing IMG4 component AVE...
Sending AVE (1148324 bytes)...
Extracting armfw_g12p.im4p...
Personalizing IMG4 component GFX...
Sending GFX (784756 bytes)...
Extracting adc-zelus-d79.im4p...
Personalizing IMG4 component ISP...
Sending ISP (9891932 bytes)...
Extracting t8030pmp.im4p...
Personalizing IMG4 component PMP...
Sending PMP (303220 bytes)...
Extracting 018-01496-004.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (13213 bytes)...
[... 1181 lines cut ...]
[15:03:50.0600-GMT]{4>8} CHECKPOINT END: RESTORED:[0x1613] cleanup_recovery_os_volume
restore-step-ids = {0x11030677:45;0x1103065B:53}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 152
restore-step-user-progress = 29
[15:03:50.0600-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:45;0x1103065B:53;0x11030647:56}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 152
restore-step-user-progress = 29
[15:03:50.0600-GMT]{4>8} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x11030677:45;0x1103065B:53}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image}
restore-step-uptime = 152
restore-step-user-progress = 29
[15:03:50.0600-GMT]{4>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x11030677:45;0x1103065B:53;0x11030648:57}
restore-step-names = {0x11030677:perform_main_os_prepare;0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 152
restore-step-user-progress = 29

ERROR: Unable to restore device
tim@tim-bluechip-BUSINESSline-MT-Pro-G:~$ 

THX for help

nikias, 2020-11-22:
Can you make sure you are using latest code from git, especially libusbmuxd and libimobiledevice? I recently pushed a change that from my understanding should help fixing these kind of issues.

yunyhx, 2020-11-22:
I'm not sure where can I see if it's the current version?

yunyhx, 2020-11-22:
tim@tim-bluechip-BUSINESSline-MT-Pro-G:~$ sudo apt-get install \
> build-essential \
> checkinstall \
> git \
> autoconf \
> automake \
> libtool-bin \
> libplist-dev \
> libusbmuxd-dev \
> libssl-dev \
> usbmuxd
[sudo] Passwort für tim: 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
autoconf ist schon die neueste Version (2.69-11.1).
automake ist schon die neueste Version (1:1.16.1-4ubuntu6).
git ist schon die neueste Version (1:2.25.1-1ubuntu3).
libssl-dev ist schon die neueste Version (1.1.1f-1ubuntu2).
libtool-bin ist schon die neueste Version (2.4.6-14).
checkinstall ist schon die neueste Version (1.6.2+git20170426.d24a630-2ubuntu1).
build-essential ist schon die neueste Version (12.8ubuntu1.1).
libplist-dev ist schon die neueste Version (2.2.0-4).
libusbmuxd-dev ist schon die neueste Version (2.0.2-3).
usbmuxd ist schon die neueste Version (1.1.1-2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
tim@tim-bluechip-BUSINESSline-MT-Pro-G:~$

nikias, 2020-11-23:
You only compiled libimobiledevice, but you might also need to compile libusbmuxd from source. But you can try. Make sure to run `sudo ldconfig` after installing the libs and make sure that idevicerestore is using the new ones. Run
```
ldd `which idevicerestore`
```
and check the output if it points to `/usr/local/lib/libimobiledevice*`.

arceuss, 2020-11-29:
> You only compiled libimobiledevice, but you might also need to compile libusbmuxd from source. But you can try. Make sure to run `sudo ldconfig` after installing the libs and make sure that [...]

im getting the same issue and it does point to /usr/local/lib/libimobiledevice + lubusbmuxd is installed