libimobiledevice/idevicerestore #388: iPhone 12 Series, iOS 14.3, fails consistently after sending NORData (FDR)

a-gunter, 2020-12-16:
This seems to affect the 12 series on iOS 14.3.

iOS 14.2.1 and below works fine, and I have not noticed this issue on other models for iOS 14.3, just the 12 series (12 mini, 12, 12 pro, 12 pro max).

After sending NORData, the device quickly fails to receive message from FDR and fails shortly after. The error message is ERROR: usbmuxd_recv_timeout returned -54 (Connection reset by peer).

I've reproduced this on a couple of phones so far, both of which can restore fine to iOS 14.2.1. Here's the debug log:
[iPhone_12Pro_14.3.txt](https://github.com/libimobiledevice/idevicerestore/files/5704957/iPhone_12Pro_14.3.txt)

NguyenVanVuong, 2020-12-18:
In recovery mode send restore add rsep.

a-gunter, 2020-12-18:
> In recovery mode send restore add rsep.

@NguyenVanVuong aren't we already sending that in RestoreSEPImageData? I tried adding the RestoreSEP into the firmware_files dict but that was unsuccessful.

a-gunter, 2020-12-18:
For whatever reason, the device disconnects after the NORData is sent on 14.3, which doesn't happen on earlier iOS versions. I've only tested this on Mac

daxiongaijingxiang, 2020-12-21:
I have a solution

yogendra123456, 2020-12-21:
hi  @daxiongaijingxiang ;

I have the same problem. Can you tell me the the solution.

Thanks in advance....

daxiongaijingxiang, 2020-12-21:
you can Hook iTnues messages, you can see your code enterRestoreModel error!  your miss RestoreSEP file and rsepfirmware command.

vipinbeni, 2020-12-21:
hi @daxiongaijingxiang  i also facing same issue please  tell what is going wrong what to do for this issue  
iOS 14.3, fails consistently after sending NORData (FDR)

Thanks a lot

nikias, 2020-12-21:
I made a patch after looking into MobileDevice framework. In fact, there is a new `rsepfirmware` command that is sent after the RestoreSEP image is uploaded to the device. I don't have a device of this generation so I can't test myself. Let me know if this works:

https://gist.github.com/nikias/c11c856b6cfff41a07d5299fd57167df

I also found something that could be related to the blank screen problem (#381), apparently it uses the new `RestoreDCP` component for that, but I haven't figured out yet how it is used. Need to check a USB dump I guess...

Note: "FDR error" is not an actual error. The connection is closed, but the condition is not handled correctly and thus reported as an error. I didn't find the time yet to take care of that.

a-gunter, 2020-12-21:
Thanks @nikias, this patch works. And thanks for the note, I thought about changing the title to this issue but didn't get around to it.

I can help out with a USB dump from iTunes / idevicerestore to get more info on the black screen.

yogendra123456, 2020-12-22:
hi @nikias and @daxiongaijingxiang,

After update above patch **resolved FDR error** but now i am getting such type error. may be it is a connection error but i am not getting it...

About to send filesystem...
Connecting to ASR
10:37:25 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:432 idevice_connect(): ERROR: Connecting to usbmuxd failed: -107 (Unknown error)
Retrying connection...
Received 272 bytes:
C:\libimobiledevice-vs-master\idevicerestore\src\common.c:printing 272 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Checksum Chunks</key>
        <true/>
        <key>Command</key>
        <string>Initiate</string>
</dict>
</plist>
Connected to ASR
Validating the filesystem
10:37:27 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:605 idevice_connection_send(): internal_connection_send 612, sent 612
Received 325 bytes:
C:\libimobiledevice-vs-master\idevicerestore\src\common.c:printing 325 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Command</key>
        <string>OOBData</string>
        <key>OOB Length</key>
        <integer>72</integer>
        <key>OOB Offset</key>
        <integer>0</integer>
</dict>
</plist>
10:37:27 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:605 idevice_connection_send(): internal_connection_send 72, sent 72
Received 325 bytes:
C:\libimobiledevice-vs-master\idevicerestore\src\common.c:printing 325 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Command</key>
        <string>OOBData</string>
        <key>OOB Length</key>
        <integer>64</integer>
        <key>OOB Offset</key>
        <integer>0</integer>
</dict>
</plist>
10:37:27 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:605 idevice_connection_send(): internal_connection_send 64, sent 64
10:37:27 C:\libimobiledevice-vs-master\libimobiledevice\src\idevice.c:739 internal_connection_receive(): ERROR: usbmuxd_recv returned -108 (Unknown error)
ERROR: Unable to receive data from ASR
ERROR: Unable to receive validation packet
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device

Thanks in advanced..

closed 2021-01-04 by a-gunter