libimobiledevice/idevicerestore #390: ERROR: Unable to receive data from ASR

vipinbeni, 2020-12-21:
Please help same problem in Recovery asa ell as normal mode 

C:\Users\Administrator\Desktop\libimobile2019-master>idevicerestore -e -l -y
Found device in Recovery mode
Identified device as n53ap, iPhone6,2
NOTE: Running non-interactively, automatically selecting latest available version
Selected firmware 12.5 (build 16H20)
Verifying 'iPhone_4.0_64bit_12.5_16H20_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.5
Product Build: 16H20 Major: 16
INFO: device serial number is DX3TH4ZCFRC5
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-87199-018.dmg
Found ECID 7754761725176
Getting ApNonce in recovery mode... ef b1 f7 0a c1 e3 94 62 dc e2 4c 68 64 7e cf ee ac 73 df dc
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 38 03 5e 51 f7 ea d9 7f 46 df fa e3 38 f0 cb 8d 6a 39 a2 9a
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (731057 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (11857 bytes)...
Extracting 038-87170-018.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (9204 bytes)...
ramdisk-size=0x10000000
Extracting 038-87170-018.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (91608302 bytes)...
Extracting DeviceTree.n53ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (109009 bytes)...
Extracting kernelcache.release.iphone6...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (13337001 bytes)...
Waiting for device to enter restore mode...
About to restore device...
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device c0006e3b9c9a6641433e2b47f64e03551ab0c2df has successfully entered restore mode
Hardware Information:
BoardID: 2
ChipID: 35168
UniqueChipID: 7754761725176
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.iphone6.RELEASE.im4p...
Personalizing IMG4 component LLB...
[... 993 lines cut ...]
[05:12:06.0398-GMT]{4>7} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x065B] asr_and_invert_image [0]D(failed to restore APFS image)
restore-step-results = {0x1107065B:{0:14}}
restore-step-codes = {0x1107065B:{0:14}}
restore-step-domains = {0x1107065B:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107065B:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 19
restore-step-user-progress = 1
[05:12:06.0403-GMT]{4>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=1 [sync=true] (first failure)
[05:12:06.0404-GMT]{4>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:49;0x11030647:50}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030647:cleanup_check_result}
restore-step-uptime = 19
restore-step-user-progress = 1
[05:12:06.0407-GMT]{4>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103065B:49}
restore-step-names = {0x1103065B:asr_and_invert_image}
restore-step-uptime = 19
restore-step-user-progress = 1
[05:12:06.0411-GMT]{4>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103065B:49;0x11030648:51}
restore-step-names = {0x1103065B:asr_and_invert_image;0x11030648:cleanup_send_final_status}
restore-step-uptime = 19
restore-step-user-progress = 1

ERROR: Unable to restore device

vipinbeni, 2020-12-22:
Hi @nikias Sir  ,

Please help me still getting this issue 
Found device in Recovery mode
Identified device as n53ap, iPhone6,2
NOTE: Running non-interactively, automatically selecting latest available version
Selected firmware 12.5 (build 16H20)
Verifying 'iPhone_4.0_64bit_12.5_16H20_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.5
Product Build: 16H20 Major: 16
INFO: device serial number is DX3TH4ZCFRC5
Device supports Image4: true
Variant: Customer Erase Install (IPSW)

.....

Personalizing IMG4 component iBoot...
Extracting sep-firmware.n53.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n53.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Checking filesystems (15)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 000002552229CDD0 (-2). 0/2 bytes
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Creating partition map (11)
Creating filesystem (12)
About to send filesystem...
Connected to ASR
Validating the filesystem
ERROR: Unable to receive data from ASR
ERROR: Unable to receive validation packet
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
Got status message
Status: Fail

vipinbeni, 2021-01-11:
Hi Please help to resolve this error 

libmobiledevice\libimobiledevice-vs\libimobiledevice\src\idevice.c:523 internal_connection_receive(): ERROR: usbmuxd_recv returned -108 (Unknown error)
ERROR: Unable to receive data from ASR
ERROR: Unable to receive validation packet
ERROR: ASR was unable to validate the filesystem
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
12:14:21 J:\libmobiledevice\libimobiledevice-vs\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
12:14:21 J:\libmobiledevice\libimobiledevice-vs\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 48498 bytes following
12:14:21 J:\libmobiledevice\libimobiledevice-vs\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 48498 bytes
12:14:21 J:\libmobiledevice\libimobiledevice-vs\libimobiledevice\src\property_list_service.c:250 internal_plist_receive_timeout(): supress printing 48498 bytes plist...

xing181512, 2022-05-26:
@vipinbeni  I have the same error, have you solved it? how to solve