libimobiledevice/idevicerestore #391: Error in building using mingw

vipinbeni, 2020-12-21:
Please help where  i am doing wrong 

[error.txt](https://github.com/libimobiledevice/idevicerestore/files/5723858/error.txt)

msys64\home\Administrator\builds\idevicerestore\src/socket.c:343: undefined reference to `__imp_shutdown'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-socket.o: in function `socket_close':
C:\msys64\home\Administrator\builds\idevicerestore\src/socket.c:348: undefined reference to `__imp_closesocket'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-socket.o: in function `socket_receive_timeout':
C:\msys64\home\Administrator\builds\idevicerestore\src/socket.c:376: undefined reference to `__imp_recv'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\msys64\home\Administrator\builds\idevicerestore\src/socket.c:376: undefined reference to `__imp_recv'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\msys64\home\Administrator\builds\idevicerestore\src/socket.c:376: undefined reference to `__imp_recv'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: idevicerestore-socket.o: in function `socket_send':
C:\msys64\home\Administrator\builds\idevicerestore\src/socket.c:391: undefined reference to `__imp_send'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:490: idevicerestore.exe] Error 1
make[2]: Leaving directory '/home/Administrator/builds/idevicerestore/src'
make[1]: *** [Makefile:428: all-recursive] Error 1
make[1]: Leaving directory '/home/Administrator/builds/idevicerestore'
make: *** [Makefile:360: all] Error 2

nirhen, 2020-12-21:
i got you https://github.com/libimobiledevice/idevicerestore/pull/387

vipinbeni, 2021-01-08:
Working

closed 2021-01-08 by vipinbeni