libimobiledevice/idevicerestore #393: iPhone 12 mini restoring firmware failed

rohit713, 2021-01-11:
When I am going to restore the iPhone 12 mini with iPhone13,1_14.3_18C66_Restore.ipsw downloaded firmware. It give me the following error-
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to receive message from FDR 0X55e5c2cf4d90 (-2). 0/2 bytes
ERROR: Unable to restore device

Please help me to resolve this issue.

demhademha, 2021-01-11:
What mode is your device in?

rohit713, 2021-01-11:
when I start the process it was in normal and activated mode. But now it is in recovery mode.

demhademha, 2021-01-11:
Could you provide the full log?

rohit713, 2021-01-11:
idevicerestore -u 00008101-000338A111F0001E -i 0x338a111f0001e -e /myData/iPhone13,1_14.3_18C66_Restore.ipsw -C /myData/
Found device in Recovery mode
Identified device as d52gap, iPhone13,1
Extracting BuildManifest from IPSW
Product Version: 14.3
Product Build: 18C66 Major: 18
INFO: device serial number is C6KDNDBA0GPP
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.

################################ [ WARNING ] #################################
You are about to perform an ERASE restore. ALL DATA on the target device
will be IRREVERSIBLY DESTROYED. If you want to update your device without
erasing the user data, hit CTRL+C now and restart without -e or --erase
command line switch.
If you want to continue with the ERASE, please type YES and press ENTER.
##############################################################################
>
Invalid input. Please type YES or hit CTRL+C to abort.
> yes
Invalid input. Please type YES or hit CTRL+C to abort.
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/data/BITRASER/General/bitR_Firmware/iPhone13,1_14.3_18C66_Restore/038-82963-078.dmg'
Found ECID 906689371963422
Getting ApNonce in recovery mode... ad c2 c1 20 aa fe 8c 6d e7 68 5c b9 cb 69 34 eb 4b 60 a1 c8 d2 33 40 8e fe 27 db 3e db e4 1e 74
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 06 e2 8a f7 70 b9 5e ca 70 21 4b 6d 26 bf cd d9 ce db 5a d3
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d52g.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (992454 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6723.62.3
iBoot build-style=RELEASE
Sending

rohit713, 2021-01-11:
idevicerestore -u 00008101-000338A111F0001E -i 0x338a111f0001e -e /data/iPhone13,1_14.3_18C66_Restore.ipsw -C /data
Found device in Recovery mode
Identified device as d52gap, iPhone13,1
Extracting BuildManifest from IPSW
Product Version: 14.3
Product Build: 18C66 Major: 18
INFO: device serial number is C6KDNDBA0GPP
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> 
Invalid input. Please type YES or hit CTRL+C to abort.
> yes
Invalid input. Please type YES or hit CTRL+C to abort.
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/data/BITRASER/General/bitR_Firmware/iPhone13,1_14.3_18C66_Restore/038-82963-078.dmg'
Found ECID 906689371963422
Getting ApNonce in recovery mode... ad c2 c1 20 aa fe 8c 6d e7 68 5c b9 cb 69 34 eb 4b 60 a1 c8 d2 33 40 8e fe 27 db 3e db e4 1e 74 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 06 e2 8a f7 70 b9 5e ca 70 21 4b 6d 26 bf cd d9 ce db 5a d3 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d52g.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (992454 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6723.62.3
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2340~iphone.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (29192 bytes)...
Extracting h13_ane_fw_styx_d5x.im4p...
Personalizing IMG4 component ANE...
Sending ANE (3051352 bytes)...
Extracting aopfw-iphone13aop.im4p...
Personalizing IMG4 component AOP...
Sending AOP (2258728 bytes)...
Extracting AppleAVE2FW_H13.im4p...
Personalizing IMG4 component AVE...
Sending AVE (1133576 bytes)...
Extracting armfw_g13p.im4p...
Personalizing IMG4 component GFX...
Sending GFX (2388328 bytes)...
Extracting adc-pallas-d5xg.im4p...
Personalizing IMG4 component ISP...
Sending ISP (15784520 bytes)...
Extracting t8101pmp.im4p...
Personalizing IMG4 component PMP...
Sending PMP (399144 bytes)...
Extracting rans.t8101.release.im4p...
Personalizing IMG4 component RestoreANS...
Sending RestoreANS (794867 bytes)...
Extracting iphone13dcp.im4p...
Personalizing IMG4 component RestoreDCP...
Sending RestoreDCP (3815208 bytes)...
Extracting 038-83075-083.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (14929 bytes)...
Extracting SmartIOFirmware_ASCv4.im4p...
[... 51 lines cut ...]
Extracting batterylow1@2340~iphone.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@2340~iphone-lightning.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.d52gap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting liquiddetect@2340~iphone-lightning.im4p...
Personalizing IMG4 component Liquid...
Extracting lowpowermode@2340~iphone-lightning.im4p...
Personalizing IMG4 component LowPowerWallet0...
Extracting recoverymode@2340~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting WirelessPower.iphone13.im4p...
Personalizing IMG4 component WCHFirmwareUpdater...
Extracting iBoot.d52g.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.d52g.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.d52g.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
ERROR: Could not read data (-256). Aborting.
ERROR: Unable to receive message from FDR 0x55e5c2cf4d90 (-2). 0/2 bytes
ERROR: Unable to restore device

a-gunter, 2021-01-11:
Have you updated idevicerestore in the past couple of weeks? This seems like the issue that was fixed with commit 7fd59d571648b0b418b4c3728b0fb64898feba8c

rohit713, 2021-01-11:
I had updated it on december 2 week. Ok I will again check it and after updating the library I will share that issue is resolved or not. And thanks for your help.

rohit713, 2021-01-12:
@a-gunter Thanks for the suggestion, yes Its working after updating this patch.

closed 2021-01-12 by rohit713