libimobiledevice/idevicerestore #395: iBridge 5.1 ERROR: get_option_with_type: option 'NorImageData' value of a different type than expected, using default

armwdev, 2021-01-12:
Most likely the format of “NorImageData” has changed or encrypted after iBridge 5.0.1. From my observation, all the device tree img4 files seem to be encrypted after iBridge 5.0.1 as it's not the case for the iBridge 4.6 and below versions. I've tried to downgrade to 4.6 but got TSS request error. Not sure how to make idevicerestore to accomplish the restore function with this situation. 

Thanks for the reply.

nikias, 2021-01-12:
@rickmark any idea about this maybe?

mh-cbon, 2022-04-03:
I have been facing that error too. It is a "recent" macbook (not sure which)

I changed `restore.c` file to
```
diff --git a/src/restore.c b/src/restore.c
index ef907ba..da737bc 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -1252,13 +1252,14 @@ int restore_send_nor(restored_client_t restore, struct idevicerestore_client_t*
 	plist_dict_set_item(dict, "LlbImageData", plist_new_data((char*)llb_data, (uint64_t) llb_size));
 	free(llb_data);
 
-	if (client->build_major >= 20) {
-		// Starting with M1 macs, it seems that NorImageData is now a dict.
-		// Sending an array like previous versions results in restore success but the machine will SOS after rebooting.
-		norimage = plist_new_dict();
-	} else {
-		norimage = plist_new_array();
-	}
+	// if (client->build_major >= 20) {
+	// 	// Starting with M1 macs, it seems that NorImageData is now a dict.
+	// 	// Sending an array like previous versions results in restore success but the machine will SOS after rebooting.
+	// 	norimage = plist_new_dict();
+	// } else {
+	// 	norimage = plist_new_array();
+	// }
+	norimage = plist_new_dict();
 
 	plist_dict_iter iter = NULL;
 	plist_dict_new_iter(firmware_files, &iter);
@@ -1310,16 +1311,17 @@ int restore_send_nor(restored_client_t restore, struct idevicerestore_client_t*
 		component_data = NULL;
 		component_size = 0;
 
-		if (client->build_major >= 20) {
-			plist_dict_set_item(norimage, component, plist_new_data((char*)nor_data, (uint64_t)nor_size));
-		} else {
-			/* make sure iBoot is the first entry in the array */
-			if (!strncmp("iBoot", component, 5)) {
-				plist_array_insert_item(norimage, plist_new_data((char*)nor_data, (uint64_t)nor_size), 0);
-			} else {
-				plist_array_append_item(norimage, plist_new_data((char*)nor_data, (uint64_t)nor_size));
-			}
-		}
+		// if (client->build_major >= 20) {
+		// 	plist_dict_set_item(norimage, component, plist_new_data((char*)nor_data, (uint64_t)nor_size));
+		// } else {
+		// 	/* make sure iBoot is the first entry in the array */
+		// 	if (!strncmp("iBoot", component, 5)) {
+		// 		plist_array_insert_item(norimage, plist_new_data((char*)nor_data, (uint64_t)nor_size), 0);
+		// 	} else {
+		// 		plist_array_append_item(norimage, plist_new_data((char*)nor_data, (uint64_t)nor_size));
+		// 	}
+		// }
+		plist_dict_set_item(norimage, component, plist_new_data((char*)nor_data, (uint64_t)nor_size));
 
 		free(comp);
 		free(comppath);
```

rebuilt.

And somehow it got 

```
Got status message
Status: Restore Finished
ReverseProxy[Ctrl]: (status=2) Terminated
Cleaning up...
DONE
progress: 6 1.000000
```

:clap: :clap: 
:clap: :clap: 
:clap: :clap: 

But, and i dont really understand that, i still have this error into my last log..
[irestore3.log](https://github.com/libimobiledevice/idevicerestore/files/8405125/irestore3.log)

Anyways, thanks a lot for all of this, i was afraid i had to garbage this computer. Happy to face next steps.

nikias, 2022-04-03:
Yes that's the fix. I need to make a check for T2 and macOS M1 firmwares which use a dict while iPhone devices use an array.

which error do you refer to exactly in your log?

mh-cbon, 2022-04-03:
the one in the title `ERROR: get_option_with_type: option 'NorImageData' value of a different type than expected, using default` then it says `install_iboot_stage2: host failed to provide firmware` at line 3895.

 I have re executed the procedure again to choose option 2, it worked!

But now it is stuck doing inernet recovery and wont let me boot it using a fedora usb stick. Clicking on the uefi icon bring me back to internet recovery. Lets see.

a-gunter, 2022-04-03:
Yeah once you restore T2 it defaults to internet recovery, and I'm not sure if you can boot from a flash drive on T2 unless you explicitly allow it.

I ran into this same issue with the T2s and now I check the SupportedImageTypes from the build manifest in order to determine what type of device I'm running (M1 / T2 vs iPhones / iPads). The T2 devices all show up as 'iBridge' and the M1 devices all contain 'Mac'.

mh-cbon, 2022-04-03:
I am not sure i understood all you said, but ftr and while i was on it, this is the spec i get from irecovery  -q, in case it helps.

```
$ sudo irecovery -q
CPID: 0x8012
CPRV: 0x10
BDID: 0x3a
ECID: 0x0015484202024026
CPFM: 0x03
SCEP: 0x01
IBFL: 0x3c
SRTG: iBoot-3401.0.0.1.16
SRNM: N/A
IMEI: N/A
NONC: 9cc163f9cae9bb365f9da5021319a0c2ac4bf1f3158528c0a8bb9dfe46df94f5
SNON: 305b1abb63f6de9171f26a2f38fb5753eaad4245
MODE: DFU
PRODUCT: iBridge2,14
MODEL: j152fap
NAME: Apple T2 MacBookPro16,1 (j152f)
```

And also here is the value that dfu-util would print before and after the flash.

```
# before
Found Runtime: [05ac:1227] ver=0000, devnum=21, cfg=1, intf=0, path="1-8", alt=0, name="UNKNOWN", serial="CPID:8012 CPRV:10 CPFM:03 SCEP:01 BDID:3A
ECID:0015484202024026 IBFL:3C SRTG:[iBoot-3401.0.0.1.16]"

# after
Found Runtime: [05ac:1227] ver=0000, devnum=19, cfg=1, intf=0, path="1-3", alt=0, name="UNKNOWN", serial="CPID:8012 CPRV:10 CPFM:03 SCEP:01 BDID:3A
 ECID:0015484202024026 IBFL:3C SRTG:[iBoot-3401.0.0.1.16]"
```

I dont know how to read that, i can only figure some numbers has changed.

rickmark, 2022-04-04:
One could in theory turn off the security mode via checkra1n right after restore - a good restore goes to “full security” and has a new APNonce - meaning by default it would be blank and need to use internet recovery.  I’ve noted malicious workstations performing “restores” but instead of the disk being blank (as it should be due to a new disk key) it already has a volume, be weary of this case

nikias, 2022-04-09:
Finally figured out how this works. The commit message says it all. See 190473b75b1f0f142cf878f79d19ae53e0cf8146.

closed 2022-04-09 by nikias