libimobiledevice/idevicerestore #397: Can't restore ios 14.4 on Windows

tausift0, 2021-02-13:
I tried both iPhone X and iPhone 7 and I get the same error.  What am I doing wrong?

idevicerestore "C:\Firmware\iPhone10,6/iPhone10,6.ipsw"
Found device in Normal mode
Identified device as d221ap, iPhone10,6
Extracting BuildManifest from IPSW
Product Version: 14.4
Product Build: 18D52 Major: 18
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-96302-062.dmg
Found ECID 361275545808814
Checking if device requires stashbag...
ERROR: img4_create_local_manifest: Unhandled component 'Ap,SystemVolumeCanonicalMetadata' - can't create manifest
ERROR: Unable to create preboard manifest.

tausift0, 2021-02-13:
And when I try in Recovery mode

idevicerestore "C:\Firmware\iPhone10,6/iPhone10,6.ipsw"
Found device in Recovery mode
Identified device as d221ap, iPhone10,6
Extracting BuildManifest from IPSW
Product Version: 14.4
Product Build: 18D52 Major: 18
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 038-96302-062.dmg
Found ECID 361275545808814
Getting ApNonce in dfu mode... 32 23 8c c8 cc 17 48 48 09 1c b9 b2 24 d8 ce 42 6f e5 d6 6f 6a 47 ce d9 d3 3d 99 79 0c 5c f3 5e
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... b1 1f ad 2c 6c 20 c1 f2 3f b0 96 46 c2 0a 28 29 58 56 4f cf
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
NOTE: device is not in DFU mode, assuming recovery mode.
ERROR: could not retrieve device serial number. Can't continue.

closed 2021-02-13 by tausift0

goopy13, 2021-06-11:
@tausift0 Did you ever find a solution to this?

tausift0, 2021-06-11:
Yes. I learned how to compile libi for Windows and then compiled it myself using the latest release. It had all these problems fixed. If you want I can send a download link for the exe that I compiled or you can compile yourself.

goopy13, 2021-06-14:
@tausift0 Can you point me to the tutorial you used?

tausift0, 2021-06-14:
Check this
https://github.com/libimobiledevice/idevicerestore/issues/378#issuecomment-778547518

fsinnes, 2021-08-02:
@tausift0 can i get a download link?