libimobiledevice/idevicerestore #398: ERROR: Unable to receive message from FDR

Cryptiiiic, 2021-02-13:
The 7fd59d571648b0b418b4c3728b0fb64898feba8c commit 721 for iPhone 12 actually ends up breaking a11 and other devices fdr during restore.
Full Log:
`
Tag found
Extracting iBoot.d22.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Tag found
Personalizing IMG4 component RestoreSEP...
Tag found
Personalizing IMG4 component SEP...
Tag found
common.c:supressed printing 10276528 bytes plist...
Sending NORData now...
Done sending NORData
idevice_event_cb: device REDACTED (udid: ffffffffffffffffffffffffffffffff00000017) disconnected
ERROR: Unable to receive message from FDR 0x7fbb1ad11bc0 (-2). 0/2 bytes
FDR 0x7fbb1ad11bc0 terminating...
ERROR: Could not read data (-256). Aborting.
`
Happens on most devices, in this specific case I was using iPhone10,3 d22ap X to restore to 14.3/14.4

Cryptiiiic, 2021-02-22:
Was able to succeed with restore after commenting out the rsepfirmware command block, this sep panic happens in a11-a13 unless I comment this out: [recovery.c#229](https://github.com/libimobiledevice/idevicerestore/blob/master/src/recovery.c#L229-#L235)

opa334, 2021-02-22:
[A10 seems to be affected too](https://twitter.com/kha5060/status/1363916224908963854/photo/1)

nikias, 2021-02-23:
Are you sure this is idevicerestore and not futurerestore like in the screenshot? I recently restored an iPhone 7 (A10) with the latest codebase without any issues. Also please note that the FDR error is not an actual error condition but a connection closed event. It clearly fails though (seems like the device reboots).

opa334, 2021-02-23:
@nikias Yeah after digging some more, I think you can disregard my A10 concerns. It's either because that user tried to restore to 14.2 (potential SEP incompatiblity) or it's something else unrelated to idevicerestore as many users succeeded restores on A10, just not this guy for some reason. The original issue reported by @Cryptiiiic does exists tho.

Cryptiiiic, 2021-03-03:
@nikias actually this issue is caused by restore mode kernel being incompatible with restoresep, closing as its not related to idr itself, related to my device.

closed 2021-03-03 by Cryptiiiic

akamidzi, 2022-01-06:
15:05:13.980:  Start flash
15:05:13.980:  Checking PC DiskSpace
15:05:13.980:  Detecting Connection
15:05:13.982:  Found a device with mode: Recovery 
15:05:13.982:  Detecting ProductType
15:05:13.982:  matching device type through original information
15:05:13.984:  Checking iDevice firmware G:/IPONE IPSW/5s/iPhone_4.0_64bit_12.5.5_16H62_Restore.ipsw
15:05:13.984:  Extract BuildManifest.plist
15:05:13.984:  Extracting Restore.plist from IPSW
15:05:13.994:  Firmware version: 12.5.5
15:05:13.994:  Product Build: 16H62 Major: 16
15:05:13.996:  Is Image4 supported: TRUE
15:05:13.997:  Extracting Firmware
15:05:14.000:  Reading ECID
15:05:14.000:  Found ECID 4357995126324
15:05:14.000:  Reading NONCE
15:05:14.000:  Requesting SHSH
15:05:15.259:  Enter Recovery Mode
15:05:23.715:  Sending AppleLogo
15:05:23.740:  Sending Ramdisk
15:05:31.409:  Sending DeviceTree
15:05:31.429:  Sending RestoreSep
15:05:31.562:  Sending KernelCache
15:05:39.086:  Restoring
15:05:39.087:  About to restore device... 
15:05:41.267:  Sending RootTicket
15:05:41.282:  Waiting for NAND
15:05:41.403:  Sending NORData
15:05:54.668:  ERROR: Unable to receive message from FDR 0DE5C050 (-4). 0/2 bytes
15:06:43.229:  The forced stop flashing flag was detected and the flashing is being stopped
15:06:43.236:  ERROR: User cancel.
15:06:43.229:  Failed to restore device(-9999)
15:06:43.240:  ERROR: Unable to restore iDevice(-9999).

akamidzi, 2022-01-06:
please  help my email( akamidzi@gmail.com)

Mrkwame, 2024-01-28:
ERROR: Unable to receive message from FDR 0000000003F031A0 (-4). 0/2 bytes

Please help (samuelkusali110@gmail.com