libimobiledevice/idevicerestore #400: On Windows reaching out gg.apple.com:443 causes blocking connection due to untrusted certificate

hinoble, 2021-02-25:
Hi, we have only windows PCs and I had to compile whole idevice* solution in VS. We have barely used firmware update feature (mostly reading ideviceinfo) and there was no problem even with updating ipsw, I believe up to iOS v 10.x but now when we need to update ipsw (with the latest updates of everything, including usbmuxd) I have faced the following behaviour:
...
Mounting filesystems (16)
Mounting filesystems (16)
Mounting filesystems (16)
Mounting filesystems (16)
FDR 03CBCB88 got sync message
Connecting to FDR client at port 49158
FDR Received 131 bytes
Got device identifier 6b8b4567327b23c6643c9869
FDR connected in reply to sync message, starting command thread
FDR 030FEE38 got plist message
FDR Received 59 bytes
FDR sending 52 bytes:
FDR Sent 52 bytes
FDR 030FEE38 waiting for message...
ERROR: Unable to receive message from FDR 030FEE38 (-2). 0/2 bytes
FDR 030FEE38 terminating...
FDR 03CBCB88 got sync message
Connecting to FDR client at port 49158
FDR Received 131 bytes
Got device identifier 6633487374b0dc5119495cff
FDR connected in reply to sync message, starting command thread
FDR 030FEDA8 got proxy message
Got proxy command with 1 bytes
FDR 030FEDA8 proxy command data too short, retrying
FDR 030FEDA8 got proxy message
Got proxy command with 17 bytes
FDR 030FEDA8 Proxy connect request to gg.apple.com:443
FDR 030FEDA8 got payload of 517 bytes, now try to proxy it
Sending 517 bytes of data
FDR 030FEDA8 terminating...

This appears in the log for 3 times.

The further debug info shows this message (log from iPhone 8) also 3 times:

AMFDRHttpMessageSendSync: ^^^^^^^^^^^^^^^^^^^^^^^^^ BEGIN httpRequest ^^^^^^^^^^^^^^^^^^^^^^^^^
_AMFDRHttpLogRequest: https://gg.apple.com/ca/authorize
_AMFDRHttpLogRequest: Content-Type: application/x-pem-file

... lots of geaders with device info and PEM certificate and end with:

_AMFDRHttpMessageSendSync: ^^^^^^^^^^^^^^^^^^^^^^^^^ END httpRequest ^^^^^^^^^^^^^^^^^^^^^^^^^
_AMSupportHttpURLSessionSendSyncNoRetry: HTTP request failed with error Error Domain=NSURLErrorDomain Code=-1200 "An SSL error has occurred and a secure connection to the server cannot be made." UserInfo={NSErrorFailingURLStringKey=https://gg.apple.com/ca/authorize, NSLocalizedRecoverySuggestion=Would you like to connect to the server anyway?, _kCFStreamErrorDomainKey=3, _NSURLErrorFailingURLSessionTaskErrorKey=LocalDataTask <FB0D3508-631C-46B2-A794-47C0E464110D>.<1>, _NSURLErrorRelatedURLSessionTaskErrorKey=(
    "LocalDataTask <FB0D3508-631C-46B2-A794-47C0E464110D>.<1>"
), NSLocalizedDescription=An SSL error has occurred and a secure connection to the server cannot be made., NSErrorFailingURLKey=https://gg.apple.com/ca/authorize, NSUnderlyingError=0x10a207390 {Error Domain=kCFErrorDomainCFNetwork Code=-1200 "(null)" UserInfo={_kCFStreamPropertySSLClientCertificateState=0, _kCFNetworkCFStreamSSLErrorOriginalValue=-9816, _kCFStreamErrorDomainKey=3, _kCFStreamErrorCodeKey=-9816}}, _kCFStreamErrorCodeKey=-9816}
AMSupportHttpURLSessionSendSync: HTTP request failed (status=16)
AMSupportHttpURLSessionSendSync: HTTP request failed, httpResponseData is NULL
_AMFDRHttpMessageSendSync: HTTP request failed (status=16)
_AMFDRHttpMessageSendSync: HTTP request failed, httpResponseDict is NULL
_AMFDRHttpMessageSendSync: HTTP server returned unexpected FDR response code -1
_AMFDRHttpMessageSendSync: ^^^^^^^^^^^^^^^^^^^^^^^^^ Attempt 1 of 3 failed, sleeping for 1 seconds ^^^^^^^^^^^^^^^^^^^^^^^^^

What I "like" is this message: `NSLocalizedRecoverySuggestion=Would you like to connect to the server anyway?` looks like mocking.. 

So whatever I do - I cannot bypass this. 
As far as I understand the phone uses socket to send data to Apple CSR server via my PC and is blocked due to the untrusted certificate? Why it was working before? Why it stopped working afterwards?

Please help me to find a solution and thanks for great job!

nikias, 2021-02-26:
The device creates a proxy through your PC, so the question is can you connect to gg.apple.com securely directly from your PC?

nikias, 2021-02-26:
Uh scratch that, it also shows up as intrusted here. The server closed the connection (-9816 == errSSLClosedNoNotify) so maybe it's just Apple who messed something up?

hinoble, 2021-02-26:
I think they have enforced strict policy for SSL connections. But does idevicerestore work for you with -le flags, Nikias?

nikias, 2021-02-26:
> But does idevicerestore work for you with -le flags, Nikias?

That wouldn't make a difference regarding that specific issue...

Echoooo0827, 2024-11-19:
How to find a way to create a certificate and customize it?