libimobiledevice/idevicerestore #401: Unable to fetch tss for Rose

hinoble, 2021-03-15:
I have some problem with iPhone 11 and 12 (previous  models work)
I use the -l (latest option) ipsw is downloaded from apple, but I am getting this error:
```
Updating Rose (65)

Sending Rose TSS request...

Request URL set to https://gs.apple.com/TSS/controller?action=2

TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.

ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)

ERROR: Unable to fetch Rose ticket
```
iTunes restore without problem. 

Probably there was an update for this chip, but I have missed it (I cannot pull, the code is very modified), could someone point me where to look for solution?

hinoble, 2021-03-16:
I have updated the question with more debug data. Please can someone help me?

hinoble, 2021-03-16:
Ok I have found the problem: the Rap,BoardID and Rap,ChipID values are wrong (read from parameters.plist). Not sure what caused this though.

nikias, 2021-03-16:
It looks like an endian problem, parameters has them as data nodes, for example the Rap,ChipID is 06 20 in bytes, the integer value is 8198 which is 0x2006 in hex. I need to check where the conversion happens.

hinoble, 2021-03-16:
Sorry Nikias, just found MY problem (as usual) - this was old libplist, it was not reading the values correctly. Updated libplist and now it works. Sorry for your list time.

nikias, 2021-03-16:
No worries, I didn't check yet :)

closed 2021-03-16 by nikias

hinoble, 2021-03-16:
oh, I was too fast with conclusion, i just put the Rose functions in a test and tested before starting restore

```
	restore_get_rose_firmware_data(restore, client, build_identity, NULL);

	return 0;

	// start the restore process
```

This way the values were reading correct:

```
        <key>Rap,BoardID</key>
        <integer>8</integer>
        <key>Rap,ChipID</key>
        <integer>8198</integer>
        <key>Rap,ECID</key>
        <integer>-1</integer>
```

but after running a test I've got this again:

```
        <key>Rap,BoardID</key>
        <integer>2048</integer>
        <key>Rap,ChipID</key>
        <integer>1568</integer>
        <key>Rap,ECID</key>
        <integer>4325429784801731584</integer>
```

Probably this code makes the problem:

```
	/* add Rap,* tags from info dictionary to parameters */
	plist_dict_merge(&parameters, p_info);
```

If you are still here we can debug together.

hinoble, 2021-03-16:
You were right. I forgot to set __LITTLE_ENDIAN__ in idevicerestore project.

xing181512, 2022-06-03:
@hinoble  I also have the same problem, tss.c and  libplist  is completely updated. how you solved it？

CyberOneLPDN, 2023-09-09:
Hello i'm facing the same problem 

TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.

and i'm searching for solution to fix it 
do you find why it doesn't work ?