libimobiledevice/idevicerestore #402: [iPhone8] [D20AP] [iPhone10,1] ERROR: Unable to receive message from FDR 0x7f05d8001320 (-2). 0/2 bytes

Un1Gfn, 2021-04-05:
There are multiple errors and title is only the first one. See [idevicerestore.txt](https://github.com/libimobiledevice/idevicerestore/files/6257429/idevicerestore.txt)

(Arch Linux) uname -r 5.11.11\
libimobiledevice [1.3.0](https://github.com/libimobiledevice/libimobiledevice/releases/tag/1.3.0)\
libirecovery [11324701c89a2ef2ace1686fe25eeec9ad8adac8](https://github.com/libimobiledevice/libirecovery/tree/11324701c89a2ef2ace1686fe25eeec9ad8adac8)\
idevicerestore [4c154e82197039fd0991c69f563706bb17eb13ff](https://github.com/libimobiledevice/idevicerestore/tree/4c154e82197039fd0991c69f563706bb17eb13ff)

```plain
$ sha1sum ./iPhone_4.7_P3_14.4.2_18D70_Restore.ipsw
```

[076b9e90ff4eb3eab348334861ccdf6c5f0c5e25](https://www.theiphonewiki.com/wiki/Firmware/iPhone/14.x#iPhone_8)

```plain
# idevicerestore --erase ./iPhone_4.7_P3_14.4.2_18D70_Restore.ipsw
```

[idevicerestore.txt](https://github.com/libimobiledevice/idevicerestore/files/6257429/idevicerestore.txt)

Un1Gfn, 2021-04-05:
Second attempt fails with a different cable (both USB-A to lightning). This time the following becomes the first error encountered.

```plain
[=====================================             ]  72.0%ERROR: Unable to send data to ASR. Sent 0 of 20 bytes.
```

Full log [idevicerestore2.txt](https://github.com/libimobiledevice/idevicerestore/files/6257479/idevicerestore2.txt)

demhademha, 2021-04-05:
Hi, I found building **libusb** and **libplist** from scratch helpful

Un1Gfn, 2021-04-05:
> Hi, I found building **libusb** and **libplist** from scratch helpful

@demhademha Should I build "the latest release" or "the latest commit in master branch"?

nikias, 2021-04-05:
Make sure to also use libusmbuxd from git master.

Un1Gfn, 2021-04-05:
```plain
$ pactree idevicerestore-git -a | tr '|`-' '   '
```

I will try building git-master variant of all the following packages (non-exhaustive).

```plain
idevicerestore-git
  libirecovery-git
    libusb
  libimobiledevice
    libusbmuxd
      libusb
      libplist
    usbmuxd
      libimobiledevice
```

demhademha, 2021-04-05:
Could you ensure that all of the projects are compiled via **git**

Un1Gfn, 2021-04-05:
I will, but I need couple of days 'cause I cannot reinstall all of them on my everyday laptop. That's gonna remove 700+ packages. I'd rather do it on my build PC when I'm back home.