libimobiledevice/idevicerestore #407: MacBook Pro 15'' 2018 won't finish iBridge installation and go to recovery mode

multihad, 2021-05-18:
Found device in Recovery mode
Identified device as j680ap, iBridge2,3
Extracting BuildManifest from IPSW
Product Version: 5.3
Product Build: 18P4556 Major: 18
INFO: device serial number is C02DFDSFDFSDFS exp
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '/Users/macbookpro/Desktop/iBridge2,1,iBridge2,10,iBridge2,12,iBridge2,14,iBridge2,15,iBridge2,16,iBridge2,19,iBridge2,20,iBridge2,21,iBridge2,22,iBridge2,3,iBridge2,4,iBridge2,5,iBridge2,6,iBridge2,7,iBridge2,8_5.3_18P4556_Restore/038-69925-193.dmg'
Found ECID 5928812160352294
Getting ApNonce in recovery mode... 69 18 92 4d e1 13 f2 14 51 64 44 b9 c9 b8 92 a4 57 ac d6 62 54 3c 62 ef e8 b4 42 1b a6 c6 21 25 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 4a 77 13 e1 3d a6 01 62 ae 37 e5 7d 0d 38 f4 2d d1 a0 77 05 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.j680.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (1189528 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-6723.101.4
iBoot build-style=RELEASE
Extracting 038-69900-193.dmg.trustcache...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (10182 bytes)...
ramdisk-size=0x8000000
Extracting 038-69900-193.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (98285100 bytes)...
Extracting DeviceTree.j680ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (35665 bytes)...
Extracting sep-firmware.j680.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (4114154 bytes)...
Extracting kernelcache.release.ibridge2p...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (10026706 bytes)...
Waiting for device to enter restore mode...
About to restore device... 
Connecting now...
Connected to com.apple.mobile.restored, version 15
Device 00008012-0015103926C20026 has successfully entered restore mode
Hardware Information:
BoardID: 11
ChipID: 32786
UniqueChipID: 5928812160352294
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Updating NAND Firmware (58)
Got status message
Status: Disk Failure
Log is available:
[01:21:23.0095-GMT]{3>8} CHECKPOINT NOTICE: Image4 device: AP nonce clearable
entering ramrod_clear_ap_nonce
[01:21:23.0348-GMT]{3>8} CHECKPOINT NOTICE: AP nonce consumed
[01:21:23.0348-GMT]{3>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: auto-boot=false
[01:21:23.0348-GMT]{3>8} CHECKPOINT NOTICE: Pre-existing NVRAM variable: restore-outcome=initial_monitor_no_return
[01:21:23.0349-GMT]{3>8} CHECKPOINT PROGRESS: START (unknown) -> (initial_engine_no_return)
[01:21:23.0349-GMT]{3>8} CHECKPOINT NOTICE: NVRAM access available on initial check
restore-outcome = initial_engine_no_return
executing /usr/sbin/nvram restore-outcome=initial_engine_no_return
[... 350 lines cut ...]
restore-step-user-progress = 0
[01:21:32.0617-GMT]{3>8} CHECKPOINT END: RESTORED:[0x1613] cleanup_recovery_os_volume
restore-step-ids = {0x1103060E:17}
restore-step-names = {0x1103060E:update_NAND_firmware}
restore-step-uptime = 10
restore-step-user-progress = 0
[01:21:32.0617-GMT]{3>8} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103060E:17;0x11030647:20}
restore-step-names = {0x1103060E:update_NAND_firmware;0x11030647:cleanup_check_result}
restore-step-uptime = 10
restore-step-user-progress = 0
[01:21:32.0618-GMT]{3>8} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103060E:17}
restore-step-names = {0x1103060E:update_NAND_firmware}
restore-step-uptime = 10
restore-step-user-progress = 0
[01:21:32.0618-GMT]{3>8} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103060E:17;0x11030648:21}
restore-step-names = {0x1103060E:update_NAND_firmware;0x11030648:cleanup_send_final_status}
restore-step-uptime = 10
restore-step-user-progress = 0

ERROR: Unable to successfully restore device
ERROR: Unable to restore device
macbookpro@MacBook-Pro-2 ~ %

nikias, 2021-05-18:
I think the error is visible here:
```
Update NAND - FTL version mismatch. Erase install required
failed to update NAND
```
You have to perform an erase restore. Please also ALWAYS paste the command line used, not just the output.

multihad, 2021-05-18:
ok Sir

esabook, 2022-01-03:
I just tried on like this device, this error with `idevicerestore -e -l -y -d`
[idevicerestore -e -l -y -d.log](https://github.com/libimobiledevice/idevicerestore/files/7800028/idevicerestore.-e.-l.-y.-d.log)

historically:
delete apfs OSVolume, Preboot, Recovery
xartutil --erase-all

Last known partition:
Apfs container
---- Apfs (unreadable after xartutil)
Ext4

a-gunter, 2022-01-03:
This was the same issue I ran into with BridgeOS where NorImageData was supposed to be a dict instead of an array (mentioned it in Telegram). I only got to test one Mac though, so I couldn't confirm if the issue was for all iBridge devices. 

In the log:
----
get_option_with_type: option 'NorImageData' value of a different type than expected, using default install_iboot_stage2: host failed to provide firmware
----

Looks like this might be the case for all iBridge devices like you suspected, @nikias

mh-cbon, 2022-04-03:
@a-gunter i can confirm this error here. by any chance can you give some guidance in patching this ?