libimobiledevice/idevicerestore #409: on cmake construction

XanaduAPI, 2021-06-01:
About this project, do you plan to support cmake construction?

I use it on windows, and now I use mingw32 to build it. As a result, I can't debug the program well (maybe my knowledge is too narrow?).

So, do you plan to support using cmake / MSVC to build?

neheb, 2021-08-02:
I have (or had) a meson setup that I use since autotools is horrible. Interested?

XanaduAPI, 2021-08-03:
> I have (or had) a meson setup that I use since autotools is horrible. Interested?

Thank you for your reply, but I am currently using cmake and do not intend to switch to other build systems for the time being.

demhademha, 2021-08-03:
> I have (or had) a meson setup that I use since autotools is horrible. Interested?

Yes: https://github.com/libimobiledevice/libimobiledevice/issues/1195
If you could share your `meson.build` file, (alongside any other changes), it'd mean one tool would be done
Regards

neheb, 2021-08-03:
Hmm it turns out I only had libplist, libusbmuxd, and libimobiledevice converted. Unfortunately, those were done in a VM that has since been nuked.

libplist one is here: https://github.com/neheb/libplist/commit/5bd0419a48ef02898d253deba21a9d1d7ebc50a5

It's a bit old. Could use some cleanup.

demhademha, 2021-08-03:
Then again, it seems that more people would prefer cmake over meson.
What do you think

neheb, 2021-08-03:
I disagree.

Having written both, meson is much easier to write and is easier to use than CMake.

XanaduAPI, 2021-08-03:
Cmake can be well used with IDE such as visual studio or clion.

Sorry, I don't like command line / vim editing code very much.

XanaduAPI, 2021-08-03:
Of course, meson is also a good option. At least better than autotools.

neheb, 2021-08-03:
There's no VSCode plugin?

XanaduAPI, 2021-08-03:
> There's no VSCode plugin?

To me, vscode is not as friendly as visual studio / clion. So I don't like using vscode very much

neheb, 2021-08-03:
CLion seems to have support: https://blog.jetbrains.com/clion/2021/01/working-with-meson-in-clion-using-compilation-db/

demhademha, 2021-08-03:
> I disagree.
> 
> 
> 
> Having written both, meson is much easier to write and is easier to use than CMake.

Absolutely

XanaduAPI, 2021-08-03:
> CLion seems to have support: https://blog.jetbrains.com/clion/2021/01/working-with-meson-in-clion-using-compilation-db/

It seems good, but I'm not going to learn it at present.

Well, what we're talking about is that this project may need a more advanced construction method, right?

demhademha, 2021-08-03:
> > CLion seems to have support: https://blog.jetbrains.com/clion/2021/01/working-with-meson-in-clion-using-compilation-db/
> 
> 
> 
> It seems good, but I'm not going to learn it at present.
We can't just use `X` build system - meson would be good as its cross platform, easy to configure (for users) and easy to build upon (for devs)

> 
> 
> Well, what we're talking about is that this project may need a more advanced construction method, right? 
Although autoconf works, (windows support is somewhat lacking). The meson configuration is also `significantly` faster than doing `./configure && make`
Regards