libimobiledevice/idevicerestore #411: TSS fails with M1 MacBook Pro

rnagy, 2021-06-07:
Hi

I've managed to wipe my disk completely while playing around on the M1 and now I am trying
to restore the laptop with idevicerestore, however the TSS request fails while doing so.

Debug log is attached.

```
Found device in Recovery mode
Identified device as j293ap, MacBookPro17,1
The following firmwares are currently being signed for MacBookPro17,1:
  [1] 11.4 (build 20F71)
  [2] 11.3.1 (build 20E241)
  [3] 11.3 (build 20E232)
  [4] 11.2.3 (build 20D91)
  [5] 11.2.2 (build 20D80)
  [6] 11.2.1 (build 20D74)
  [7] 11.2 (build 20D64)
  [8] 11.1 (build 20C69)
  [9] 11.0.1 (build 20B29)
Select the firmware you want to restore: 1
Selected firmware 11.4 (build 20F71)
Verifying 'UniversalMac_11.4_20F71_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.4
Product Build: 20F71 Major: 20
INFO: device serial number is FVFFK08DQ05P
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'UniversalMac_11.4_20F71_Restore/018-17379-086.dmg'
Found ECID 2375052739346462
Getting ApNonce in recovery mode... 12 45 5f 47 c0 43 48 fd 91 9a f6 15 5d 50 02 d6 a5 f2 ca 4d 7f c3 ef a7 6c 37 44 d4 33 19 1a f4 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 29 9f 5b ca 14 e1 67 54 ab 28 5f 6e d5 8d 7c 9c a1 ca 82 85 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... 29 9f 5b ca 14 e1 67 54 ab 28 5f 6e d5 8d 7c 9c a1 ca 82 85 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... 29 9f 5b ca 14 e1 67 54 ab 28 5f 6e d5 8d 7c 9c a1 ca 82 85 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=8&MESSAGE=An internal error occurred.
ERROR: TSS request failed (status=8, message=An internal error occurred.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device (recovery OS Root Ticket)
```
[debug_out.txt](https://github.com/libimobiledevice/idevicerestore/files/6608519/debug_out.txt)

sywesk, 2021-06-07:
Thanks for testing this out ! It does seem like the third TSS call for the recovery OS is failing. I'll look into it as soon as I can get my hands on a M1 mbp. It may not be soon sadly :(

rnagy, 2021-06-07:
Anything i can do to help maybe? :)

nikias, 2021-06-07:
@rnagy your output looks a bit mixed up, could you maybe dump stderr and stdout separately?

rnagy, 2021-06-07:
> @rnagy your output looks a bit mixed up, could you maybe dump stderr and stdout separately?

sure thing. attached.
[debug_stdout.txt](https://github.com/libimobiledevice/idevicerestore/files/6610029/debug_stdout.txt)
[debug_stderr.txt](https://github.com/libimobiledevice/idevicerestore/files/6610030/debug_stderr.txt)

nikias, 2021-06-07:
Ok I found a bad entry in the last request. Can you try and see if it works (or at least gets further) when you remove / comment out these two lines: https://github.com/libimobiledevice/idevicerestore/blob/d2fc0a88fa1219c71247afadf54da2a214521226/src/tss.c#L884-L885

rnagy, 2021-06-07:
> src/tss.c

new debug logs attached

```
root@ubuntu:/media/ubuntu/180C558E0C556830# idevicerestore --erase --latest
Found device in Recovery mode
Identified device as j293ap, MacBookPro17,1
The following firmwares are currently being signed for MacBookPro17,1:
  [1] 11.4 (build 20F71)
  [2] 11.3.1 (build 20E241)
  [3] 11.3 (build 20E232)
  [4] 11.2.3 (build 20D91)
  [5] 11.2.2 (build 20D80)
  [6] 11.2.1 (build 20D74)
  [7] 11.2 (build 20D64)
  [8] 11.1 (build 20C69)
  [9] 11.0.1 (build 20B29)
Select the firmware you want to restore: 1
Selected firmware 11.4 (build 20F71)
Verifying 'UniversalMac_11.4_20F71_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.4
Product Build: 20F71 Major: 20
INFO: device serial number is FVFFK08DQ05P
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'UniversalMac_11.4_20F71_Restore/018-17379-086.dmg'
Found ECID 2375052739346462
Getting ApNonce in recovery mode... 87 4d bf 54 00 5b c8 f1 49 7c 81 3d 08 7a 5e 4d 8b 8c 41 b4 8e 89 6f 98 cb 07 65 66 9b 23 c4 8e 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 37 02 77 10 e1 2c 9a 49 c4 c0 b6 81 97 6f f1 02 f0 df 2c cf 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... 37 02 77 10 e1 2c 9a 49 c4 c0 b6 81 97 6f f1 02 f0 df 2c cf 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... 37 02 77 10 e1 2c 9a 49 c4 c0 b6 81 97 6f f1 02 f0 df 2c cf 
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=69&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=69, message=This device isn't eligible for the requested build.)
Sending TSS request attempt 1... ERROR: Unable to send TSS request
ERROR: Unable to get SHSH blobs for this device (recovery OS Root Ticket)
```

[debug_stdout.txt](https://github.com/libimobiledevice/idevicerestore/files/6610638/debug_stdout.txt)
[debug_stderr.txt](https://github.com/libimobiledevice/idevicerestore/files/6610639/debug_stderr.txt)

sywesk, 2021-06-07:
Hello @rnagy , I opened a merge request that should fix the problem. Can you try it ? 
@nikias is the mastermind behind this PR, don't thank me he did all the work, i just tested :joy:

nikias, 2021-06-07:
@rnagy I merged it, at least the failing TSS should work now. I have to say I am expecting some more failures later on, but let's find out :)

rnagy, 2021-06-07:
Goes further this time, new logs attached.

```
Found device in Recovery mode
Identified device as j293ap, MacBookPro17,1
The following firmwares are currently being signed for MacBookPro17,1:
  [1] 11.4 (build 20F71)
  [2] 11.3.1 (build 20E241)
  [3] 11.3 (build 20E232)
  [4] 11.2.3 (build 20D91)
  [5] 11.2.2 (build 20D80)
  [6] 11.2.1 (build 20D74)
  [7] 11.2 (build 20D64)
  [8] 11.1 (build 20C69)
  [9] 11.0.1 (build 20B29)
Select the firmware you want to restore: 1
Selected firmware 11.4 (build 20F71)
Verifying 'UniversalMac_11.4_20F71_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 11.4
Product Build: 20F71 Major: 20
INFO: device serial number is FVFFK08DQ05P
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'UniversalMac_11.4_20F71_Restore/018-17379-086.dmg'
Found ECID 2375052739346462
Getting ApNonce in recovery mode... 7d a2 89 83 32 c3 95 4a f3 a4 ef 3b fa c2 e9 ac 69 e7 ec 64 da 71 66 6a 83 67 7e 58 d2 3c 73 5b 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... c0 e8 cc f5 07 94 6e da b1 f0 23 0e e9 55 96 ac 68 67 c8 fe 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... c0 e8 cc f5 07 94 6e da b1 f0 23 0e e9 55 96 ac 68 67 c8 fe 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... c0 e8 cc f5 07 94 6e da b1 f0 23 0e e9 55 96 ac 68 67 c8 fe 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.j293.RELEASE.im4p (Firmware/dfu/iBEC.j293.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (484195 bytes)...
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.
```
[debug_stdout.txt](https://github.com/libimobiledevice/idevicerestore/files/6610931/debug_stdout.txt)
[debug_stderr.txt](https://github.com/libimobiledevice/idevicerestore/files/6610932/debug_stderr.txt)

rnagy, 2021-06-07:
Sorry, correct attachment this time. 
[debug_stdout.txt](https://github.com/libimobiledevice/idevicerestore/files/6611046/debug_stdout.txt)
[debug_stderr.txt](https://github.com/libimobiledevice/idevicerestore/files/6611048/debug_stderr.txt)

Also it seems that If i manually power-cycle the machine it detects that the device is coming back
2 times and then it fails with this:

```
Sending RestoreSEP (7664088 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.mac13g (kernelcache.release.mac13g)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (23357915 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 000870190eda001e (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 000870190eda001e (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
```

However, usbmuxd is running and the device is back on the support.apple.com/mac/restore screen.

rnagy, 2021-06-07:
If I try to put it into DFU mode, with holding ctrl-option-shift down as the apple recovery page suggest,
it will fail because i am in DFU mode and not in restore mode. I think there might be a step missing that I should do maybe.

sywesk, 2021-06-07:
@rnagy can you execute the command `lsusb -t -v` on the host when you plug in the macbook in restore mode ? The pid did change for mac minis, i could be different for macbooks

rnagy, 2021-06-07:
```
[10780.657694] usb 2-1: new high-speed USB device number 47 using xhci_hcd
[10780.808283] usb 2-1: New USB device found, idVendor=05ac, idProduct=1281, bcdDevice= 0.00
[10780.808295] usb 2-1: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[10780.808299] usb 2-1: Product: Apple Mobile Device (Recovery Mode)
[10780.808302] usb 2-1: Manufacturer: Apple Inc.
[10780.808305] usb 2-1: SerialNumber: SDOM:01 CPID:8103 CPRV:11 CPFM:03 SCEP:01 BDID:24 ECID:000870190EDA001E IBFL:FD SRNM:[FVFFK08DQ05P]
```
```
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/3p, 480M
        ID 05e3:0610 Genesys Logic, Inc. Hub
        |__ Port 1: Dev 3, If 2, Class=Video, Driver=uvcvideo, 480M
            ID 13d3:56bc IMC Networks 
        |__ Port 1: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
            ID 13d3:56bc IMC Networks 
        |__ Port 1: Dev 3, If 3, Class=Video, Driver=uvcvideo, 480M
            ID 13d3:56bc IMC Networks 
        |__ Port 1: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
            ID 13d3:56bc IMC Networks 
        |__ Port 2: Dev 4, If 0, Class=Chip/SmartCard, Driver=, 12M
            ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 2: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        ID 0781:55ab SanDisk Corp. 
    |__ Port 3: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
        ID 0781:5581 SanDisk Corp. Ultra
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 47, If 0, Class=Application Specific Interface, Driver=, 480M
        ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]
    |__ Port 1: Dev 47, If 1, Class=Vendor Specific Class, Driver=, 480M
        ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
```

sywesk, 2021-06-07:
Okay so it's not in restore mode, but rather in the recovery mode, which is kinda the step before. It seems like your mac did not boot the restore kernel. You can see that in the lsusb output. I'll inspect your logs

rnagy, 2021-06-07:
If i start the device in DFU mod eand then execute idevicerestore, it can see the apple logo and a status bar appearing but the device never goes into recovery mode and just blanks the screen at one point.
[Uploading debug_stdout.txt…]()

sywesk, 2021-06-07:
@rnagy can you try removing line 747 until 859 from tss.c  and check if it works ?

rnagy, 2021-06-07:
No change.

sywesk, 2021-06-07:
Well I removed my message because I misread your output :pray: 
After reading correctly it seemed that the issue is now a bit different, the restore kernel isn't booting. So you managed to get iBEC to start, and to send all the components needed to start the restore kernel.

rnagy, 2021-06-07:
Okay I've removed my comment as well. I've tried with the following builds and neither of them boot up:
```
  [1] 11.4 (build 20F71)
  [2] 11.3.1 (build 20E241)
  [3] 11.3 (build 20E232)
```

sywesk, 2021-06-07:
Seems like it's a recovery issue. e might not be sending everything :thinking:

xeu100, 2021-07-24:
Any progress on this? Running a VM of macOS just to DFU restore my mb is a pain and this tool seems like a great way around that.

sywesk, 2022-03-26:
@cheesycod have you tried building from source the latest version ? marcan managed to dfu m1 mbps using it

cheesycod, 2022-06-07:
it worked

cheesycod, 2022-06-07:
I managed to fix my boot looping Mac with this so thanks

abuyaseen, 2022-09-15:
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.
Local boot, Board 0x8 (j314cap)/Rev 0x6
::
::	BUILD_TAG: iBoot-7429.81.3
::
::	BUILD_STYLE: RELEASE
::
::	USB_SERIAL_NUMBER: SDOM:01 CPID:6001 CPRV:11 CPFM:01 SCEP:01 BDID:08 ECID:000C69DE2E40011E IBFL:ED SRNM:[

abuyaseen, 2022-09-15:
i try dfu restore after send ibss 100%  mac turn off and never back to recovery

abuyaseen, 2022-09-16:
mac@cs-MacBook-Pro idevicerestore % idevicerestore -l
idevicerestore 1.0.0-111-gd0921e4
Found device in DFU mode
ECID: 349xxxxxxxxxxxxx
Identified device as j314cap, MacBookPro18,4
The following firmwares are currently being signed for MacBookPro18,4:
  [1] 12.6 (build 21G115)
  [2] 12.5.1 (build 21G83)
  [3] 12.5 (build 21G72)
  [4] 12.4 (build 21F79)
  [5] 12.3.1 (build 21E258)
  [6] 12.3 (build 21E230)
  [7] 12.2.1 (build 21D62)
  [8] 12.2 (build 21D49)
  [9] 12.1 (build 21C52)
  [10] 12.0.1 (build 21A559)
Select the firmware you want to restore: 5
Selected firmware 12.3.1 (build 21E258)
Verifying 'UniversalMac_12.3.1_21E258_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.3.1
Product Build: 21E258 Major: 21
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Performing macOS restore
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'UniversalMac_12.3.1_21E258_Restore/018-25400-296.dmg'
Getting ApNonce in dfu mode... dc 65 9c 7a b7 73 ab f7 22 07 6b 33 3d 98 78 52 6e fc 55 0f 32 f6 dd 5c 7e 3d 2d 10 23 80 0d 80 
Trying to fetch new SHSH blob
Getting SepNonce in dfu mode... c1 53 ee c4 7c 5c 59 31 05 d9 3e d9 1f 38 2d 14 f8 8d 3c 0e 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in dfu mode... c1 53 ee c4 7c 5c 59 31 05 d9 3e d9 1f 38 2d 14 f8 8d 3c 0e 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in dfu mode... c1 53 ee c4 7c 5c 59 31 05 d9 3e d9 1f 38 2d 14 f8 8d 3c 0e 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBSS.j314c.RELEASE.im4p (Firmware/dfu/iBSS.j314c.RELEASE.im4p)...
Personalizing IMG4 component iBSS...
Sending iBSS (963562 bytes)...
[==================================================] 100.0%
ERROR: Device did not reconnect in DFU or recovery mode. Possibly invalid iBSS. Reset device and try again.
ERROR: Unable to place device into recovery mode from DFU mode
mac@cs-MacBook-Pro idevicerestore %

jfbortolotti, 2023-01-20:
I'm having similar issue as [abuyaseen](https://github.com/abuyaseen) trying to recover an MBA M1:
idevicerestore --latest
idevicerestore 1.0.0-128-g7b89019
Found device in Recovery mode
ECID: 7961607634812958
Identified device as j313ap, MacBookAir10,1
The following firmwares are currently being signed for MacBookAir10,1:
  [1] 13.1 (build 22C65)
  [2] 13.0.1 (build 22A400)
  [3] 13.0 (build 22A380)
  [4] 12.6 (build 21G115)
  [5] 12.5.1 (build 21G83)
  [6] 12.5 (build 21G72)
  [7] 12.4 (build 21F79)
  [8] 12.3.1 (build 21E258)
  [9] 12.3 (build 21E230)
  [10] 12.2.1 (build 21D62)
  [11] 12.2 (build 21D49)
  [12] 12.1 (build 21C52)
  [13] 12.0.1 (build 21A559)
  [14] 11.6 (build 20G165)
  [15] 11.5.2 (build 20G95)
  [16] 11.5.1 (build 20G80)
  [17] 11.5 (build 20G71)
  [18] 11.4 (build 20F71)
  [19] 11.3.1 (build 20E241)
  [20] 11.3 (build 20E232)
  [21] 11.2.3 (build 20D91)
  [22] 11.2.2 (build 20D80)
  [23] 11.2.1 (build 20D74)
  [24] 11.2 (build 20D64)
  [25] 11.1 (build 20C69)
  [26] 11.0.1 (build 20B29)
Select the firmware you want to restore: 4
Selected firmware 12.6 (build 21G115)
Verifying 'UniversalMac_12.6_21G115_Restore.ipsw'...
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 12.6
Product Build: 21G115 Major: 21
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update the device without erasing user data.
Performing macOS restore
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'UniversalMac_12.6_21G115_Restore/078-70950-018.dmg'
Getting ApNonce in recovery mode... 09 e6 59 fa 3c 9d e1 ae 76 60 ae bf 3e 11 4a 2b 77 b8 9e 6b d9 aa f6 89 11 c7 be 61 ff 67 81 c0 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 65 1d 1a c3 ae 4f 29 2d 02 08 9d 26 14 13 9a 8c bf a8 d9 bb 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... 65 1d 1a c3 ae 4f 29 2d 02 08 9d 26 14 13 9a 8c bf a8 d9 bb 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Getting SepNonce in recovery mode... 65 1d 1a c3 ae 4f 29 2d 02 08 9d 26 14 13 9a 8c bf a8 d9 bb 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.j313.RELEASE.im4p (Firmware/dfu/iBEC.j313.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (497779 bytes)...
ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.

I can see the device usb enumeration in recovery mode:  usb 3-1.1: Product: Apple Mobile Device (Recovery Mode)

Looking for some help to get this working on my mba M1 so I can use it again.

cheesycod, 2023-03-15:
@jfbortolotti Use monterey ipsw

cheesycod, 2023-03-15:
**For anyone having issues, just use a monterey IPSW, its more reliable**