libimobiledevice/idevicerestore #415: recovery_enter_restore Cannot enter Restore mode

XanaduAPI, 2021-06-16:
ProductType:iPhone9,1

When I was brushing the machine, I encountered the following error:
----------------------------------------------------------------------------------------------------
Extracting kernelcache.release.iphone9 (kernelcache.release.iphone9)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (15185746 bytes)...
irecv_event_cb: device 000a69822405e526 (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 000a69822405e526 (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode. 
Please make sure that usbmuxd is running. 
----------------------------------------------------------------------------------------------------
Not all devices will be like this, limited to specific devices.
What can I do to continue?

nikias, 2021-06-16:
This can have multiple reasons, could be a bad signature, broken device....

So tell me, how long does it take from
```
Waiting for device to enter restore mode...
```
to
```
irecv_event_cb: device 000a69822405e526 (udid: N/A) connected in Recovery mode
```
?

XanaduAPI, 2021-06-16:
17:11:23.491 "DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response "
17:11:23.492 "NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity "
17:11:23.495 "Extracting sep-firmware.d10.RELEASE.im4p (Firmware/all_flash/sep-firmware.d10.RELEASE.im4p)... "
17:11:23.497 "Personalizing IMG4 component RestoreSEP... "
17:11:23.499 "Tag found "
17:11:23.500 "Sending RestoreSEP (1386971 bytes)... "
17:11:23.531 "DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response "
17:11:23.533 "NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity "
17:11:23.535 "Extracting kernelcache.release.iphone9 (kernelcache.release.iphone9)... "
17:11:23.564 "Personalizing IMG4 component RestoreKernelCache... "
17:11:23.566 "Tag found "
17:11:23.569 "Sending RestoreKernelCache (15185746 bytes)... "
17:11:26.211 "DEBUG: Waiting for device to disconnect... "
17:11:26.663 "irecv_event_cb: device 000a69822405e526 (udid: N/A) disconnected "
17:11:26.664 "Waiting for device to enter restore mode... "
17:12:01.709 "irecv_event_cb: device 000a69822405e526 (udid: N/A) connected in Recovery mode "
17:12:01.711 "ERROR: Device failed to enter restore mode. "
17:12:01.712 "Please make sure that usbmuxd is running. "
17:12:02.210 "found device with ECID 000a69822405e526 "
17:12:02.211 "Setting to interface 0:0 "
17:12:02.213 "WARNING: set interface failed, error -8 "
17:12:02.214 "found device with ECID 000a69822405e526 "
17:12:02.216 "Setting to interface 0:0 "
17:12:02.218 "WARNING: set interface failed, error -8 "
17:12:02.219 "found device with ECID 000a69822405e526 "
17:12:02.221 "Setting to interface 0:0 "
17:12:02.222 "WARNING: set interface failed, error -8 "

Under normal circumstances, it seems that the device should enter a progress bar interface, but it directly retreats to recovery mode

IPSW:iPhone_4.7_P3_14.6_18F72_Restore

nikias, 2021-06-16:
Well I see a ~30 second delay here:
```
17:11:26.664 "Waiting for device to enter restore mode... "
17:12:01.709 "irecv_event_cb: device 000a69822405e526 (udid: N/A) connected in Recovery mode "
```
But if it doesn't show a progress bar then the device didn't like the images that get sent to it...

XanaduAPI, 2021-06-16:
@nikias 
So...
What should I do? Change a firmware?

XanaduAPI, 2021-06-16:
I changed a firmware and got the same error.

17:28:15.017 "DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response "
17:28:15.019 "NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity "
17:28:15.021 "Extracting sep-firmware.d10.RELEASE.im4p (Firmware/all_flash/sep-firmware.d10.RELEASE.im4p)... "
17:28:15.022 "Personalizing IMG4 component RestoreSEP... "
17:28:15.023 "Tag found "
17:28:15.025 "Sending RestoreSEP (1387099 bytes)... "
17:28:15.057 "DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response "
17:28:15.059 "NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity "
17:28:15.061 "Extracting kernelcache.release.iphone9 (kernelcache.release.iphone9)... "
17:28:15.091 "Personalizing IMG4 component RestoreKernelCache... "
17:28:15.092 "Tag found "
17:28:15.095 "Sending RestoreKernelCache (15185930 bytes)... "
17:28:17.743 "DEBUG: Waiting for device to disconnect... "
17:28:18.209 "irecv_event_cb: device 000a69822405e526 (udid: N/A) disconnected "
17:28:18.211 "Waiting for device to enter restore mode... "
17:28:53.256 "irecv_event_cb: device 000a69822405e526 (udid: N/A) connected in Recovery mode "
17:28:53.257 "ERROR: Device failed to enter restore mode. "
17:28:53.259 "Please make sure that usbmuxd is running. "
17:28:53.757 "found device with ECID 000a69822405e526 "
17:28:53.759 "Setting to interface 0:0 "
17:28:53.761 "WARNING: set interface failed, error -8 "
17:28:53.762 "found device with ECID 000a69822405e526 "
17:28:53.764 "Setting to interface 0:0 "
17:28:53.765 "WARNING: set interface failed, error -8 "
17:28:53.766 "found device with ECID 000a69822405e526 "
17:28:53.767 "Setting to interface 0:0 "
17:28:53.768 "WARNING: set interface failed, error -8 "

IPSW:iPhone_4.7_P3_14.7_18G5042c_Restore.ipsw

ShyanJMC, 2021-06-16:
First execute as root;

usbmuxd -f -U root

Then execute 'idevicerestore --latest' as root.

XanaduAPI, 2021-06-17:
> First execute as root;
> 
> usbmuxd -f -U root
> 
> Then execute 'idevicerestore --latest' as root.

I don't think I'm clear enough.

I can brush other devices, but this one can't.

Will this device cause USB muxd to exit? Obviously not.

And I tried many firmware, and I couldn't finish it

ShyanJMC, 2021-06-17:
> > First execute as root;
> > usbmuxd -f -U root
> > Then execute 'idevicerestore --latest' as root.
> 
> I don't think I'm clear enough.
> 
> I can brush other devices, but this one can't.
> 
> Will this device cause USB muxd to exit? Obviously not.
> 
> And I tried many firmware, and I couldn't finish it

I had the same issue, it doesn't matter the firmware, if can not enter in restore mode mode is bacause "usbmuxd" doesn't have properly access to the device. Executing "usbmuxd -f -U root" as root before execute "idevicerestore --latest" will fix the issue.

XanaduAPI, 2021-06-17:
> I had the same issue, it doesn't matter the firmware, if can not enter in restore mode mode is bacause "usbmuxd" doesn't have properly access to the device. Executing "usbmuxd -f -U root" as root [...]

I use it on Windows 10 :(

XanaduAPI, 2021-06-17:
I'm in recovery_ send_ Irecv in kernelcache_ usb_ control_ An output log has been added to the transfer. It is found that the return value is - 1. Will this help?

14:02:07.597 "DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response "
14:02:07.600 "NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity "
14:02:07.603 "Extracting sep-firmware.d10.RELEASE.im4p (Firmware/all_flash/sep-firmware.d10.RELEASE.im4p)... "
14:02:07.605 "Personalizing IMG4 component RestoreSEP... "
14:02:07.607 "Tag found "
14:02:07.609 "Sending RestoreSEP (1386971 bytes)... "
14:02:07.639 "DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response "
14:02:07.643 "NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity "
14:02:07.646 "Extracting kernelcache.release.iphone9 (kernelcache.release.iphone9)... "
14:02:07.680 "Personalizing IMG4 component RestoreKernelCache... "
14:02:07.682 "Tag found "
14:02:07.686 "Sending RestoreKernelCache (15185746 bytes)... "
14:02:08.076 "ERROR: irecv_usb_control_transfer : -1 "
14:02:10.334 "DEBUG: Waiting for device to disconnect... "
14:02:10.843 "irecv_event_cb: device 000a69822405e526 (udid: N/A) disconnected "
14:02:10.847 "[XANADU]:[0]Ignore here for Recovery mode connection... "
14:02:10.851 "Waiting for device to enter restore mode... "
14:02:45.888 "irecv_event_cb: device 000a69822405e526 (udid: N/A) connected in Recovery mode "
----------------------------------------------------------------------------------------------------

XanaduAPI, 2021-06-17:
Well, the same thing happens on MacOS.
:(

fweth, 2021-06-27:
Got the same situation:

```
Sending RestoreKernelCache (19117143 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
```
The device immediately enters restore mode (from what it shows on screen) but maybe not the progress bar mode it is supposed to.  Have `usbmuxd` running in another window as root and tried `idevicerestore` both as root and non-root.  No success so far :(

XanaduAPI, 2021-06-29:
This sentence was not carried out successfully.
irecv_usb_control_transfer(client->recovery->client, 0x21, 1, 0, 0, 0, 0, 5000);

I try to perform other identification, some devices can succeed, some still make mistakes.

demhademha, 2021-07-08:
Have there been any other developments regarding this issue?
I've got an iPad 6, which can't update to 14.6
Could this be a regression somewhere, because I was able to update an iPad 6 to 14.4 a while back

XanaduAPI, 2021-07-12:
At present, there is no progress

nikias, 2022-04-14:
Can this device be successfully restored on macOS or Windows with iTunes?

xanadu-lee, 2022-04-19:
int recovery_send_kernelcache(struct idevicerestore_client_t* client, plist_t build_identity)
{
	const char* component = "RestoreKernelCache";
	irecv_error_t recovery_error = IRECV_E_SUCCESS;

	if (client->recovery == NULL) {
		if (recovery_client_new(client) < 0) {
			return -1;
		}
	}

	if (recovery_send_component(client, build_identity, component) < 0) {
		error("ERROR: Unable to send %s to device.\n", component);
		return -1;
	}

	int ioctl_error = irecv_usb_control_transfer(client->recovery->client, 0x21, 1, 0, 0, 0, 0, 5000);
#if defined(_WIN32)
	if(ioctl_error < 0)
	{
		// Xanadu:修复某些设备无法进入刷机模式的BUG
		irecv_usb_control_transfer(client->recovery->client, 0xA1, 3, 0, 0, 0, 0, 5000);
		irecv_usb_control_transfer(client->recovery->client, 0x41, 3, 0, 0, 0, 0, 5000);
	}
#endif

	if (client->restore_boot_args) {
		char setba[256];
		strcpy(setba, "setenv boot-args ");
		strcat(setba, client->restore_boot_args);
		recovery_error = irecv_send_command(client->recovery->client, setba);
	}

	recovery_error = irecv_send_command(client->recovery->client, "bootx");
	if (recovery_error != IRECV_E_SUCCESS) {
		error("ERROR: Unable to execute %s\n", component);
		return -1;
	}

	return 0;
}

These two sentences can increase the success rate.
		irecv_usb_control_transfer(client->recovery->client, 0xA1, 3, 0, 0, 0, 0, 5000);
		irecv_usb_control_transfer(client->recovery->client, 0x41, 3, 0, 0, 0, 0, 5000);