libimobiledevice/idevicerestore #417: iphone SE A1723 failed to restore

jeremiegithub, 2021-06-18:
Hi.
After an ios update, Iphone se A1723 keeps booting in recovery mode. Update/restore with itunes failed stuck on "verifying iphone restore" and seems idevicesrestore does the same:
FDR Sent 52 bytes
FDR 0xb3a00c88 waiting for message...
ERROR: Unable to receive message from FDR 0xb3a00c88 (-2). 0/2 bytes
FDR 0xb3a00c88 terminating...
FDR 0x175fba0 timeout waiting for command
FDR 0x175fba0 waiting for message...
Enters in a loop here !!!???
Any help would be appreciate thanks !
Jérémie.

Fr translation here in case:
Bonjour. Après une mise à jour mon iphone se a1723 est bloqué en mode récupération/restauration. J'ai essayé de restaurer ios via itunes, mais ce dernier reste bloqué sur "vérification de la restauration auprès d'apple". On dirait que idevicerestore fait la même chose. Merci d'avance pour votre aide.

jeremiegithub, 2021-06-19:
edit: don't want to jailbreak nor to hack anything (unless I have no choice), just make it work again !

nikias, 2021-06-19:
@jeremiegithub if you run it again with `-d` does it print a "previous restore log" ?

jeremiegithub, 2021-06-20:
Hi. Thanks a lot for helping. Here is the log:

pi@raspberrypi:/usr/local/bin $ sudo idevicerestore -d -y -e -R /home/pi/iphone/img/latest > /home/pi/iphone/log.txt
irecv_event_cb: device 001c6dca102b60a6 (udid: N/A) connected in DFU mode
opening device 05ac:1227...
found device with ECID 001c6dca102b60a6
Setting to configuration 1
Setting to interface 0:0
opening device 05ac:1227...
found device with ECID 001c6dca102b60a6
Setting to configuration 1
Setting to interface 0:0
Getting device's ECID for TSS request
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,SystemVolumeCanonicalMetadata
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry LLB
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry OS
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry RestoreLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry RestoreRamDisk
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
[... 211 lines cut ...]
Connecting to FDR client at port 49159
FDR Received 131 bytes
Got device identifier 1f16e9e81190cde766ef438d
FDR connected in reply to sync message, starting command thread
FDR 0x1a2e200 waiting for message...
FDR 0xb3a00d90 waiting for message...
FDR 0xb3a00d90 got plist message
FDR Received 59 bytes
FDR sending 52 bytes:
FDR Sent 52 bytes
FDR 0xb3a00d90 waiting for message...
ERROR: Unable to receive message from FDR 0xb3a00d90 (-2). 0/2 bytes
FDR 0xb3a00d90 terminating...
FDR 0x1a2e200 timeout waiting for command
FDR 0x1a2e200 waiting for message...
No data to read (timeout)
FDR 0x1a2e200 timeout waiting for command
FDR 0x1a2e200 waiting for message...
No data to read (timeout)
FDR 0x1a2e200 timeout waiting for command
FDR 0x1a2e200 waiting for message...
ERROR: Could not read data (-256). Aborting.
idevice_event_cb: device 001c6dca102b60a6 (udid: ffffffffffffffffffffffffffffffff00000001) disconnected
FDR 0x1a2e200 terminating...
ERROR: Unable to restore device

jeremiegithub, 2021-06-20:
I see no track of previous restore tries.

jeremiegithub, 2021-06-20:
strange thing, any idevicexxx commands ends to no device found whatever i'm in recovery or dfu mode but irecovery is able to :
pi@raspberrypi:/usr/local/bin $ irecovery -q
CPID: 0x8003
CPRV: 0x01
BDID: 0x02
ECID: 0x001c6dca102b60a6
CPFM: 0x03
SCEP: 0x01
IBFL: 0x1d
SRTG: N/A
SRNM: FDDWPFG9HTVL
IMEI: N/A
NONC: 048cad03bc6da99e51317958ec8ee885ea139b17
SNON: c714d1bfc22065e89f83282b6da7282cdd54d306
MODE: Recovery
PRODUCT: iPhone8,4
MODEL: n69ap
NAME: iPhone SE