libimobiledevice/idevicerestore #419: idevicerestore: symbol lookup error: idevicerestore: undefined symbol: irecv_send_command_breq

tmhgh, 2021-07-03:
idevicerestore: symbol lookup error: idevicerestore: undefined symbol: irecv_send_command_breq

I get this error when I try to run idevicerestore. Built from the latest clone from this repo

nikias, 2021-07-03:
You need latest code for libirecovery as well.

tmhgh, 2021-07-03:
I am

tmhgh, 2021-07-03:
Thanks for the reply, I am using the latest from the github repos of all these, libplist, libusbmuxd, libimobiledevice, and libirecovery.

nikias, 2021-07-03:
You probably have another libirecovery still around. Make sure it uses the right one. Here's the commit:
https://github.com/libimobiledevice/libirecovery/commit/47934949e0015165a4562b08e824adb3f664c0ea

tmhgh, 2021-07-03:
Okay, thanks I'll check it out.

tmhgh, 2021-07-03:
Thanks, that was the problem. I have a libirecovery from a different repo lying about, Thanks again

closed 2021-07-03 by nikias