libimobiledevice/idevicerestore #421: checksum verification / preboard manifest (download latest method AND local file method wont work)

fsinnes, 2021-07-21:
Here is a transcript of what I am experiencing starting right before downloaded finishes up

downloading: 96%
downloading: 97%
downloading: 98%
downloading: 99%

Verifying 'iPhone_4.7_P3_14.7_18G69_Restore.ipsw'...
ERROR: Can't open 'iPhone_4.7_P3_14.7_18G69_Restore.ipsw' for checksum verification

C:\Users\MET DEV\Downloads\libimobiledevice.1.2.1-r1122-win-x64>idevicerestore iPhone_4.7_P3_14.7_18G69_Restore.ipsw
Found device in Normal mode
Identified device as d10ap, iPhone9,1
Extracting BuildManifest from IPSW
Product Version: 14.7
Product Build: 18G69 Major: 18
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 018-60086-003.dmg
Found ECID 5071055981379622
Checking if device requires stashbag...
ERROR: img4_create_local_manifest: Unhandled component 'Ap,SystemVolumeCanonicalMetadata' - can't create manifest
ERROR: Unable to create preboard manifest.

unsure where to go from here

nikias, 2021-07-22:
Just guessing, but I assume that libzip is to blame here.

fsinnes, 2021-07-22:
@nikias any idea how to go about fixing the issue