libimobiledevice/idevicerestore #425: iPhone 7 Plus, 15.0 beta 4: Unhandled component 'Ap,SystemVolumeCanonicalMetadata'

skortela, 2021-08-05:
I tried to update iPhone 7 Plus using 15.0 beta 4 image.
https://www.theiphonewiki.com/wiki/Beta_Firmware/iPhone/15.x#iPhone_7_Plus

Failed.

`[12:07:05.433][37372] : Found device in Normal mode
[12:07:05.578][37372] : Identified device as d111ap, iPhone9,4
[12:07:05.578][37372] : Extracting BuildManifest from IPSW
[12:07:05.679][37372] : Product Version: 15.0
[12:07:05.679][37372] : Product Build: 19A5307g Major: 19
[12:07:05.718][37372] : Device supports Image4: true
[12:07:05.729][37372] : Variant: Developer Upgrade Install (IPSW)
[12:07:05.729][37372] : This restore will update your device without erasing user data.
[12:07:05.765][37372] : Checking IPSW for required components...
[12:07:05.815][37372] : All required components found in IPSW
[12:07:05.820][37372] : Extracting filesystem from IPSW: 018-62408-008.dmg
[12:07:16.327][37372] : Extracting filesystem from IPSW completed
[12:07:16.332][37372] : Found ECID 8450107637255462
[12:07:16.363][37372] : Checking if device requires stashbag...
[12:07:16.385][37372] : ERROR: img4_create_local_manifest: Unhandled component 'Ap,SystemVolumeCanonicalMetadata' - can't create manifest
[12:07:16.386][37372] : ERROR: Unable to create preboard manifest.
[12:07:16.527][37372] : Process finished
`

demhademha, 2021-08-05:
Perhaps, put your device in recovery mode first, then try

nikias, 2021-08-05:
If you haven't tried from recovery mode yet, let's see if we can skip this component and see if it continues.

skortela, 2021-08-06:
I skipped all unhandled components
```
[09:02:09.850][33208] : img4_create_local_manifest: Unhandled component 'Ap,SystemVolumeCanonicalMetadata' - skipped
[09:02:09.850][33208] : img4_create_local_manifest: Unhandled component 'SystemVolume' - skipped
```
And now it continues to recovery mode and later to restore mode. Finally it stops to unknown data request 'FirmwareUpdaterPreflight'
```
[09:07:49.296][33208] : Updating Baseband completed.
[09:07:49.316][33208] : Updating SE Firmware (59)
[09:07:49.329][33208] : Unknown data request 'FirmwareUpdaterPreflight' received
```
Process does not continue from that point.
Also hangs to same in 'eraser' mode.

nikias, 2021-08-06:
Try this patch:
```diff
diff --git a/src/restore.c b/src/restore.c
index e090594..ff605a4 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -3550,6 +3550,10 @@ int restore_device(struct idevicerestore_client_t* client, plist_t build_identit
                plist_dict_set_item(opts, "RestoreBundlePath", plist_new_string("/tmp/Per2.tmp"));
                // FIXME: not required for iOS 5?
                //plist_dict_set_item(opts, "SourceRestoreBundlePath", plist_new_string("/tmp"));
+
+               plist_dict_set_item(opts, "SupportedDataTypes", restore_supported_data_types());
+               plist_dict_set_item(opts, "SupportedMessageTypes", restore_supported_message_types());
+
                // FIXME: new on iOS 5 ?
                plist_dict_set_item(opts, "SystemImageType", plist_new_string("User"));
                // FIXME: does this have any effect actually?
```

skortela, 2021-08-09:
Thanks Nikias.
Now it succeeded to update.

Btw, is it safe to skip those unhandled components 'Ap,SystemVolumeCanonicalMetadata' & 'SystemVolume'?
Do you commit your changes to master?

nikias, 2021-08-09:
I will commit the changes, yes. I think it is safe to skip it the missing types, they don't seem to be part of the "local manifest". One way to tell for sure is if the device asks for a passcode after the restore and shows a "Restore in Progress" screen, then you know something is missing.

demhademha, 2021-08-09:
Will this change affect < 15.0 beta 4 (for example, 14.7.1)

nikias, 2021-08-09:
yes, I have reports of broken restores, and this should help.

nikias, 2021-08-10:
@skortela check commits 18f6e42f261386f10c93f16d31c927704fdd78a3 and 313b0d5ea1f33ceb78c860745befed8d660f447e.

skortela, 2021-08-10:
SW update succeeded with those commits.

nikias, 2021-08-10:
Perfect, thanks for confirming!

skortela, 2021-08-12:
Works perfectly!

closed 2021-08-12 by skortela

nikias, 2021-08-12:
Great, thanks!

proceso501, 2021-12-18:
Hi, I have the same issue:

"ERROR: img4_create_local_manifest: Unhandled component 'Ap,SystemVolumeCanonicalMetadata' - can't create manifest
ERROR: Unable to create preboard manifest."

How i can solve it? How I can apply the Nikias patch? I have the same problem in Windows 10 and Kali Linux, Just the same. Can you help me? I try by myself many days and dedicated many hours to investigue and i cant do it without help.

My Iphone 11 (12,1) and Ipad 7.5 (6 Gen) was actualiced to the ios 15.1 version and have many problems, i want doengrade, or erase all the ipad and iphone content and install or downgrade to the ios 14 or similar. In the two devices have the same error too.

Thanks you very much in advance.