libimobiledevice/idevicerestore #426: Unkown data request 'Personalized Data' received (IPHONE 8)

slackjeff, 2021-08-05:
Everything works correctly, however it is paused at 50% and I get this message:
Unkown data request 'Personalized Data' received
Use arch linux, and iphone 8 in restore mode.

Version idevicerestore: 1.0.1

nikias, 2021-08-05:
Did you build idevicerestore yourself? Because there is specific code to handle this (see https://github.com/libimobiledevice/idevicerestore/blob/dfa05a8c417e785799a0d8ea0f9a58ed89a13085/src/restore.c#L3260) and I wonder if it's just due to an outdated build.

ksaadDE, 2021-08-14:
> Everything works correctly, however it is paused at 50% and I get this message:
> Unkown data request 'Personalized Data' received
> Use arch linux, and iphone 8 in restore mode.
> 
> Version idevicerestore: 1.0.1

Arch Linux? Which version?
Did you compile the idevicerestore yourself?

fuomag9, 2021-08-17:
Same issue here idevicerestore 1.0.0 from ubuntu apt

demhademha, 2021-08-17:
> Same issue here,  idevicerestore 1.0.0 from ubuntu apt

You need to build everything from source - ubuntu is behind the latest git releases.

closed 2022-04-09 by nikias