libimobiledevice/idevicerestore #428: Got error 'This device isn't eligible for the requested build.'

skortela, 2021-08-13:
I tried to update iPhone XS to latest beta version with idevicerestore. However it fails to TSS server error:
'This device isn't eligible for the requested build.'
What could cause this?

Also seems that whole process get aborted without actual retry.
```
[10:09:54.810][3456] : Found device in Normal mode
[10:09:54.892][3456] : Identified device as d331pap, iPhone11,6
[10:09:54.892][3456] : Extracting BuildManifest from IPSW
[10:09:55.054][3456] : Product Version: 15.0
[10:09:55.054][3456] : Product Build: 19A5318f Major: 19
[10:09:55.060][3456] : Device supports Image4: true
[10:09:55.070][3456] : Variant: Developer Erase Install (IPSW)
[10:09:55.070][3456] : This restore will erase your device data.
[10:09:55.070][3456] : Checking IPSW for required components...
[10:09:55.210][3456] : All required components found in IPSW
[10:09:55.213][3456] : Using cached filesystem from 'C:/ipsw/iPhone11,2,iPhone11,4,iPhone11,6,iPhone12,3,iPhone12,5_15.0_19A5318f_Restore/018-62379-017.dmg'
[10:09:55.213][3456] : Extracting filesystem from IPSW completed
[10:09:55.219][3456] : Found ECID 2636148192706606
[10:09:55.219][3456] : Getting ApNonce in normal mode... b4 6a 0f c6 89 f6 21 6f 1e 3e 0c 75 f3 77 54 8a a8 09 c0 1f 8a cc e9 ad 9f fb ce 3c 52 35 74 a3 
[10:09:55.236][3456] : Trying to fetch new SHSH blob
[10:09:55.236][3456] : Getting SepNonce in normal mode... a4 af 60 dc 4b 2b 4b 00 84 8b b4 60 62 98 4d e3 8c 84 58 f0 
[10:09:55.446][3456] : Request URL set to https://gs.apple.com/TSS/controller?action=2
[10:09:55.446][3456] : Sending TSS request attempt 1... [10:09:56.277][3456] : TSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
[10:09:56.277][3456] : TSS request failed (status=94, message=This device isn't eligible for the requested build.), trying again...
[10:09:58.283][3456] : Request URL set to https://17.171.36.30/TSS/controller?action=2
```

skortela, 2021-08-13:
Here is log:
[idevicerestore.log](https://github.com/libimobiledevice/idevicerestore/files/6982275/idevicerestore.log)

nikias, 2021-08-13:
Thanks, did this device work with an earlier beta like beta 4? This error usually means your device can't be restored because Apple doesn't sign that particular iOS version anymore. But it's the latest beta so hmm...

skortela, 2021-08-13:
Beta 4 image had exact same problem.

a-gunter, 2021-08-13:
same thing also happens for XS Max & 11 Pro Max... so I guess it's consistent across all models that use this .ipsw.

the .ipsw for 7 Plus / 8 Plus doesn't have this issue though. 15.0 beta 5 works fine with latest commits

nikias, 2021-08-13:
Ok we need someone to start a restore with Apple Configurator 2 or Finder or iTunes and dump the network communication after switching it to use http instead of https with:
```bash
defaults write com.apple.AMPDevicesAgent AuthInstallSigningServerURL http://gs.apple.com:80
```
Then we can see what is going on I guess...

nikias, 2021-08-14:
Can you try if this works:
```diff
@@ -2235,6 +2235,10 @@ int get_tss_response(struct idevicerestore_client_t* client, plist_t build_ident
                return -1;
        }

+       if (client->image4supported) {
+               plist_dict_set_item(parameters, "_OnlyFWComponents", plist_new_bool(1));
+       }
+
        /* add tags from manifest */
        if (tss_request_add_ap_tags(request, parameters, NULL) < 0) {
                error("ERROR: Unable to add common tags to TSS request\n");
```

skortela, 2021-08-16:
Update failed. Here is new log.
[new_idevicerestore.log](https://github.com/libimobiledevice/idevicerestore/files/6990167/new_idevicerestore.log)

nikias, 2021-08-16:
Hmm that is weird, the `Yonkers,*` and `Savage,*` tags are not skipped (only for the local manifest).  Can you make sure the patch applied correctly? The components don't have `Trusted: true` set, so they should get filtered out here: https://github.com/libimobiledevice/idevicerestore/blob/master/src/tss.c#L937

skortela, 2021-08-16:
Sorry, my mistake. I will re-test.

nikias, 2021-08-16:
Don't be sorry, it could easily happen that I also screwed up something.

skortela, 2021-08-16:
Now it fails to another error.
```
[14:37:10.723][13044] : Sending iBEC (1385652 bytes)...
[14:37:10.919][13044] : Waiting for device to disconnect...
[14:37:40.937][13044] : ERROR: Device did not disconnect. Possibly invalid iBEC. Reset device and try again.
```
[idevicerestore.log](https://github.com/libimobiledevice/idevicerestore/files/6992124/idevicerestore.log)

nikias, 2021-08-16:
Hmm... does the screen go dark when it sends the iBEC? Can you re-run and add option `-k` ? It will keep a file `iBEC` in your local directory. Then upload it together with the new debug output so I can check if it was incorrectly personalized (stitched).

nikias, 2021-08-16:
Never mind, I am seeing what's missing. The TSS request doesn't have the iBEC in it amongst other components. Will check how to add them properly.

nikias, 2021-08-16:
Undo my previous patch for `idevicerestore.c` and try this new patch:
```diff
diff --git a/src/tss.c b/src/tss.c
index ae46e15..e9eb9c3 100644
--- a/src/tss.c
+++ b/src/tss.c
@@ -933,13 +933,24 @@ int tss_request_add_ap_tags(plist_t request, plist_t parameters, plist_t overrid
                        continue;
                }

+               plist_t info_dict = plist_dict_get_item(manifest_entry, "Info");
+               if (!info_dict) {
+                       continue;
+               }
+
+               if (_plist_dict_get_bool(parameters, "ApSupportsImg4")) {
+                       if (!plist_dict_get_item(info_dict, "RestoreRequestRules")) {
+                               debug("DEBUG: %s: Skipping '%s' as it doesn't have RestoreRequestRules\n", __func__, key);
+                               continue;
+                       }
+               }
+
                if (_plist_dict_get_bool(parameters, "_OnlyFWComponents")) {
                        if (!_plist_dict_get_bool(manifest_entry, "Trusted")) {
-                               debug("DEBUG: %s: Skipping '%s' as it is not trusted", __func__, key);
+                               debug("DEBUG: %s: Skipping '%s' as it is not trusted\n", __func__, key);
                                continue;
                        }

-                       plist_t info_dict = plist_dict_get_item(manifest_entry, "Info");
                        if (!_plist_dict_get_bool(info_dict, "IsFirmwarePayload") && !_plist_dict_get_bool(info_dict, "IsSecondaryFirmwarePayload") && !_plist_dict_get_bool(info_dict, "IsFUDFirmware")) {
                                debug("DEBUG: %s: Skipping '%s' as it is neither firmware nor secondary nor FUD firmware payload\n", __func__, key);
                                continue;
```

skortela, 2021-08-17:
Good work, now update succeeded with that patch.

nikias, 2021-08-17:
Great!

nikias, 2021-08-17:
See commit 9a9ad5dab9a88705bb0b3261220ba2a52acc31dc.

closed 2021-08-17 by skortela