libimobiledevice/idevicerestore #430: Add Levels/Colours to logging Output

stellarpower, 2021-08-14:
Hi,

Trying to factory reset an iPad - a friend gave it to me and don't have windows to hand; would never pay for an Apple device ;) - so afraid I am not in the least knowledgeable.

I am using a docker build and was having some issues, I then rebuilt and got further. The output during the whole process is really great, verbosity is only a good thing in situations like this. However what I think would help would be adding some different logging levels to the output. I have several `Unable to receive message from FDR`, and it appeared to hang for a few minutes, but now looks to have run to completion. Without knowing what the FDR is, it would probably help the inexperienced user like me (and perhaps also for triaging bugs) to know if messages like these can be considered fatal/errors/warnings, etc., to guide the overall process. And for bonus points, some coloured output in the logging would also be great - I know there are several libraries that can just be incorporated in C++ to do this, but afraid I don't know of any pure-C ones.

Thanks for liberating my free iPad! :)

nikias, 2021-08-14:
I very much agree that the logging output needs to be improved, and yes, also colorized! I am currently working on moving some common code all around in the different libraries into a new [libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue) where I also added support for printing colorized output.
The FDR "error" in fact is also not an error, it is usually a "connection closed" event, so the message is a bit misleading indeed.
Thanks for your feedback.

stellarpower, 2021-08-14:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEgAAABICAYAAABV7bNHAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAArSURBVHic7cEBDQAAAMKg909tDjegAAAAAAAAAAAAAAAAAAAAAAAAAAA+DFFIAAEctgHwAAAAAElFTkSuQmCC###24x24:true###][Tracking image][https://github.com/notifications/beacon/ABGF2APPP4UQRPEGGPPM22DT4ZEZHA5CNFSM5CEUXFOKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGWJ5E3A.gif]

I suspected as much, more of a warning w.r.t. the process as a whole. That all sounds good, thanks for responding so quickly!