libimobiledevice/idevicerestore #436: iPhone 5C: couldn't find tag SNON

skortela, 2021-09-22:
I have tested erasure with two different iPhone 5C models. Erasure is failing for same error.
First there is warning about SNON

[13:09:46.374][1076] : [libirecovery] irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string  NONC:05FE405753166F125559E7C9AC558654F107C7E9

Device enters to restore mode and then fails to error:

[13:10:23.095][1076] : Starting FDR listener thread
[13:10:23.096][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:25.100][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:27.113][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:29.127][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:31.140][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:33.142][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:35.144][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:37.158][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:39.172][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:41.185][1076] : [libusbmuxd] usbmuxd_connect: Connect failed, Error code=3
[13:10:41.185][1076] : ERROR: Unable to connect to FDR client (-2)
[13:10:41.185][1076] : ERROR: Failed to start FDR Ctrl channel
[13:10:41.226][1076] : ERROR: Unable to restore device

[iPhone 5C erasure.log](https://github.com/libimobiledevice/idevicerestore/files/7210311/iPhone.5C.erasure.log)

So whats wrong?

skortela, 2021-09-24:
Debug log:
[idevicerestore_debug.log](https://github.com/libimobiledevice/idevicerestore/files/7224480/idevicerestore_debug.log)
 
Note! Earlier version (august) of idevicerestore works.

I can see following error in debug log:
Failed to parse checkpoint result from checkpoint plist

doronz88, 2021-09-27:
I'm not sure, but an FDR socket related issue has just been resolved (#444), eventhough the error seems to occur before this part. Try again with current comitted version.

sywesk, 2021-09-27:
@doronz88 the fix #444 is not related to this sadly. I'm not sure what's causing your problem though. The SNON problem doesn't seem blocking (ie no SEP Nonce in the USB descriptor) but it could be a clue as to where the problem comes from.

I think that we'll also need usbmuxd's logs if possible (see `journalctl - xeu usbmuxd`) @skortela

nikias, 2021-09-28:
The iPhone 5C doesn't have an SEP, so it also doesn't have a SepNonce (SNON). I think the problem here is that it also cannot do FDR, so it fails to connect in this case. The error code 3 is a "connection refused" error, but this is a) not propagated properly from libusbmuxd to libimobiledevice (only results in IDEVICE_E_UNKNOWN_ERROR, I definitely need to improve this as it waits 10 x 2 seconds until giving up), and b) not handled in idevicerestore - in fact this is currently ignored.
The checkpoint error message though might make sense, maybe it is different for older device versions. It fails here:
```c
			// Get checkpoint complete
			node = plist_dict_get_item(message, "CHECKPOINT_COMPLETE");
			if (!node || plist_get_node_type(node) != PLIST_BOOLEAN) {
				debug("Failed to parse checkpoint result from checkpoint plist\n");
				return -1;
			}
```
@skortela can you reconfigure libimobiledevice with `--enable-debug` and run it again? Then we also see all the messages (including the checkpoint ones) when you run idevicerestore with `-d`.

skortela, 2021-09-28:
`-enable-debug` seemed to be correct param.

Here is debug log:
[debug_log_28_9.log](https://github.com/libimobiledevice/idevicerestore/files/7243918/debug_log_28_9.log)

nikias, 2021-09-28:
Same output. You need to basically do `./configure --enable-debug` in your libimobiledevice source tree, and don't forget to `sudo make install`.

skortela, 2021-09-28:
Ok, Maybe now..
[newlog.log](https://github.com/libimobiledevice/idevicerestore/files/7244273/newlog.log)

nikias, 2021-09-30:
@skortela I didn't manage to fix the FDR connection refused handling yet, but I think the checkpoint message parsing error could be worked around with this:
```diff
diff --git a/src/restore.c b/src/restore.c
index b1089bf..47c93dd 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -3764,7 +3764,7 @@ int restore_device(struct idevicerestore_client_t* client, plist_t build_identit
                else if (!strcmp(type, "CheckpointMsg")) {
                        uint64_t ckpt_id;
                        uint64_t ckpt_res;
-                       uint8_t ckpt_complete;
+                       uint8_t ckpt_complete = 0;
                        // Get checkpoint id
                        node = plist_dict_get_item(message, "CHECKPOINT_ID");
                        if (!node || plist_get_node_type(node) != PLIST_UINT) {
@@ -3781,11 +3781,9 @@ int restore_device(struct idevicerestore_client_t* client, plist_t build_identit
                        plist_get_uint_val(node, &ckpt_res);
                        // Get checkpoint complete
                        node = plist_dict_get_item(message, "CHECKPOINT_COMPLETE");
-                       if (!node || plist_get_node_type(node) != PLIST_BOOLEAN) {
-                               debug("Failed to parse checkpoint result from checkpoint plist\n");
-                               return -1;
+                       if (PLIST_IS_BOOLEAN(node)) {
+                               plist_get_bool_val(node, &ckpt_complete);
                        }
-                       plist_get_bool_val(node, &ckpt_complete);
                        if (ckpt_complete)
                                info("Checkpoint %" PRIu64 " complete with code %" PRIu64 "\n", ckpt_id, ckpt_res);
                }
```
I think if `CHECKPOINT_COMPLETE` is not present we can assume it is considered `false`. Let me know if this works.

skortela, 2021-10-04:
Thanks @nikias, no it succeeded!

Here is full log if interested:
[succeeded.log](https://github.com/libimobiledevice/idevicerestore/files/7275724/succeeded.log)

closed 2021-10-04 by skortela