libimobiledevice/idevicerestore #439: iPhone 5 cannot be restored

lorinbudaca, 2021-09-23:
Tried erasing an iPhone 5 and an iPad 4 but encountered this issue. I understand that usbmuxd fails to connect to the device in FDR mode. 
Compiled with the latest commits.

```
~# idevicerestore -e -d iPhone_4.0_32bit_10.3.4_14G61_Restore
irecv_event_cb: device <CONFIDENTIAL> (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string  NONC:<CONFIDENTIAL>
found device with ECID <CONFIDENTIAL>
Setting to configuration 1
Setting to interface 0:0
Identified device as n42ap, iPhone5,2
Extracting BuildManifest from IPSW
Product Version: 10.3.4
Product Build: 14G61 Major: 14
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string  NONC:<CONFIDENTIAL>
found device with ECID <CONFIDENTIAL>
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is <CONFIDENTIAL>
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> yes
Invalid input. Please type YES or hit CTRL+C to abort.
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'iPhone_4.0_32bit_10.3.4_14G61_Restore/058-74968-065.dmg'
Getting device's ECID for TSS request
Found ECID <CONFIDENTIAL>
Getting ApNonce in recovery mode... 05 fe 40 57 53 16 6f 12 55 59 e7 c9 ac 55 86 54 f1 07 c7 e9 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbFDRSecurityKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Applying restore request rules for entry DeviceTree
DEBUG: Applying restore request rules for entry KernelCache
DEBUG: Applying restore request rules for entry LLB
DEBUG: Applying restore request rules for entry OS
DEBUG: Applying restore request rules for entry RecoveryMode
DEBUG: Applying restore request rules for entry RestoreDeviceTree
DEBUG: Applying restore request rules for entry RestoreKernelCache
DEBUG: Applying restore request rules for entry RestoreLogo
DEBUG: Applying restore request rules for entry RestoreRamDisk
DEBUG: Applying restore request rules for entry iBEC
DEBUG: Applying restore request rules for entry iBSS
DEBUG: Applying restore request rules for entry iBoot
common.c:printing 3872 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
[... 467 lines cut ...]
Connected to com.apple.mobile.restored, version 14
Device <CONFIDENTIAL> has successfully entered restore mode
Hardware Information:
BoardID: 2
ChipID: <CONFIDENTIAL>
UniqueChipID: <CONFIDENTIAL>
ProductionMode: true
Previous restore exit status: 0x100
Starting FDR listener thread
Connecting to FDR client at port 1082
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
Retrying connection...
ERROR: Unable to connect to FDR client (-2)
ERROR: Failed to start FDR Ctrl channel
Failed to parse checkpoint result from checkpoint plist
ERROR: Unable to restore device

```

labradorcode, 2021-09-30:
make idevicerestore-dirclean libimobiledevice-dirclean libirecovery-dirclean libusbmuxd-dirclean usbmuxd-dirclean libplist-dirclean

Hi, 
I have the same problem,
ProductName: iPhone OS
ProductType: iPhone5,2
ProductVersion: 10.3.4

I have built idevicerestore and other libs with recent commits,

- idevicerestore built with the commit 65cc89f80d3f9b8417038c02258f830a6b09615e 
- libimobiledevice built with the commit b3d35fbcf7a1ac669c2e80fbd58920941a5d4c0c 
- libimobiledevice-glue built with the commit 499a5578b15235d00bc492068635de45bec1807d 
- libirecovery built with the commit 0165076dedcad80c9a6886424cdc0992a88ce2ae 
- libusbmuxd built with the commit 2ec5354a6ff2ba5e2740eabe7402186f29294f79 
- libusb version 1.0.23
- libplist build with the commit cf7a3f3d7c06b197ee71c9f97eb9aa05f26d63b5 

**idevicerestore output log**
[iphone5.2.txt](https://github.com/libimobiledevice/idevicerestore/files/7258348/iphone5.2.txt)

**ideviceinfo output log**
[ideviceinfo.txt](https://github.com/libimobiledevice/idevicerestore/files/7258431/ideviceinfo.txt)

**usbmuxd output log**
[usbmuxd.txt](https://github.com/libimobiledevice/idevicerestore/files/7258412/usbmuxd.txt)

nikias, 2021-09-30:
Can you try with this patch:
```diff
diff --git a/src/restore.c b/src/restore.c
index b1089bf..47c93dd 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -3764,7 +3764,7 @@ int restore_device(struct idevicerestore_client_t* client, plist_t build_identit
                else if (!strcmp(type, "CheckpointMsg")) {
                        uint64_t ckpt_id;
                        uint64_t ckpt_res;
-                       uint8_t ckpt_complete;
+                       uint8_t ckpt_complete = 0;
                        // Get checkpoint id
                        node = plist_dict_get_item(message, "CHECKPOINT_ID");
                        if (!node || plist_get_node_type(node) != PLIST_UINT) {
@@ -3781,11 +3781,9 @@ int restore_device(struct idevicerestore_client_t* client, plist_t build_identit
                        plist_get_uint_val(node, &ckpt_res);
                        // Get checkpoint complete
                        node = plist_dict_get_item(message, "CHECKPOINT_COMPLETE");
-                       if (!node || plist_get_node_type(node) != PLIST_BOOLEAN) {
-                               debug("Failed to parse checkpoint result from checkpoint plist\n");
-                               return -1;
+                       if (PLIST_IS_BOOLEAN(node)) {
+                               plist_get_bool_val(node, &ckpt_complete);
                        }
-                       plist_get_bool_val(node, &ckpt_complete);
                        if (ckpt_complete)
                                info("Checkpoint %" PRIu64 " complete with code %" PRIu64 "\n", ckpt_id, ckpt_res);
                }
```
I think if `CHECKPOINT_COMPLETE` is not present we can assume it is considered `false`. Let me know if this works.

labradorcode, 2021-10-01:
> Can you try with this patch:
>
> ```diff
> diff --git a/src/restore.c b/src/restore.c
> index b1089bf..47c93dd 100644
> --- a/src/restore.c
> +++ b/src/restore.c
> @@ -3764,7 +3764,7 @@ int restore_device(struct [...]

Perfect, it works. Tried with the commit 65cc89f80d3f9b8417038c02258f830a6b09615e 

cheers mate

```

Got status message
Status: Restore Finished
Cleaning up...
DONE
progress: 6 1.000000

```

nikias, 2021-10-01:
Pushed the change with commit ef5f24be48f42e94064a2e4acec41a2fe95519fc.

closed 2021-10-03 by nikias