libimobiledevice/idevicerestore #441: AS/M1: Support Mac OS Monterey (12.0)

marcan, 2021-09-24:
Restoring the latest 12.0 beta7 (UniversalMac_12.0_21A5522h_Restore.ipsw) fails with the following:

```[16:27:44.0226-GMT]{3>7} CHECKPOINT BEGIN: RESTORED:[0x0655] is_host_compatible
restore-step-ids = {0x11030655:12}
restore-step-names = {0x11030655:is_host_compatible}
restore-step-uptime = 6
restore-step-user-progress = 0
host/device compatibility check found that a message is missing or is not supported (optional): FirmwareUpdaterPreflight

host/device compatibility check found that a message is missing or is not supported (required): BootabilityBundle

host/device compatibility check found that a message is missing or is not supported (optional): RestoreLocalPolicy

host/device compatibility check found that a message is missing or is not supported (optional): AuthInstallCACert

Detected a mismatch between the messages the device can send and what the host can support.
Checkpoint engine recorder path set to /mnt5
[16:27:44.0228-GMT]{3>7} CHECKPOINT FAILURE:(FAILURE:10) RESTORED:[0x0655] is_host_compatible [0]D(This host version is unsupported. You may need to update your host tools.)
```

marcan, 2021-09-24:
FWIW, BootabilityBundle is a new Preboot component that is used as a shared library so that newer OSes can get correct BootPolicy and bootability configuration even when the OS doing so (via the boot picker) is older. Presumably idevicerestore needs support for sending this (it's just a directory in the ipsw).

marcan, 2021-09-26:
I did a bit of digging into this. Here's a preliminary patch with the basic scaffolding: https://github.com/marcan/idevicerestore/commit/6abbd0afcd8b5e2533c6763e53c5c01773dc9c9a

That changes the BuildManifest identity picker logic to use the full variant name (since the 12.0 IPSWs don't have the GlobalSigning flag to identify the RecoveryOS variant), and adds a basic stub for BootabilityBundle.

The actual BootabilityBundle request looks like this:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>MsgType</key>
        <string>DataRequestMsg</string>
        <key>DataType</key>
        <string>BootabilityBundle</string>
        <key>DataPort</key>
        <integer>49170</integer>
</dict>
</plist>
```

It expects you to connect to the DataPort and dump in a cpio archive of the BootabilityBundle data, constructed like this:

* `<ipsw>/BootabilityBundle/Restore/Bootability/* -> ./`
* `<ipsw>/BootabilityBundle/Restore/Firmware/Bootability.dmg.trustcache -> ./Bootability.trustcache`

The cpio should be in the portable ASCII format as produced by `cpio -c`, e.g. `find . | cpio -R 0:0 -c -o > ../BootabilityBundle.cpio`.

Building a cpio like that and dumping it into the DataPort via iproxy/netcat allows the restore to continue (nc needs to be killed at the end, to close the connection and get the remote end to continue). I'm guessing you'd want to handle this by implementing cpio internally? (it's a pretty simple format).

With this process, the restore completes and macOS Monterey 12.0 beta7 boots correctly.

nikias, 2021-09-30:
Yes cpio handling internally makes most sense. Thanks for sharing your findings.

commit b34d4618 (2021-11-19): restore: Support sending BootabilityBundle (Monterey)

This is used on macOS 12+ (Monterey and later) to allow older OSes to
perform the bless2/bootability process for newer OSes, by putting a
shared library in the Preboot partition that contains the specific logic
required to make a given macOS install bootable.

Closes: #441
closed 2021-11-19 by nikias