libimobiledevice/idevicerestore #442: AS/M1: Support the iMac M1

marcan, 2021-09-24:
Trying to restore 11.4 (UniversalMac_11.4_20F71_Restore.ipsw) fails with:

```
Checkpoint 4873 complete with code 0
Updating Veridian (66)
Checkpoint 4881 complete with code 0
Unknown operation (81)
ERROR: restore_send_firmware_updater_data: Got unknown updater name 'AppleTCON'.
```

TCON is the timing controller for the display AIUI, so it's unsurprising that this would be unique to the iMac or to Macs with a built-in screen in general.

nikias, 2021-09-24:
Yes I found the AppleTCON already but since I don't have the hardware I cannot test it. I started to write the code for it but then realized I need to capture the TSS request for it. This is what I have so far (but it won't work obviously):

https://gist.github.com/nikias/aadb2aed43b80710618e3fdef9b72601

Maybe you can help with capturing the TSS request?

Also interestingly you didn't have issues with recovery mode which is failing on M1 MacBook (Pro).

marcan, 2021-09-25:
Sure, how do you usually do that? Is it just plaintext HTTP?

The other issue I had was a missing USB PID in usbmuxd, but you saw that PR already.

nikias, 2021-09-25:
It uses https but basically you can do
```
defaults write com.apple.AMPDevicesAgent AuthInstallSigningServerURL http://gs.apple.com:80
```
And then capture network traffic with wireshark, while restoring with Apple Configurator.

marcan, 2021-09-26:
Request:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>@BBTicket</key>
	<true/>
	<key>@Baobab,Ticket</key>
	<true/>
	<key>@HostPlatformInfo</key>
	<string>mac</string>
	<key>@UUID</key>
	<string>F2106519-0A83-4541-869F-E8EA1DE24E85</string>
	<key>@VersionInfo</key>
	<string>libauthinstall-776.140.3</string>
	<key>Baobab,BoardID</key>
	<integer>3</integer>
	<key>Baobab,ChipID</key>
	<integer>2133</integer>
	<key>Baobab,ECID</key>
	<data>
	AAAAAAAAAAADMzUBFQQVOg==
	</data>
	<key>Baobab,Life</key>
	<integer>15</integer>
	<key>Baobab,ManifestEpoch</key>
	<integer>0</integer>
	<key>Baobab,ProductionMode</key>
	<true/>
	<key>Baobab,SecurityDomain</key>
	<integer>1</integer>
	<key>Baobab,TCON</key>
	<dict>
		<key>Digest</key>
		<data>
		olkY54c5YzwGBzv4JFUufx+0gPbqXbT1f1cU3y7CRyk=
		</data>
		<key>EPRO</key>
		<true/>
	</dict>
	<key>Baobab,UpdateNonce</key>
	<data>
	DkTeak3ADmOdDHa5kGan5Fk8MbjIpYmDI1jDPLC+q/8=
	</data>
</dict>
</plist>
```
Response:
```
STATUS=0&MESSAGE=SUCCESS&REQUEST_STRING=<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<plist version="1.0">
<dict>
	<key>@ServerVersion</key>
	<string>2.1.0</string>
	<key>Baobab,Ticket</key>
	<data>
	MIIHLhYESU00TQIBADGCAXb/hOqFnEKCAW0wggFpFgRNQU5CMYIBX/+E6oWc
<snip>
	</data>
</dict>
</plist>
```

commit 1ea3fe75 (2021-09-26): restore: Add support for updating AppleTCON

Closes: #442
closed 2021-09-26 by nikias