libimobiledevice/idevicerestore #450: Separate `idevicerestore` the binary from `libirestore` the framework

rickmark, 2021-10-06:
There are a number of both high-level and low-level APIs in this repository.  In order to better facilitate modularity and code reuse, I suggest we factor out the codebase like so:

* `idevicerestore` is a binary dependent on `libirestore`, which would expose a new high-level API (prototyped as `sequencer.c` but could actually be a CPP state-machine with C API interfaces - as state machines in C are a nightmare, and `libplist` utilizes CPP internally in places for similar reasons).
** This high-level API allows one to request a restore sequence on an iDevice, and should be a C API providing a series of options similar to what can be set on the program as flags today.  This high-level API also provides C-style callbacks that can be set to expose progress.  The `idevicerestore` program would format these callbacks into text mode choices and progress. 
*** EXAMPLE: selecting IPSW when ipsw.me has more than one choice, progress indicators for transfer, and info about stages)
* We create Cython bindings for `libirestore` as a separate PyPI package, allowing multiple versions of Python to install compile-time bindings for the library.
** BONUS: GNOME shell could then implement a PyGTK application to provide UI based device restore/recovery similar to what Finder provides on macOS
** We better layer the low-level APIs into an acyclic-directed graph.  For example idevicerestore -> IPSW -> IMG4. IPSW -> TSS -> CURL, and otherwise so each low-level API C file is compiled without dependence on any of its consumers.  This would institute thoughtful, layered design to each of the 
* We move `dfu.c` to a `libirecovery-2.0` as it is a low level USB protocol meaning all USB code is handled by the union of `libirescue` and `libusbmuxd` removing the USB code from `libirestore` entirely
** We merge in code from `macvdmtool` to allow for USB-PD VDMs in `libirescue` to allow for pushing a connected device to DFU without user intervention (this is for USB-C devices like the T2, M1, and iPad Pro)
** In `libirescue` we create `device.c(pp)` that would represent an abstract device in any state for operations like `get_state`,`get_ap_nonce` etc.

XanaduAPI, 2021-10-13:
Looks like a good proposal!

rickmark, 2021-10-13:
I'd like @nikias or @marcan to weigh in a little here too...

I think the most controversial part of the proposal is to move the `dfu.c` code to `libirestore` (it would necessitate a version bump and a conditional in `configure.ac`) but I do think moving all of the USB code out of `libirecovery` is a really nice design touch.  In this case it would be possible to drop the `libusb` code and dependency from `libirestore` as it would be transitive via `libusbmuxd` and `libirecovery`.  For similar reasons it might also be wise to move Plist services from `libimobiledevice` to `libusbmuxd` again, making it not a dependency for `libirecovery`, and I cannot think of any reason to depend on `libusbmuxd` where you wouldn't want PList services, but there are plenty of places you may want it without the entire `libimobiledevice` import.

I'm only really conflicted about `lockdownd` in this case.  Perhaps because a restore device runs a different mode of lockdown it makes sense to have the two diverge (`com.apple.restored` does not make use of pairing, encryption etc)