libimobiledevice/idevicerestore #462: iPhone6s ipsw Verification fails IOS15.2

hiromi-hhh, 2021-12-15:
It outputs the following error and exits.

```
~
Selected firmware 15.2 (build 19C56)
Verifying 'iPhone_4.7_15.2_19C56_Restore.ipsw'...
Checksum does not match.
Downloading firmware (https://updates.cdn-apple.com/2021FCSWinter/fullrestores/071-92007/FED491BD-0E89-4696-86AA-7F6911F50133/iPhone_4.7_15.2_19C56_Restore.ipsw)
downloading: 1%
downloading: 2%
downloading: 3%
downloading: 4%
downloading: 5%
downloading: 6%
downloading: 7%
~
downloading: 95%
downloading: 96%
downloading: 97%
downloading: 98%
downloading: 99%

Verifying 'iPhone_4.7_15.2_19C56_Restore.ipsw'...
ERROR: File download failed (checksum mismatch).
```

Used on MacOS (Bigsur 11.4). The library is the latest one.

Packjackisback, 2022-08-30:
Same problem on ipad prro

Jackalhz1980, 2023-07-17:
Ive experienced the same. Ive included a little more detail.  Hopefully it helps.

(root㉿kali)-[/opt/libimobiledevice-1.2.1]
└─# idevicerestore -RPnldy ./iPhone10,3,iPhone10,6_16.5.1_20F75_Restore.ipsw.lock
progress: 0 0.000000
irecv_event_cb: device 001a40591087c02e (udid: N/A) connected in DFU mode
progress: 0 0.100000
Found device in DFU mode
opening device 05ac:1227...
found device with ECID 001a40591087c02e
Setting to configuration 1
Setting to interface 0:0
progress: 0 0.200000
Identified device as d22ap, iPhone10,3
*   Trying 104.26.9.156:443...
* Connected to api.ipsw.me (104.26.9.156) port 443 (#0)
* found 423 certificates in /etc/ssl/certs
* GnuTLS ciphers: NORMAL:-ARCFOUR-128:-CTYPE-ALL:+CTYPE-X509:-VERS-SSL3.0
* ALPN: offers h2,http/1.1
* SSL connection using TLS1.3 / ECDHE_RSA_AES_256_GCM_SHA384
*   server certificate verification SKIPPED
*   server certificate status verification SKIPPED
*   common name: sni.cloudflaressl.com (matched)
*   server certificate expiration date OK
*   server certificate activation date OK
*   certificate public key: EC/ECDSA
*   certificate version: #3
*   subject: C=US,ST=California,L=San Francisco,O=Cloudflare\, Inc.,CN=sni.cloudflaressl.com
*   start date: Sun, 05 Feb 2023 00:00:00 GMT
*   expire date: Sun, 04 Feb 2024 23:59:59 GMT
*   issuer: C=US,O=Cloudflare\, Inc.,CN=Cloudflare Inc ECC CA-3
* ALPN: server accepted h2
* using HTTP/2
* h2h3 [:method: GET]
* h2h3 [:path: /v3/device/iPhone10,3]
* h2h3 [:scheme: https]
* h2h3 [:authority: api.ipsw.me]
* h2h3 [user-agent: InetURL/1.0 idevicerestore/1.0.0]
* h2h3 [accept: */*]
* Using Stream ID: 1 (easy handle 0xd509d0)
> GET /v3/device/iPhone10,3 HTTP/2
Host: api.ipsw.me
user-agent: InetURL/1.0 idevicerestore/1.0.0
accept: */*

< HTTP/2 200 
< date: Mon, 17 Jul 2023 03:10:05 GMT
< content-type: application/json
< vary: Accept-Encoding
< vary: Origin
< access-control-allow-origin: *
< access-control-expose-headers: X-Data-Location, X-Server-Location
< cache-control: public, max-age=14400
< content-md5: 3H+ffEczLm3pqhazcSqv6A==
< content-security-policy: default-src https: 'unsafe-eval' 'unsafe-inline'; object-src 'none'
< expires: Mon, 17 Jul 2023 04:08:28 GMT
< x-content-type-options: nosniff
< x-data-location: database
< x-frame-options: DENY
< x-frame-options: DENY
< x-server-location: f4a0475957db
< x-xss-protection: 1; mode=block
< strict-transport-security: max-age=15552000; preload
< last-modified: Mon, 17 Jul 2023 03:08:28 GMT
< cf-cache-status: HIT
< report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=A5Bzv7HsDD0cDKjG2YunjoDl%2FSCWRX0U%2BgXc7gYXDr5x7f9gcgILy%2FujAKm7z4EKdJVj0U3iL%2B%2B6KGg0Q7xJdUw9BugLvRSe8gKa%2B2C18%2BFo%2BRuD18%2F5HYHjEqsy"}],"group":"cf-nel","max_age":604800}
< nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
< server: cloudflare
< cf-ray: 7e7f4a350df645f0-DFW
< alt-svc: h3=":443"; ma=86400
[... 67 lines cut ...]
downloading: 17%
downloading: 18%
downloading: 19%
downloading: 20%
downloading: 21%
downloading: 22%
downloading: 23%
downloading: 24%
downloading: 25%
downloading: 26%
downloading: 27%
downloading: 28%
downloading: 29%
downloading: 30%
downloading: 31%
downloading: 32%
downloading: 33%
downloading: 34%
* Failure writing output to destination
* Closing connection 0

Verifying 'iPhone10,3,iPhone10,6_16.5.1_20F75_Restore.ipsw'...
ERROR: File download failed (checksum mismatch).
                                                                                                                                                                                              
┌──(root㉿kali)-[/opt/libimobiledevice-1.2.1]