libimobiledevice/idevicerestore #463: 15.3 beta flashing fails: 'ERROR: Unable to find any build identities'

skortela, 2021-12-21:
iPhone 8 Plus flashing fails. Same problems seen on other iPhones also with iOS 15.3 beta image.

Problem cannot be reproduced on older (few months older) idevicerestore versions.
I have feeling that this might be regression from this:
https://github.com/libimobiledevice/idevicerestore/commit/ef2db7acb3b177aab665267934e3d64c95cfe758#diff-25a6634263c1b1f6fc4697a04e2b9904ea4b042a89af59dc93ec1f5d44848a26
```
idevicerestore.exe -e iPhone_5.5_P3_15.3_19D5026g_Restore.ipsw
[12:11:50.663][14968] : Found device in Normal mode
[12:11:50.794][14968] : Identified device as d211ap, iPhone10,5
[12:11:50.797][14968] : Extracting BuildManifest from IPSW
[12:11:50.820][14968] : Product Version: 15.3
[12:11:50.822][14968] : Product Build: 19D5026g Major: 19
[12:11:50.827][14968] : Device supports Image4: true
[12:11:50.829][14968] : ERROR: Unable to find any build identities
```

marcan, 2021-12-21:
Ha, that's because the betas use different variant names...

```
                                <key>Variant</key>
                                <string>Developer Upgrade Install (IPSW)</string>
```

Should still work for release versions though. I guess we need to try both strings?

nikias, 2022-04-14:
Fixed with 2eb8865b0fcbe9908a64e9b8b396a83ab0c28c2d.

closed 2022-04-14 by nikias