libimobiledevice/idevicerestore #473: iPad Air 5th gen, idevicerestore fails, Got unknown updater name 'AppleTypeCRetimer'.

skortela, 2022-03-28:
Erasing of iPad Air 5th gen fails.

```
[08:57:38.872][22828] : Updating Veridian (66)
[08:57:38.873][22828] : Checkpoint 4294972177 complete with code 0
[08:57:38.873][22828] : Unknown operation (79)
[08:57:38.875][22828] : Checkpoint 4886 complete with code 0
[08:57:38.876][22828] : Unknown operation (82)
[08:57:38.885][22828] : ERROR: restore_send_firmware_updater_data: Got unknown updater name 'AppleTypeCRetimer'.
[08:57:38.885][22828] : ERROR: Unable to send FirmwareUpdater data
[08:57:38.885][22828] : ERROR: Unable to successfully restore device
```
[idevicerestore.log](https://github.com/libimobiledevice/idevicerestore/files/8360293/idevicerestore.log)

nikias, 2022-04-04:
Can you apply this patch and see if it works?
```diff
diff --git a/src/restore.c b/src/restore.c
index ef907ba..958d863 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -2683,6 +2683,147 @@ static plist_t restore_get_tcon_firmware_data(restored_client_t restore, struct
 	return response;
 }
 
+static plist_t restore_get_timer1_firmware_data(restored_client_t restore, struct idevicerestore_client_t* client, plist_t build_identity, plist_t p_info)
+{
+	char *comp_name = NULL;
+	char *comp_path = NULL;
+	plist_t comp_node = NULL;
+	unsigned char* component_data = NULL;
+	unsigned int component_size = 0;
+	ftab_t ftab = NULL;
+	ftab_t rftab = NULL;
+	uint32_t ftag = 0;
+	plist_t parameters = NULL;
+	plist_t request = NULL;
+	plist_t response = NULL;
+	plist_t node = NULL;
+	int ret;
+
+	/* create Timer1 request */
+	request = tss_request_new(NULL);
+	if (request == NULL) {
+		error("ERROR: Unable to create Timer TSS request\n");
+		free(component_data);
+		return NULL;
+	}
+
+	parameters = plist_new_dict();
+
+	/* add manifest for current build_identity to parameters */
+	tss_parameters_add_from_manifest(parameters, build_identity);
+
+	plist_dict_set_item(parameters, "ApProductionMode", plist_new_bool(1));
+	if (client->image4supported) {
+		plist_dict_set_item(parameters, "ApSecurityMode", plist_new_bool(1));
+		plist_dict_set_item(parameters, "ApSupportsImg4", plist_new_bool(1));
+	} else {
+		plist_dict_set_item(parameters, "ApSupportsImg4", plist_new_bool(0));
+	}
+
+	/* add Timer,* tags from info dictionary to parameters */
+	plist_dict_merge(&parameters, p_info);
+
+	/* add required tags for Timer1 TSS request */
+	tss_request_add_timer1_tags(request, parameters, NULL);
+
+	plist_free(parameters);
+
+	info("Sending Timer1 TSS request...\n");
+	response = tss_request_send(request, client->tss_url);
+	plist_free(request);
+	if (response == NULL) {
+		error("ERROR: Unable to fetch Timer1 ticket\n");
+		free(component_data);
+		return NULL;
+	}
+
+	if (plist_dict_get_item(response, "Timer1,Ticket")) {
+		info("Received Timer1 ticket\n");
+	} else {
+		error("ERROR: No 'Timer1,Ticket' in TSS response, this might not work\n");
+	}
+
[... 180 lines cut ...]
+
+	/* apply overrides */
+	if (overrides) {
+		plist_dict_merge(&request, overrides);
+	}
+
+	return 0;
+}
+
 static size_t tss_write_callback(char* data, size_t size, size_t nmemb, tss_response* response) {
 	size_t total = size * nmemb;
 	if (total != 0) {
diff --git a/src/tss.h b/src/tss.h
index 3590aed..8e485bf 100644
--- a/src/tss.h
+++ b/src/tss.h
@@ -48,6 +48,7 @@ int tss_request_add_vinyl_tags(plist_t request, plist_t parameters, plist_t over
 int tss_request_add_rose_tags(plist_t request, plist_t parameters, plist_t overrides);
 int tss_request_add_veridian_tags(plist_t request, plist_t parameters, plist_t overrides);
 int tss_request_add_tcon_tags(plist_t request, plist_t parameters, plist_t overrides);
+int tss_request_add_timer1_tags(plist_t request, plist_t parameters, plist_t overrides);
 
 int tss_request_add_ap_img4_tags(plist_t request, plist_t parameters);
 int tss_request_add_ap_img3_tags(plist_t request, plist_t parameters);
```

pekn, 2022-04-05:
I tested above patch with iPad Air 5th Gen.. It still fails, here is last lines from log:

...
Sending IsFUDFirmware for iBootData...
Checkpoint 4874 complete with code 0
Updating gas gauge software (47)
Checkpoint 4294972161 complete with code 0
Updating gas gauge softwTSS server returned: STATUS=94&MESSAGE=This device isn't eligible for the requested build.
ERROR: TSS request failed (status=94, message=This device isn't eligible for the requested build.)
ERROR: Unable to fetch Timer1 ticket
ERROR: restore_send_firmware_updater_data: Couldn't get AppleTypeCRetimer firmware data
ERROR: Unable to send FirmwareUpdater data
ERROR: Unable to successfully restore device

nikias, 2022-04-05:
Oh I think I forgot something here, can you try after applying this patch as well?
```diff
diff --git a/src/tss.c b/src/tss.c
index 084ad10..fe16ec9 100644
--- a/src/tss.c
+++ b/src/tss.c
@@ -490,6 +490,48 @@ int tss_parameters_add_from_manifest(plist_t parameters, plist_t build_identity)
        }
        node = NULL;

+       /* add Timer,BoardID,1 */
+       node = plist_dict_get_item(build_identity, "Timer,BoardID,1");
+       if (node) {
+               plist_dict_set_item(parameters, "Timer,BoardID,1", plist_copy(node));
+       }
+       node = NULL;
+
+       /* add Timer,BoardID,2 */
+       node = plist_dict_get_item(build_identity, "Timer,BoardID,2");
+       if (node) {
+               plist_dict_set_item(parameters, "Timer,BoardID,2", plist_copy(node));
+       }
+       node = NULL;
+
+       /* add Timer,ChipID,1 */
+       node = plist_dict_get_item(build_identity, "Timer,ChipID,1");
+       if (node) {
+               plist_dict_set_item(parameters, "Timer,ChipID,1", plist_copy(node));
+       }
+       node = NULL;
+
+       /* add Timer,ChipID,2 */
+       node = plist_dict_get_item(build_identity, "Timer,ChipID,2");
+       if (node) {
+               plist_dict_set_item(parameters, "Timer,ChipID,2", plist_copy(node));
+       }
+       node = NULL;
+
+       /* add Timer,SecurityDomain,1 */
+       node = plist_dict_get_item(build_identity, "Timer,SecurityDomain,1");
+       if (node) {
+               plist_dict_set_item(parameters, "Timer,SecurityDomain,1", plist_copy(node));
+       }
+       node = NULL;
+
+       /* add Timer,SecurityDomain,2 */
+       node = plist_dict_get_item(build_identity, "Timer,SecurityDomain,2");
+       if (node) {
+               plist_dict_set_item(parameters, "Timer,SecurityDomain,2", plist_copy(node));
+       }
+       node = NULL;
+
        /* add build identity manifest dictionary */
        node = plist_dict_get_item(build_identity, "Manifest");
        if (!node || plist_get_node_type(node) != PLIST_DICT) {
```

pekn, 2022-04-05:
now it fails with different message:

...
Sending Timer1 TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168

nikias, 2022-04-05:
Yes I figured it out. Wait a little for me to make a new patch!

nikias, 2022-04-06:
New patch here. Undo the previous patches first.
```diff
diff --git a/src/restore.c b/src/restore.c
index ef907ba..0909a7c 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -2683,6 +2683,196 @@ static plist_t restore_get_tcon_firmware_data(restored_client_t restore, struct
 	return response;
 }
 
+static plist_t restore_get_timer_firmware_data(restored_client_t restore, struct idevicerestore_client_t* client, plist_t build_identity, plist_t p_info)
+{
+	char comp_name[64];
+	char *comp_path = NULL;
+	plist_t comp_node = NULL;
+	unsigned char* component_data = NULL;
+	unsigned int component_size = 0;
+	ftab_t ftab = NULL;
+	ftab_t rftab = NULL;
+	uint32_t ftag = 0;
+	plist_t parameters = NULL;
+	plist_t request = NULL;
+	plist_t response = NULL;
+	plist_t node = NULL;
+	const char* ticket_name = NULL;
+	uint32_t tag = 0;
+	int ret;
+
+	/* create Timer request */
+	request = tss_request_new(NULL);
+	if (request == NULL) {
+		error("ERROR: Unable to create Timer TSS request\n");
+		return NULL;
+	}
+
+	parameters = plist_new_dict();
+
+	/* add manifest for current build_identity to parameters */
+	tss_parameters_add_from_manifest(parameters, build_identity);
+
+	/* add Timer,* tags from info dictionary to parameters */
+	plist_t info_array = plist_dict_get_item(p_info, "InfoArray");
+	if (!info_array) {
+		error("ERROR: Could not find InfoArray in info dictionary\n");
+		plist_free(parameters);
+		return NULL;
+	} else {
+		plist_t info_dict = plist_array_get_item(info_array, 0);
+		plist_t hwid = plist_dict_get_item(info_dict, "HardwareID");
+		uint64_t u64val;
+		uint8_t bval;
+		tag = (uint32_t)_plist_dict_get_uint(info_dict, "TagNumber");
+		char key[64];
+
+		plist_dict_set_item(parameters, "TagNumber", plist_new_uint(tag));
+		plist_t node = plist_dict_get_item(info_dict, "TicketName");
+		if (node) {
+			ticket_name = plist_get_string_ptr(node, NULL);
+			plist_dict_set_item(parameters, "TicketName", plist_copy(node));
+		}
+
+		sprintf(key, "Timer,ChipID,%u", tag);
+		u64val = _plist_dict_get_uint(hwid, "ChipID");
+		plist_dict_set_item(parameters, key, plist_new_uint(u64val));
+
+		sprintf(key, "Timer,BoardID,%u", tag);
+		u64val = _plist_dict_get_uint(hwid, "BoardID");
+		plist_dict_set_item(parameters, key, plist_new_uint(u64val));
+
+		sprintf(key, "Timer,ECID,%u", tag);
[... 300 lines cut ...]
+
+	/* apply overrides */
+	if (overrides) {
+		plist_dict_merge(&request, overrides);
+	}
+
+	return 0;
+}
+
 static size_t tss_write_callback(char* data, size_t size, size_t nmemb, tss_response* response) {
 	size_t total = size * nmemb;
 	if (total != 0) {
diff --git a/src/tss.h b/src/tss.h
index 3590aed..8464f11 100644
--- a/src/tss.h
+++ b/src/tss.h
@@ -48,6 +48,7 @@ int tss_request_add_vinyl_tags(plist_t request, plist_t parameters, plist_t over
 int tss_request_add_rose_tags(plist_t request, plist_t parameters, plist_t overrides);
 int tss_request_add_veridian_tags(plist_t request, plist_t parameters, plist_t overrides);
 int tss_request_add_tcon_tags(plist_t request, plist_t parameters, plist_t overrides);
+int tss_request_add_timer_tags(plist_t request, plist_t parameters, plist_t overrides);
 
 int tss_request_add_ap_img4_tags(plist_t request, plist_t parameters);
 int tss_request_add_ap_img3_tags(plist_t request, plist_t parameters);
```

pekn, 2022-04-06:
Tested above patch. Still same error:

...
Sending Timer1,Ticket TSS request...
Request URL set to https://gs.apple.com/TSS/controller?action=2
TSS server returned: STATUS=168&MESSAGE=An internal error occurred.
ERROR: tss_send_request: Unhandled status code 168

nikias, 2022-04-06:
Did you run with debug output? I want to see the actual request that gets sent.

pekn, 2022-04-06:
Tried to run with debug output but output wasn't very good (Timer TSS request was not shown completely)..
Anyway, added code to dump Timer TSS request to a file.
Here's its contents:

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>@BBTicket</key>
	<true/>
	<key>@HostPlatformInfo</key>
	<string>mac</string>
	<key>@VersionInfo</key>
	<string>libauthinstall-850.0.2</string>
	<key>@UUID</key>
	<string>B60CDDCE-C693-66A2-23B7-ADF922273D85</string>
	<key>@Timer1,Ticket</key>
	<true/>
	<key>Timer,BoardID,1</key>
	<integer>1</integer>
	<key>Timer,ChipID,1</key>
	<integer>26112</integer>
	<key>Timer,SecurityDomain,1</key>
	<integer>1</integer>
	<key>Timer,SecurityMode,1</key>
	<true/>
	<key>Timer,ProductionMode,1</key>
	<true/>
	<key>Timer,ECID,1</key>
	<integer>75916162930065438</integer>
	<key>Timer,Nonce,1</key>
	<data>
	fC2wnTEOzgs21QyG5MPmZBaElIzW/AMmKw0O2Rps+8M=
	</data>
	<key>Timer,RTKitOS,1</key>
	<dict>
		<key>Digest</key>
		<data>
		ZiJ9JxG+O61dS1DE9YJtPtxnLbK8GrmF4ilVT3F1vgLWrzdM3q+tp3AvHgTi
		qICo
		</data>
		<key>Trusted</key>
		<true/>
	</dict>
	<key>Timer,RestoreRTKitOS,1</key>
	<dict>
		<key>Digest</key>
		<data>
		ZiJ9JxG+O61dS1DE9YJtPtxnLbK8GrmF4ilVT3F1vgLWrzdM3q+tp3AvHgTi
		qICo
		</data>
		<key>Trusted</key>
		<true/>
	</dict>
</dict>
</plist>

pekn, 2022-04-06:
Here is output of 'idevicerestore -e -l &> output.txt'
(Doesn't look quite OK though, not sure why)
[https://www.dropbox.com/s/04sdudr5l9lfiah/output.txt?dl=0](url)

nikias, 2022-04-06:
Ah OK I see what is missing, need to check the code again...

nikias, 2022-04-06:
Try this updated patch after undoing the previous one:
```diff
diff --git a/src/restore.c b/src/restore.c
index ef907ba..48eca78 100644
--- a/src/restore.c
+++ b/src/restore.c
@@ -2683,6 +2683,204 @@ static plist_t restore_get_tcon_firmware_data(restored_client_t restore, struct
 	return response;
 }
 
+static plist_t restore_get_timer_firmware_data(restored_client_t restore, struct idevicerestore_client_t* client, plist_t build_identity, plist_t p_info)
+{
+	char comp_name[64];
+	char *comp_path = NULL;
+	plist_t comp_node = NULL;
+	unsigned char* component_data = NULL;
+	unsigned int component_size = 0;
+	ftab_t ftab = NULL;
+	ftab_t rftab = NULL;
+	uint32_t ftag = 0;
+	plist_t parameters = NULL;
+	plist_t request = NULL;
+	plist_t response = NULL;
+	plist_t node = NULL;
+	const char* ticket_name = NULL;
+	uint32_t tag = 0;
+	int ret;
+
+	/* create Timer request */
+	request = tss_request_new(NULL);
+	if (request == NULL) {
+		error("ERROR: Unable to create Timer TSS request\n");
+		return NULL;
+	}
+
+	parameters = plist_new_dict();
+
+	/* add manifest for current build_identity to parameters */
+	tss_parameters_add_from_manifest(parameters, build_identity);
+
+	plist_dict_set_item(parameters, "ApProductionMode", plist_new_bool(1));
+	if (client->image4supported) {
+		plist_dict_set_item(parameters, "ApSecurityMode", plist_new_bool(1));
+		plist_dict_set_item(parameters, "ApSupportsImg4", plist_new_bool(1));
+	} else {
+		plist_dict_set_item(parameters, "ApSupportsImg4", plist_new_bool(0));
+	}
+
+	/* add Timer,* tags from info dictionary to parameters */
+	plist_t info_array = plist_dict_get_item(p_info, "InfoArray");
+	if (!info_array) {
+		error("ERROR: Could not find InfoArray in info dictionary\n");
+		plist_free(parameters);
+		return NULL;
+	} else {
+		plist_t info_dict = plist_array_get_item(info_array, 0);
+		plist_t hwid = plist_dict_get_item(info_dict, "HardwareID");
+		uint64_t u64val;
+		uint8_t bval;
+		tag = (uint32_t)_plist_dict_get_uint(info_dict, "TagNumber");
+		char key[64];
+
+		plist_dict_set_item(parameters, "TagNumber", plist_new_uint(tag));
+		plist_t node = plist_dict_get_item(info_dict, "TicketName");
+		if (node) {
+			ticket_name = plist_get_string_ptr(node, NULL);
+			plist_dict_set_item(parameters, "TicketName", plist_copy(node));
+		}
+
+		sprintf(key, "Timer,ChipID,%u", tag);
[... 308 lines cut ...]
+
+	/* apply overrides */
+	if (overrides) {
+		plist_dict_merge(&request, overrides);
+	}
+
+	return 0;
+}
+
 static size_t tss_write_callback(char* data, size_t size, size_t nmemb, tss_response* response) {
 	size_t total = size * nmemb;
 	if (total != 0) {
diff --git a/src/tss.h b/src/tss.h
index 3590aed..8464f11 100644
--- a/src/tss.h
+++ b/src/tss.h
@@ -48,6 +48,7 @@ int tss_request_add_vinyl_tags(plist_t request, plist_t parameters, plist_t over
 int tss_request_add_rose_tags(plist_t request, plist_t parameters, plist_t overrides);
 int tss_request_add_veridian_tags(plist_t request, plist_t parameters, plist_t overrides);
 int tss_request_add_tcon_tags(plist_t request, plist_t parameters, plist_t overrides);
+int tss_request_add_timer_tags(plist_t request, plist_t parameters, plist_t overrides);
 
 int tss_request_add_ap_img4_tags(plist_t request, plist_t parameters);
 int tss_request_add_ap_img3_tags(plist_t request, plist_t parameters);
```

pekn, 2022-04-06:
Now it works, output: [https://www.dropbox.com/s/uouznum277zmipl/output2.txt?dl=0](url)

nikias, 2022-04-06:
Great! Thanks for confirming. Will push the changes right away!

nikias, 2022-04-06:
Note: I deleted the old patches to not have people play with incomplete patches.

nikias, 2022-04-06:
See 7eaa1fa.

skortela, 2022-04-07:
Thanks @nikias and @pekn.
However this is still failing on my tests.

```
[10:04:40.378][19188] : ERROR: Unable to get (null)-FirmwarePreflightInfo from lockdownd
[10:04:40.378][19188] : ERROR: Unable to get FirmwarePreflightInfo

[10:04:40.379][19188] : Sending TSS request attempt 1... [10:04:41.053][19188] : TSS server returned: STATUS=8&MESSAGE=An internal error occurred.
[10:04:41.053][19188] : ERROR: TSS request failed (status=8, message=An internal error occurred.)
[10:04:41.053][19188] : ERROR: Unable to send TSS request
[10:04:41.054][19188] : ERROR: Unable to get SHSH blobs for this device
```

[idevicerestore_2.log](https://github.com/libimobiledevice/idevicerestore/files/8440936/idevicerestore_2.log)

nikias, 2022-04-07:
Yes that's on me, I broke the TSS in my "cleanup commit". Should work again after my fix https://github.com/libimobiledevice/idevicerestore/commit/ca44e4dab1ff5ebffa54859d7a6142f0ecbecc8e

nikias, 2022-04-07:
However I would love to find out how to fix the FirmwarePreflightInfo thing, let me check the code...

nikias, 2022-04-07:
@skortela can you hop on Telegram so we can look into it? Join our group https://t.me/libimobiledevice and then you can find me. Seems like
```
ideviceinfo -k FirmwarePreflightInfo
```
Doesn't return any data.

skortela, 2022-04-08:
Works now. Thanks.

closed 2022-04-08 by skortela