libimobiledevice/idevicerestore #477: Unable to compile

rayzorben, 2022-04-08:
Is there a distro that includes the packages as listed:
```
sudo apt-get install \
	build-essential \
	pkg-config \
	checkinstall \
	git \
	autoconf \
	automake \
	libtool-bin \
	libreadline-dev \
	libusb-1.0-0-dev \
	libplist-dev \
	libimobiledevice-dev \
	libimobiledevice-glue-dev \
	libcurl4-openssl-dev \
	libssl-dev \
	libzip-dev \
	zlib1g-dev
```

This is missing `libimobiledevice-glue-dev`. I have tried on debian latest, ubuntu latest, ubuntu 21, ubuntu 20 and ubuntu 19.

I tried going to each of the individual projects (`libimobiledevice-glue-dev`) and make those manually, but they also seem to have missing packages.  I built `libplist`, `libimobiledevice-glue`, then `libirecovery` and finally `idevicerestore`. When building `idevicerestore` I got the following error

```
configure: error: Package requirements (libimobiledevice-1.0 >= 1.3.0) were not met:

Requested 'libimobiledevice-1.0 >= 1.3.0' but version of libimobiledevice is 1.2.1
```

Which is odd because I built libimobiledevice from source.

nikias, 2022-04-08:
The distros don't have that package yet because it was not officially released yet. If you build from source you need to make sure to have a different prefix since it will check the default prefix first, or try to remove libimobiledevice-dev.
I suggest to try my [linux build script](https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh?dl=0) (or manually perform the steps in it). It will build and install everything in /usr/local prefix, and then you also need to tell pkg-config to look there:
```
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
```
etc.

rayzorben, 2022-04-08:
> The distros don't have that package yet because it was not officially released yet. If you build from source you need to make sure to have a different prefix since it will check the default prefix [...]

Thanks, I just tried running your script, and it doesn't build fully. I see this in the logs

```
checking for libirecovery... no
configure: error: Package requirements (libirecovery-1.0 >= 1.0.1) were not met:

No package 'libirecovery-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libirecovery_CFLAGS
```

rayzorben, 2022-04-08:
Nevermind, I got it. But now that I finally have it, doesn't look like it is working for me

```
▒▓ ~/Dow/wd  idevicerestore --latest       ✔ │ 1m 27s │ ubuntu-idevice ⬢ ▓▒░
ERROR: Unable to discover device mode. Please make sure a device is attached.
```

The iPhone is attached, I see it show up in `journalctl`. It is in restore mode.

nikias, 2022-04-08:
Is `libirecovery` properly installed? It has a udev rule to set permissions for the usb device files. Check with `irecovery -q` if it shows device info. If not, try to run it as root with `sudo irecovery -q`. If it only works with sudo, then check if you are in the group that is specified in the udev rules file `/lib/udev/rules.d/39-libirecovery.rules` (usually `plugdev`). You can run the `groups` command to see what groups your current user is in.

Richiie89, 2022-12-24:
/opt/futurerestore
mint@htpc: pts/8: 23 files totalt 820Kb -> sudo ./configure               checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking whether we need platform-specific build settings... checking for
a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for library containing pthread_create... none required
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libplist >= 1.12... yes
checking for libzip >= 0.10... yes
checking for libimobiledevice-1.0 >= 1.2.1... yes
checking for libirecovery >= 0.2.0... no
configure: error: Package requirements (libirecovery >= 0.2.0) were not met:

No package 'libirecovery' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libirecovery_CFLAGS
and libirecovery_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

Richiie89, 2022-12-24:
https://github.com/libimobiledevice/idevicerestore/issues/477#issuecomment-1093150420

(i tried running above script)
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh 

libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:31: installing './compile'
configure.ac:15: installing './missing'
futurerestore/Makefile.am: installing './depcomp'
configure.ac:4: warning: The macro `AC_CANONICAL_SYSTEM' is obsolete.
configure.ac:4: You should run autoupdate.
./lib/autoconf/general.m4:2081: AC_CANONICAL_SYSTEM is expanded from...
configure.ac:4: the top level
configure.ac:79: warning: AC_OUTPUT should be used without arguments.
configure.ac:79: You should run autoupdate.
/opt/futurerestore/external/idevicerestore /opt/futurerestore
./autogen.sh: rad 18: ./autogen.sh: Filen eller katalogen finns inte
/opt/futurerestore
/opt/futurerestore/external/img4tool /opt/futurerestore
./autogen.sh: rad 18: ./autogen.sh: Filen eller katalogen finns inte
/opt/futurerestore
/opt/futurerestore/external/tsschecker /opt/futurerestore
./autogen.sh: rad 18: ./autogen.sh: Filen eller katalogen finns inte
/opt/futurerestore
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking whether we need platform-specific build settings... checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for library containing pthread_create... none required
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libplist >= 1.12... yes
checking for libzip >= 0.10... yes
checking for libimobiledevice-1.0 >= 1.2.1... yes
checking for libirecovery >= 0.2.0... no
configure: error: Package requirements (libirecovery >= 0.2.0) were not met:
No package 'libirecovery' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libirecovery_CFLAGS
and libirecovery_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

Sat, 24 Dec 2022 18:46:47
/opt/futurerestore