libimobiledevice/idevicerestore #478: iPhone 6s cannot restore due to strange errors - possible device fault?

gabriele2000, 2022-04-12:
I'll just send the log, it's very strange.
It disconnects itself while verifying the firmware, sometimes it finished then everything crashes
`gabriele@msi-gp72m:~$ idevicerestore -e -d 6s.ipsw
ERROR: Unable to discover device mode. Please make sure a device is attached.
gabriele@msi-gp72m:~$ idevicerestore -e -d 6s.ipsw
irecv_event_cb: device 001104d420f22c26 (udid: N/A) connected in Recovery mode
Found device in Recovery mode
opening device 05ac:1281...
found device with ECID 001104d420f22c26
Setting to configuration 1
Setting to interface 0:0
Identified device as n71map, iPhone8,1
Extracting BuildManifest from IPSW
Product Version: 15.4.1
Product Build: 19E258 Major: 19
opening device 05ac:1281...
found device with ECID 001104d420f22c26
Setting to configuration 1
Setting to interface 0:0
INFO: device serial number is FVMS90AWGRY8
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
################################ [ WARNING ] #################################
# You are about to perform an *ERASE* restore. ALL DATA on the target device #
# will be IRREVERSIBLY DESTROYED. If you want to update your device without  #
# erasing the user data, hit CTRL+C now and restart without -e or --erase    #
# command line switch.                                                       #
# If you want to continue with the ERASE, please type YES and press ENTER.   #
##############################################################################
> YES
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from '6s/078-29203-013.dmg'
Getting device's ECID for TSS request
Found ECID 4790383736400934
Getting ApNonce in recovery mode... ae 80 b6 52 9d 73 f1 e3 56 21 32 65 f7 45 fa 52 91 bb 52 e8 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... bd 4e f1 31 42 78 fc 62 13 bd e6 97 f5 3c 0f 21 d1 7e 90 02 
NOTE: Unable to find BbCalibrationManifestKeyHash node
NOTE: Unable to find BbFactoryActivationManifestKeyHash node
NOTE: Unable to find BbSkeyId node
DEBUG: Applying restore request rules for entry AOP
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry Ap,SystemVolumeCanonicalMetadata
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry AppleLogo
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryCharging1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryFull
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow0
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryLow1
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry BatteryPlugin
DEBUG: Adding EPRO=true to TSS entry
DEBUG: Adding ESEC=true to TSS entry
DEBUG: Applying restore request rules for entry DeviceTree
[... 1357 lines cut ...]
FDR 0x55a604d6fbb0 waiting for message...
[===========                                       ]  20.0%FDR 0x55a604d6fbb0 timeout waiting for command
FDR 0x55a604d6fbb0 waiting for message...
[==============                                    ]  27.0%FDR 0x55a604d6fbb0 timeout waiting for command
FDR 0x55a604d6fbb0 waiting for message...
[=====================                             ]  41.0%FDR 0x55a604d6fbb0 timeout waiting for command
FDR 0x55a604d6fbb0 waiting for message...
[=============================                     ]  56.0%FDR 0x55a604d6fbb0 timeout waiting for command
FDR 0x55a604d6fbb0 waiting for message...
[====================================              ]  70.0%FDR 0x55a604d6fbb0 timeout waiting for command
FDR 0x55a604d6fbb0 waiting for message...
[===========================================       ]  85.0%FDR 0x55a604d6fbb0 timeout waiting for command
FDR 0x55a604d6fbb0 waiting for message...
[==================================================] 100.0%
Done sending filesystem
Verifying restore (14)
[=                                                 ]   2.0%FDR 0x55a604d6fbb0 timeout waiting for command
FDR 0x55a604d6fbb0 waiting for message...
[==================================================] 100.0%
ERROR: Could not read data (-256). Aborting.
FDR 0x55a604d6fbb0 terminating...
ERROR: Unable to restore device
idevice_event_cb: device 001104d420f22c26 (udid: ffffffffffffffffffffffffffffffff00000001) disconnected
gabriele@msi-gp72m:~$ 
`

demhademha, 2022-04-14:
It seems the restore completes, but it fails with an inability to read data, producing the error

gabriele2000, 2022-04-26:
[6s log with original cable.txt](https://github.com/libimobiledevice/idevicerestore/files/8561331/6s.log.with.original.cable.txt)
what do you think? @demhademha