libimobiledevice/idevicerestore #479: Doesn't detect device's recovery mode after trying to restore

DevUt, 2022-04-13:
I'm trying to restore my iphone 7
This is the log when running 
`idevicerestore --latest`

```
Checksum matches.
Extracting BuildManifest from IPSW
Product Version: 15.4.1
Product Build: 19E258 Major: 19
INFO: device serial number is -------------
Device supports Image4: true
Variant: Customer Upgrade Install (IPSW)
This restore will update your device without erasing user data.
Checking IPSW for required components...
All required components found in IPSW
Extracting filesystem from IPSW: 078-28933-013.dmg
[==================================================] 100.0%
Found ECID 6304446010106022
Getting ApNonce in recovery mode... d2 b6 dc d4 85 42 29 b3 50 3e e0 4f 88 95 8b f2 04 f0 10 c4 fc 7d e5 7e 11 35 fd e8 b1 55 23 d2 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 84 df 88 f3 62 20 39 ef c4 4b ca c5 5f 74 57 8a 02 04 82 77 
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received SHSH blobs
Extracting iBEC.d10.RELEASE.im4p (Firmware/dfu/iBEC.d10.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Sending iBEC (555160 bytes)...
Recovery Mode Environment:
iBoot build-version=iBoot-7459.102.5
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.im4p (Firmware/all_flash/applelogo@2x~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (13547 bytes)...
Extracting AppleAVE2FW_H9.im4p (Firmware/ave/AppleAVE2FW_H9.im4p)...
Personalizing IMG4 component AVE...
Sending AVE (1064499 bytes)...
Extracting 078-28761-016.dmg.trustcache (Firmware/078-28761-016.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Sending RestoreTrustCache (12300 bytes)...
ramdisk-size=0x20000000
Extracting 078-28761-016.dmg (078-28761-016.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (115984774 bytes)...
Extracting DeviceTree.d101ap.im4p (Firmware/all_flash/DeviceTree.d101ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (36740 bytes)...
Extracting sep-firmware.d101.RELEASE.im4p (Firmware/all_flash/sep-firmware.d101.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Sending RestoreSEP (1476216 bytes)...
Extracting kernelcache.release.iphone9 (kernelcache.release.iphone9)...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (16107740 bytes)...
Waiting for device to enter restore mode...
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running.
```
But the device does boot in recovery mode, 
I tried running it as root and starting 
but saw this

```
sudo usbmuxd -f -U root
[23:12:27.342][3] usbmuxd v1.1.2 starting up
[23:12:27.342][3] Using libusb 1.0.23
[23:12:27.345][3] Initialization complete
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: error [udev_hotplug_event] ignoring udev action bind
```

demhademha, 2022-04-13:
Can you try to build libusb, libplist, libimobiledevice-glue, libusbmuxd, usbmuxd, libimobiledebice, libirecovery and idevicerestore from source

DevUt, 2022-04-13:
It is ! Everything from source.

demhademha, 2022-04-14:
You are Using libusb 1.0.23
however your issue is fixed by mor recent version

DevUt, 2022-04-14:
Oops didn't notice that
I'm now using libusb 1.0.26
Still the same now the command was
`sudo idevicerestore --latest --erase -d`

The log
```
Received SHSH blobs
DEBUG: tss_response_get_path_by_entry: No entry 'iBEC' in TSS response
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.d10.RELEASE.im4p (Firmware/dfu/iBEC.d10.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (555160 bytes)...
Waiting for device to disconnect...
irecv_event_cb: device 001665dc38f020a6 (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 001665dc38f020a6 (udid: N/A) connected in Recovery mode
opening device 05ac:1281...
found device with ECID ---------------
Setting to configuration 1
Setting to interface 0:0
Recovery Mode Environment:
iBoot build-version=iBoot-7459.102.5
iBoot build-style=RELEASE
Sending RestoreLogo...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreLogo' in TSS response
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2x~iphone.im4p (Firmware/all_flash/applelogo@2x~iphone.im4p)...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (13547 bytes)...
DEBUG: AVE is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'AVE' in TSS response
NOTE: No path for component AVE in TSS, will fetch from build_identity
Extracting AppleAVE2FW_H9.im4p (Firmware/ave/AppleAVE2FW_H9.im4p)...
Personalizing IMG4 component AVE...
Tag found
Sending AVE (1064499 bytes)...
DEBUG: RestoreTrustCache is loaded by iBoot.
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreTrustCache' in TSS response
NOTE: No path for component RestoreTrustCache in TSS, will fetch from build_identity
Extracting 078-28753-016.dmg.trustcache (Firmware/078-28753-016.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Tag found
Sending RestoreTrustCache (12190 bytes)...
ramdisk-size=0x20000000
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreRamDisk' in TSS response
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 078-28753-016.dmg (078-28753-016.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (113832326 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreDeviceTree' in TSS response
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.d101ap.im4p (Firmware/all_flash/DeviceTree.d101ap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (36740 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreSEP' in TSS response
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.d101.RELEASE.im4p (Firmware/all_flash/sep-firmware.d101.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (1476216 bytes)...
DEBUG: tss_response_get_path_by_entry: No entry 'RestoreKernelCache' in TSS response
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.iphone9 (kernelcache.release.iphone9)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (16107740 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 001665dc38f020a6 (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 001665dc38f020a6 (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Please make sure that usbmuxd is running
```
Started usbmuxd as
```
sudo usbmuxd -f -U root
[12:34:40.240][3] usbmuxd v1.1.2 starting up
[12:34:40.242][3] Using libusb 1.0.26
[12:34:40.246][3] Initialization complete
```

nikias, 2022-04-14:
Now that you ran with -d we can see that the restore mode problem actually seems to be something else: the device doesn't even reach restore mode but goes back to recovery mode. Does the device show the Apple logo with an empty progress bar? How long is the delay between

```
Waiting for device to enter restore mode...
```
and
```
irecv_event_cb: device 001665dc38f020a6 (udid: N/A) connected in Recovery mode
```
?

DevUt, 2022-04-14:
Its almost instant, I see an apple logo, a very fast loading spinner and bam Back in recovery mode, and the error

nikias, 2022-04-14:
Ok now we know it's not a problem with usbmuxd at least. It also doesn't show up in usbmuxd log output.
However since it has a spinner showing it is definitely past recovery mode, so the component signing/stitching seems ok. Unfortunately this is really hard to figure out, since we don't know what it is the device isn't happy about...

demhademha, 2022-04-14:
Same issue as https://github.com/libimobiledevice/idevicerestore/issues/415 there was a patch which increased the time idevicerestore waits for the device to enter restore mode, but I can't find it

nikias, 2022-04-14:
Oh right, I remember. I don't recall a patch though. idevicerestore waits 120 seconds for the device to enter restore mode, but the device re-enters recovery mode so this won't have any effect changing that. I updated the title of this ticket reflecting the actual problem.
Maybe we can find out what's wrong with a serial cable to see some log output from the device, but I assume you don't have something like that at hand...

nikias, 2022-04-14:
Also can you try a restore on macOS or Windows via iTunes? If that also fails the device is to blame for some weird reason...

DevUt, 2022-04-14:
Yeah it fails on iTunes with 4013, guess its on the device then :(

nikias, 2022-04-14:
Uhhh that's one of those "unknown errors" which no obvious solution, but it might be either the device or the usb cable. Could also be the lightning port.
Seems like there's nothing else we can do from the software side unfortunately 😕

demhademha, 2022-04-14:
Can idevicerestore produce more debug data (does it have a --enable-debug) option with configur

nikias, 2022-04-14:
No it's already printing all it can with -d. Only difference is if you build libimobiledevice with --enable-debug it will print libimobiledevice communication debugging too, but at the point of failure this is all in recovery mode so that won't help.

demhademha, 2022-04-14:
Can you try from DFU

cyberitech, 2022-04-18:
Same issue here.
compiled all the peripheral libraries from `libimobiledevice` from source, compiled `idevicerestore` from source, and the agent acts as if it cannot communicate with `usbmuxd`

The macmini enters a recovery screen (I see apple logo and an empty status bar) until it eventually times out and reboots.
on the `usbmuxd` side of the house i see:

```
[15:25:46.973][3] usbmuxd v1.1.1 starting up
[15:25:46.973][4] Creating socket
[15:25:46.973][5] initialized config_dir to /var/lib/lockdown
[15:25:46.973][5] client_init
[15:25:46.973][5] device_init
[15:25:46.973][4] Initializing USB
[15:25:46.973][5] usb_init for linux / libusb 1.0
[15:25:46.976][4] Registering for libusb hotplug events
[15:25:46.976][4] 0 devices detected
[15:25:46.976][3] Initialization complete
[15:25:46.976][4] New client on fd 11 (pid 1012)
[15:25:46.976][5] Client command in fd 11 len 435 ver 1 msg 8 tag 9
[15:25:46.976][5] send_pkt fd 11 tag 9 msg 8 payload_length 278
[15:25:46.976][5] Client 11 now LISTENING
[15:26:35.991][4] New client on fd 12 (pid 65219)
[15:26:35.991][5] Client command in fd 12 len 438 ver 1 msg 8 tag 1
[15:26:35.991][5] send_pkt fd 12 tag 1 msg 8 payload_length 278
[15:26:35.991][5] Client 12 now LISTENING
[15:29:54.641][4] Client 12 connection closed
[15:29:54.641][4] Disconnecting client fd 12
[15:30:59.765][4] New client on fd 12 (pid 65402)
[15:30:59.765][5] Client command in fd 12 len 438 ver 1 msg 8 tag 1
[15:30:59.765][5] send_pkt fd 12 tag 1 msg 8 payload_length 278
[15:30:59.765][5] Client 12 now LISTENING
[15:34:17.729][4] Client 12 connection closed
[15:34:17.729][4] Disconnecting client fd 12
```

cyberitech, 2022-04-18:
encountered this same exact issue.
While compiling from source and installing the dependencies, i installed `usbmuxd 1.1.1` which was the latest from my package manager.
even though I did compile `usbmuxd`, which is currently at version `1.1.2` from source, it was being shadowed by the older version `1.1.1`
so the fix for me was ensuring `usbmuxd` was verion `1.1.2`

demhademha, 2022-04-19:
Glad you're issue is fixed
@nikias  as a package maintainer, can every libimobiledevice tool be built statically 
Would you mind a pr which made the version flag show what commit each tool was linked against

floating1duck, 2023-08-22:
> encountered this same exact issue. While compiling from source and installing the dependencies, i installed `usbmuxd 1.1.1` which was the latest from my package manager. even though I did compile [...]

Where can I get these version 1.1.2 ? The libmobiledevice version is at 1.1.1 only.

kuon, 2023-11-30:
If you are on archlinux, you need to install all tools from AUR using the `-git` versions, and ensure you are installing the libraries first, then `usbmuxd-git` and finally `idevicerestore-git`, the order is important because otherwise it will use older version to build.

StariaRose, 2023-12-09:
I have the same error, and I installed it from the AUR with the git packages, installed in that order, with clean builds.``sudo usbmuxd -v -f -U root`` does not detect the device. 

I tried ``sudo idevicerestore --erase --latest --debug``. It is worth noting I also downloaded the latest firmware and tried that to no avail.

I am using an Apple TV 2, version A1378. On macOS Catalina and Windows 11, I get the error code 1603 consistently. To my knowledge, this cable supports data transfer, so I don't quite know what that error is on about...

themanyone, 2024-10-27:
In this particular case, `usbmuxd` was not running, due to not having a `usbmux` user to run under.
That was causing the "cdc_ncm 4-1.3:1.1: bind() failure" in `dmesg`. And ERROR: Device failed to enter restore mode.

Now with the user, usbmux on Linux, the `usbmuxd` service starts properly, and the process continues...

```shell
su
useradd -M usbmux
systemctl start usbmuxd

idevicerestore -eR iPho...
```

Archetrix, 2025-02-06:
I have this case as well. I'm on macOS (15.3) and usbmuxd is already running from Apple's MobileDevice.framework.

My device (iPad 6) always comes back in recovery mode when it's sent to restore mode.

Is there another way of resetting&repairing an iPad by going through DFU-mode instead of recovery/restore mode? The content on device is irrelevant by now. Getting the iPad back to work is the primary goal.

Here's my full debug output of my attempt (S/N xxx'ed):
`amen@MBPro Downloads % /usr/local/bin/idevicerestore -e -R -y -d iPad_64bit_TouchID_ASTC_17.7.4_21H414_Restore.ipsw
idevicerestore 1.0.0-226-gbb5591d (libirecovery 1.2.1, libtatsu 1.0.4)
irecv_event_cb: device 000c61023a890026 (udid: N/A) connected in Recovery mode
Found device in Recovery mode
ECID: 3484361920413734
Identified device as j71bap, iPad7,5
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 17.7.4
IPSW Product Build: 21H414 Major: 21
INFO: device serial number is xxxxxxxxxxxx
Device supports Image4: true
Variant: Customer Erase Install (IPSW)
This restore will erase all device data.
Checking IPSW for required components...
All required components found in IPSW
Getting ApNonce in recovery mode... b7 bb e4 06 1d 7c xx 09 06 b8 7b a2 f2 29 16 xx 16 d7 7f ad cc e0 e3 3c dd fd 11 4f 40 52 d9 9e 
Trying to fetch new SHSH blob
Getting SepNonce in recovery mode... 9c 56 86 84 0f xx 98 30 3e a3 2e 77 73 79 xx 39 e5 9b 19 5f 
Received SHSH blobs
NOTE: No path for component iBEC in TSS, will fetch from build_identity
Extracting iBEC.ipad7b.RELEASE.im4p (Firmware/dfu/iBEC.ipad7b.RELEASE.im4p)...
Personalizing IMG4 component iBEC...
Tag found
Sending iBEC (570708 bytes)...
Waiting for device to disconnect...
irecv_event_cb: device 000c61023a890026 (udid: N/A) disconnected
Waiting for device to reconnect in recovery mode...
irecv_event_cb: device 000c61023a890026 (udid: N/A) connected in Recovery mode
Retrying connection...
Recovery Mode Environment:
iBoot build-version=iBoot-10151.140.19
iBoot build-style=RELEASE
Sending RestoreLogo...
NOTE: No path for component RestoreLogo in TSS, will fetch from build_identity
Extracting applelogo@2x~ipad.im4p (Firmware/all_flash/applelogo@2x~ipad.im4p)...
Personalizing IMG4 component RestoreLogo...
Tag found
Sending RestoreLogo (28458 bytes)...
DEBUG: AVE is loaded by iBoot.
NOTE: No path for component AVE in TSS, will fetch from build_identity
Extracting AppleAVE2FW_H9.im4p (Firmware/ave/AppleAVE2FW_H9.im4p)...
Personalizing IMG4 component AVE...
Tag found
Sending AVE (1101093 bytes)...
DEBUG: RestoreTrustCache is loaded by iBoot.
NOTE: No path for component RestoreTrustCache in TSS, will fetch from build_identity
Extracting 044-75545-014.dmg.trustcache (Firmware/044-75545-014.dmg.trustcache)...
Personalizing IMG4 component RestoreTrustCache...
Tag found
Sending RestoreTrustCache (12788 bytes)...
ramdisk-size=0x20000000
NOTE: No path for component RestoreRamDisk in TSS, will fetch from build_identity
Extracting 044-75545-014.dmg (044-75545-014.dmg)...
Personalizing IMG4 component RestoreRamDisk...
Tag found
Sending RestoreRamDisk (184556272 bytes)...
NOTE: No path for component RestoreDeviceTree in TSS, will fetch from build_identity
Extracting DeviceTree.j71bap.im4p (Firmware/all_flash/DeviceTree.j71bap.im4p)...
Personalizing IMG4 component RestoreDeviceTree...
Tag found
Sending RestoreDeviceTree (39907 bytes)...
NOTE: No path for component RestoreSEP in TSS, will fetch from build_identity
Extracting sep-firmware.j71b.RELEASE.im4p (Firmware/all_flash/sep-firmware.j71b.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Tag found
Sending RestoreSEP (1367266 bytes)...
NOTE: No path for component RestoreKernelCache in TSS, will fetch from build_identity
Extracting kernelcache.release.ipad7b (kernelcache.release.ipad7b)...
Personalizing IMG4 component RestoreKernelCache...
Tag found
Sending RestoreKernelCache (15909322 bytes)...
DEBUG: Waiting for device to disconnect...
irecv_event_cb: device 000c61023a890026 (udid: N/A) disconnected
Waiting for device to enter restore mode...
irecv_event_cb: device 000c61023a890026 (udid: N/A) connected in Recovery mode
ERROR: Device failed to enter restore mode.
Device reconnected in recovery mode, most likely image personalization failed.`

Kolgann, 2026-03-24:
> For me it worked after i started usbmuxd with this device mode `sudo USBMUXD_DEFAULT_DEVICE_MODE=3 usbmuxd -f`

I was struggling with the same issue, getting stuck at the "Device reconnected in recovery mode, most likely image personalization failed." I tried this and it worked like a charm. Absolute legend, thanks a lot. Maybe something to add to the README if multiple people are encountering it?

provokateurin, 2026-09-14:
Enabling `services.usbmuxd` solved the issue for me on NixOS 26.05.