libimobiledevice/idevicerestore #480: idevicerestore can't run: symbol lookup error undefined symbol: reverse_proxy_client_set_status_callback

NIXFOREVERNIX, 2022-04-19:
```console
$ idevicerestore
idevicerestore: symbol lookup error: idevicerestore: undefined symbol: reverse_proxy_client_set_status_callback
```
deps version:

- idevicerestore: commit 0b269e6ee66d3aafba14b3e48f27446086f09b64
- libimobiledevice: commit 7a0aedc97c66025d678afee66e735b6e3dcf0b9c
- libimobiledevice-glue: commit 00131b56d15e62ec83ab3dca9cf5372140f9b255
- libirecovery: commit 82d235703044c5af9da8ad8f77351fd2046dac47
- libplist: commit db93bae96d64140230ad050061632531644c46ad

nikias, 2022-04-19:
Well obviously it tries to use a different, older version of libimobiledevice, otherwise it would find the symbol. Check with
```
ldd `which idevicerestore`
```
What libraries it tries to use and match them with the installation location of your self-built libraries, then you will see what it tries to use.

NIXFOREVERNIX, 2022-04-19:
@nikias Thank you, Now works!

closed 2022-04-19 by NIXFOREVERNIX