libimobiledevice/idevicerestore #492: Support iOS 16.0

Lethaltail, 2022-06-07:
Trying to restore iOS 16.0 beta1 results in the following:
```[07:12:47.0288-GMT]{3&gt;6} CHECKPOINT BEGIN: RESTORED:[0x0655] is_host_compatible
restore-step-ids = {0x11030655:15}
restore-step-names = {0x11030655:is_host_compatible}
restore-step-uptime = 10
restore-step-user-progress = 0
host/device compatibility check found that a message is missing or is not supported (optional): FirmwareUpdaterPreflight

host/device compatibility check found that a message is missing or is not supported (optional): ReceiptManifest

host/device compatibility check found that a message is missing or is not supported (required): FirmwareUpdaterDataV2

host/device compatibility check found that a message is missing or is not supported (optional): RestoreLocalPolicy

host/device compatibility check found that a message is missing or is not supported (optional): AuthInstallCACert

host/device compatibility check found that a message is missing or is not supported (optional): OverlayRootDataForKeyIndex

Detected a mismatch between the messages the device can send and what the host can support.
Checkpoint engine recorder path set to /mnt5
[07:12:47.0289-GMT]{3&gt;6} CHECKPOINT FAILURE:(FAILURE:10) RESTORED:[0x0655] is_host_compatible [0]D(This host version is unsupported. You may need to update your host tools.)
restore-step-results = {0x11070655:{0:10}}
restore-step-codes = {0x11070655:{0:10}}
restore-step-domains = {0x11070655:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x11070655:"[0]D(This host version is unsupported. You may need to update your host tools.)"}
restore-step-uptime = 10
restore-step-user-progress = 0
[07:12:47.0290-GMT]{3&gt;6} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=0 [sync=true] (first failure)

doronz88, 2022-06-07:
This would take some work. The `restore_supported_data_types` needs to also tell the restored the following message types are also supported: 
* `FirmwareUpdaterPreflight`
* `ReceiptManifest`
* `FirmwareUpdaterDataV2`
* `RestoreLocalPolicy`
* `FirmwareUpdaterPreflight`
* `AuthInstallCACert`
* `OverlayRootDataForKeyIndex`

Unfortunately this is not enough since they added a new updater as well named "Cryptex1".
```python
{'MsgType': 'DataRequestMsg',
 'DataType': 'FirmwareUpdaterData',
 'Arguments': {'MessageArgInfo': {'ApProductionMode': True,
   'ApChipId': 33040,
   'ApECID': REDACTED,
   'Nonce': b'\xd9\x7fU\xc5^\x91[\xce\xc3Q1\x15\x90n\x89\x9353f)\xdd\xd3\xbau"\xd4Q77\xd5\x8f\xc9'},
  'MessageArgUpdaterLoopCount': 0,
  'DataChunkSize': 262144,
  'MessageForceRepersonalization': False,
  'MessageArgType': 'FirmwareResponseData',
  'DeviceGeneratedTags': {'BuildIdentityTags': ['ApSecurityDomain',
    'Cryptex1,ChipID',
    'Cryptex1,Type',
    'Cryptex1,SubType',
    'Cryptex1,ProductClass',
    'Cryptex1,UseProductClass',
    'Cryptex1,NonceDomain',
    'Cryptex1,Version',
    'Cryptex1,PreauthorizationVersion',
    'Cryptex1,FakeRoot',
    'Cryptex1,SystemOS',
    'Cryptex1,SystemVolume',
    'Cryptex1,SystemTrustCache',
    'Cryptex1,AppOS',
    'Cryptex1,AppVolume',
    'Cryptex1,AppTrustCache',
    'Cryptex1,MobileAssetBrainOS',
    'Cryptex1,MobileAssetBrainVolume',
    'Cryptex1,MobileAssetBrainTrustCache'],
   'ResponseTags': ['Cryptex1,Ticket']},
  'DeviceGeneratedRequest': {'Cryptex1,ProductClass': '0xF2',
   'Cryptex1,UDID': b'\x00\x00\x00\x00\x00\x00\x81\x10\x00\x12yZ&\x82\x80\x1e',
   'Cryptex1,PreauthorizationVersion': '20.1.283.5.16,0',
   'Cryptex1,ChipID': '0xFF10',
   'Cryptex1,SystemOS': {'Digest': b'\xca%]=\\\xb9Vn5\xec\xa3\xc7\xb38\xd7LMUE\xb7N\x9a.K\xa0\xce\x8e\xe2@\x8ed\x84\xa3U!o\xd0~J\r\xb6\x15}\rT\x05\xca~'},
   'Cryptex1,AppOS': {'Digest': b'\x148\x06\xcc\x14d\xa7oD\x17\xa8\xbb\xdd=jxw[\xfd\xc0\x12\xe6\xc4\xa6"\xc2\xf7\\\xf2\x86H_\x18\xcc\x80\x0f\xc9\x12\xf2\xd7\xffs\xda\xcb/\x80x\xd6'},
   'Cryptex1,Version': '20.1.283.5.16,0',
   'Cryptex1,Type': 1,
   'Cryptex1,SystemTrustCache': {'Digest': b'\xe2\xc3/F\x1cE\xaa\xc6X\xc58\xa9w\xe9\xf5\x03\xfe2Z<\x8f\xf7\xcc\xcd\x9dh\xbe\xac\xc4\xca\xc0\x81\x81]\x935\xbb\x04\x1c\xf2\xbfYc\xd5\x98\xbf\x08\xfa'},
   'Cryptex1,AppVolume': {'Digest': b'g\x1aBG\x91\xde\xc9\xbf\x1e\x03\xef\xbe\xfe6\xef\x1d>6J\xdf\x01\xd4b\xd7\xa5\x96\xeb\x91\xc9=K\x8a\x16K\xe2xR\xc4\x98D\xb6\xeb$ m_\xaf\xe1'},
   'Cryptex1,ProductionMode': True,
   'ApSecurityDomain': '0x01',
   'Cryptex1,AppTrustCache': {'Digest': b'nw!\xa0q\x1bM"\x95\x9f\xb0{\xf8\x06\x1e,\xf0\x92f\x1c\xc4\xd8\xb7V\x04\tZ\x9a\x9a0\xca\xa2\x0f\x17r\x82`l\x0f\xb1(\x18\xba\x99\xf6R\xc3\x10'},
   'Cryptex1,SystemVolume': {'Digest': b'c\xde\xcb\x93\xc4&\x1a\xb5a\xc4xQ\x1f}\xbd\x0b5\x9d#Wb\x13\xa9\xa4]\x02\x1a\x9e\x86<K]`E\x85\xba\xe6\xc1y\xd2<\x1c\xc8\xde\xa6\x9a)\x13'},
   'Cryptex1,NonceDomain': 1,
   'Cryptex1,Nonce': b'\xd9\x7fU\xc5^\x91[\xce\xc3Q1\x15\x90n\x89\x9353f)\xdd\xd3\xbau"\xd4Q77\xd5\x8f\xc9',
   '@Cryptex1,Ticket': True,
   'Cryptex1,SubType': 1},
  'MessageArgUpdaterName': 'Cryptex1'}}
```
I will also try and figure how to handle this new updater request.

doronz88, 2022-06-07:
**UPDATE:** Apple Configurator on latest macOS Monterey also fails, so I guess we'll have to use Ventura to test it. From my understanding, the update is currently available only for Apple Silicon (https://developer.apple.com/download/) and unfortunately I only have an intel computer to test it with.

Can anyone with Apple Silicon provide a .pcap file of the USB (XHC20) interface during a successful update? That would really help.

InternalLoss, 2022-06-08:
Intel had Ventura since the first hour of release - latest Xcode beta can also install the appropriate tools on non-dev OS to upgrade it.

doronz88, 2022-06-08:
So probably was just an issue with their update server or maybe they spread it in different times for different regions. I could only get Monterey beta until yesterday mid-day.
Anyhow, I already have a pcap file of a successful update process and working on it.

InternalLoss, 2022-06-08:
> So probably was just an issue with their update server or maybe they spread it in different times for different regions. I could only get Monterey beta until yesterday mid-day.
> 
> Anyhow, I already have a pcap file of a successful update process and working on it.

Awesome to hear! :D
I know slightly OT so feel free to mark as not on topic, but how would one compile idevicerestore for Windows from source? GitHub actions doesn't compile one EXE with all libraries necessary when I tried

doronz88, 2022-06-08:
I have no experience with idevicerestore on windows

doronz88, 2022-06-09:
I achieved a successful firmware update with [pymobiledevice3](https://github.com/doronz88/pymobiledevice3).
You can look at the PR for the modifications needed. I'll also create a PR for idevicerestore with these changes.

https://github.com/doronz88/pymobiledevice3/pull/269

Lethaltail, 2022-06-09:
wasnt able to try the pymobiledevice3 commit due to construct errors. Tried the PR instead. System updater went through smoothly but Savage failed to update: "bundleDataDict missing expected tag."
```
[22:15:21.0402-GMT]{3>6} CHECKPOINT END: FIRMWARE:[0x1309] update_se
restore-step-ids = {0x1103067B:63;0x11030635:71}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030635:update_device_firmware}
restore-step-uptime = 816
restore-step-user-progress = 89
Tolerated step 0x1309 has actually succeeded. Marking it as such
[22:15:21.0405-GMT]{3>6} CHECKPOINT BEGIN: FIRMWARE:[0x130D] update_savage
restore-step-ids = {0x1103067B:63;0x11030635:71;0x1103130D:88}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030635:update_device_firmware;0x1103130D:update_savage}
restore-step-uptime = 816
restore-step-user-progress = 89
entering update_savage_with_option
opening /usr/lib/updaters/libSavageUpdater_iOS.dylib
operation 59 completed at progress 0 (bug?)
ramrod_display_set_granular_progress_forced: 89.000000
opening /usr/lib/updaters/libSavageUpdater_iOS.dylib
update_generic_firmware: Supports DeviceRestoreInfo, setting ForceRepersonalization and skipping preflightTickets query for tethered restore.
populating AP ticket for updater
ramrod_updater_copy_options: Savage options: { }
creating directory (owner=25 group=25 mode=755, class=-1) /mnt9/ECB0FDA3517822CDAF6C11423CAFFED8F719183B8B9B21AF17F176B72BE06ED9F94ED34E673E80CAF02AA39E2D8C25E8//private/var//wireless/Library/Logs/CrashReporter/updater_output//Savage
ramrod_updater_copy_options: creating SavageUpdater instance with options->Options={
    APTicket = {length = 10153, bytes = 0x308227a5 1604494d 344d0201 0031821e ... 4d420313 862369ee };
    DebugLogPath = "/mnt9/ECB0FDA3517822CDAF6C11423CAFFED8F719183B8B9B21AF17F176B72BE06ED9F94ED34E673E80CAF02AA39E2D8C25E8//private/var//wireless/Library/Logs/CrashReporter/updater_output//Savage/";
    PostFDRSealing = 0;
    PreflightContext = Limited;
    RestoreInternal = 0;
    RestoreSystemPartition = "/mnt9/ECB0FDA3517822CDAF6C11423CAFFED8F719183B8B9B21AF17F176B72BE06ED9F94ED34E673E80CAF02AA39E2D8C25E8";
}
creating SavageUpdater obj
updater_log: <libSavageUpdater 0x112c056f0 [0x102adc1e8]>{} 
updater_log: SavageUpdateController::start: isProvisioned=0xF23BB19CF28E9AAF; isNGIDProvisioned=0x1FBEDC; isSocProdfused=1; persoLoop = 1
updater_log: start [input]: Options {
    APTicket = {length = 10153, bytes = 0x308227a5 1604494d 344d0201 0031821e ... 4d420313 862369ee };
    DebugLogPath = "/mnt9/ECB0FDA3517822CDAF6C11423CAFFED8F719183B8B9B21AF17F176B72BE06ED9F94ED34E673E80CAF02AA39E2D8C25E8//private/var//wireless/Library/Logs/CrashReporter/updater_output//Savage/";
    PostFDRSealing = 0;
    PreflightContext = Limited;
    RestoreInternal = 0;
    RestoreSystemPartition = "/mnt9/ECB0FDA3517822CDAF6C11423CAFFED8F719183B8B9B21AF17F176B72BE06ED9F94ED34E673E80CAF02AA39E2D8C25E8";
} 
updater_log: YonkersUpdateController::start: isProvisioned=0; isSocProdFused=1; persoLoop = 4
updater_log: start [input]: Options {
    APTicket = {length = 10153, bytes = 0x308227a5 1604494d 344d0201 0031821e ... 4d420313 862369ee };
    DebugLogPath = "/mnt9/ECB0FDA3517822CDAF6C11423CAFFED8F719183B8B9B21AF17F176B72BE06ED9F94ED34E673E80CAF02AA39E2D8C25E8//private/var//wireless/Library/Logs/CrashReporter/updater_output//Savage/";
    PostFDRSealing = 0;
    PreflightContext = Limited;
    RestoreInternal = 0;
    RestoreSystemPartition = "/mnt9/ECB0FDA3517822CDAF6C11423CAFFED8F719183B8B9B21AF17F176B72BE06ED9F94ED34E673E80CAF02AA39E2D8C25E8";
} 
updater_log: SavageUpdateController::getInstance - 0x112c0a8f0 (0x112c092b0) - UpdaterIsSupported = 1; SavageIsPresent = 1 
updater_log: SavageUpdateController::getInstance - 0x112c0a8f0 (0x112c092b0) - UpdaterIsSupported = 1; SavageIsPresent = 1 
updater_log: SavageUpdateController::execCommand: Entering execCommand: command = queryInfo, persoLoop = 1  
updater_log: SavageUpdateController::eventCmdQueryInfo: Begin Personalization 
updater_log: SavageUpdateController::eventCmdQueryInfo: Finished pre-personalizing 
updater_log: execCommand [output]: output dictionary of execCommand {
    "Savage,AllowOfflineBoot" = 1;
    "Savage,Nonce" = {length = 16, bytes = 0xa3eb7218f9d3b58b8d51689f569b0a76};
    "Savage,ProductionMode" = 1;
    "Savage,ReadECKey" = 1;
    "Savage,ReadFWKey" = 1;
    "Savage,Revision" = {length = 2, bytes = 0xac0b};
    "Savage,UID" = {length = 16, bytes = 0xf23bb19cf28e9aaf5ff8bb0c4421ea97};
} 
updater_log: SavageUpdateController::execCommand: Exiting execCommand: command = queryInfo, result = 0x0, persoLoop = 1 
copying Savage updater response for loop 0
Attempting (previously host-side) RestoreInfo flow on-device.
_logSinkCallback: Savage: SavageUpdaterGetTags [input]: DeviceInfo {
    "Savage,AllowOfflineBoot" = 1;
    "Savage,Nonce" = {length = 16, bytes = 0xa3eb7218f9d3b58b8d51689f569b0a76};
[... 160 lines cut ...]
restore-step-uptime = 816
restore-step-user-progress = 89
[22:15:21.0839-GMT]{3>6} CHECKPOINT FAILURE:(FAILURE:-1) RESTORED:[0x067B] perform_restore_installing py_device_side_restoreinfo: Failed to fetch remote firmware and request for Savage, error 99)[4]D(_AMAuthInstallUpdaterSetInfoWithCallbacks: failed to copy firmware)[5]LD(SavageErrorDomain: CreateFileData: bundleDataDict missing expected tag)
restore-step-results = {0x1107067B:{0:-1};0x11070635:{0:-1};0x1107130D:{0:98}}
[... cut ...]
[22:15:21.0903-GMT]{3>6} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103067B:63;0x11030635:71;0x1103130D:88;0x11030648:92}
restore-step-names = {0x1103067B:perform_restore_installing;0x11030635:update_device_firmware;0x1103130D:update_savage}
restore-step-uptime = 816
restore-step-user-progress = 89
ERROR: Unable to successfully restore device
Checkpoint 1608 complete with code 0
ReverseProxy[Ctrl]: (status=2) Terminated
ERROR: Unable to restore device
segmentation fault

doronz88, 2022-06-09:
The PR is still a draft and needs more work. @Lethaltail please create an issue on pymobiledevice3 if you encountered an error. This should just install as-is (as it does in github's CI) and it's much easier for more to debug (as the original writer)

doronz88, 2022-06-10:
@Lethaltail I fixed a few more bugs in my PR and got it to work on iPhone 13 Pro Max and iPhone XR. Please tell me if you're still encountering a problem. If so, please try it out on pymobiledevice3 with the `-v` flag so it'll be easier for me to debug.

Lethaltail, 2022-06-10:
Got a successful upgrade restore on my iPhone 12 mini. (after bypassing the iPhone 12 line's null SRNM quirk - #453)

Lethaltail, 2022-06-10:
This might've been me playing a little dangerously though. The device was in Recovery Mode instead of Restore Mode. Although my iPhone suffered no consequences, ignoring the lack of S/N to get to this point probably wasn't very safe.

nikias, 2022-06-10:
Hmm wait didn't I remove the serial number check?

doronz88, 2022-06-10:
@Lethaltail this is indeed a seperate bug that I didn't stumble upon probably because I really did start the update in normal mode. I think it's best @nikias will solve it in a seperate PR/commit. Also, my own PR is unrealted to this and could be merged if my solution to the `DeviceGeneratedTags` looks good enough.

nikias, 2022-06-10:
Apparently I didn't fix all the serial number checks. Will care of it, ECID is fine to make sure the device is the expected one. The PR already looks good, will check that too.

nikias, 2022-06-17:
Fixed iOS 16 support via #493.

skortela, 2022-08-01:
I tried to restore iOS 16.0 beta 4 to iPhone 8 Plus. (https://www.theiphonewiki.com/wiki/Beta_Firmware/iPhone/16.x)

Extract of "078-44523-050.dmg" is failing. There is space on drive. Also 7-zip is failing to view content of '078-44523-050.dmg', other dmg files seems to work.
Environment is Windows 11.

```
[10:03:17.322][2004] : Received Cryptex1,Ticket
[10:03:17.322][2004] : Sending FirmwareResponse data now...
[10:03:17.325][2004] : Done sending FirmwareUpdater data
[10:03:17.348][2004] : common.c:printing 444 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>MsgType</key>
	<string>DataRequestMsg</string>
	<key>DataType</key>
	<string>SourceBootObjectV4</string>
	<key>Arguments</key>
	<dict>
		<key>Variant</key>
		<string>Erase</string>
		<key>ImageName</key>
		<string>Cryptex1,SystemOS</string>
	</dict>
</dict>
</plist>
[10:03:17.348][2004] : About to send Cryptex1,SystemOS...
[10:03:17.348][2004] : DEBUG: tss_response_get_path_by_entry: No entry 'Cryptex1,SystemOS' in TSS response
[10:03:17.348][2004] : NOTE: No path for component Cryptex1,SystemOS in TSS, will fetch from build identity
[10:03:17.348][2004] : common.c:printing 173 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
</plist>
[10:03:17.348][2004] : Extracting 078-44523-050.dmg (078-44523-050.dmg)...
[10:03:17.361][2004] : ERROR: Out of memory
[10:03:17.361][2004] : ERROR: Unable to extract 078-44523-050.dmg from iPhone_5.5_P3_16.0_20A5328h_Restore.ipsw
[10:03:17.361][2004] : ERROR: Unable to extract component Cryptex1,SystemOS
[10:03:17.361][2004] : ERROR: Unable to send SourceBootObjectV4
[10:03:17.361][2004] : ERROR: Unable to successfully restore device
```

doronz88, 2022-08-04:
Seems like your hard drive is just too full. Try to clean it up before.
Alternatively, you can use [pymobiledevice3](https://github.com/doronz88/pymobiledevice3) which extract the binaries in-memory to reduce the amount of space required for the software update.

Lethaltail, 2022-08-04:
I’ve encountered this in my own testing but I neglected to mention it as I wanted to open a separate issue about this at some point.
The machine is running out of **memory** (not storage) in this scenario. The device I reproduced this on had plenty of storage available but only 2GB of RAM in all. Monitoring memory usage on a machine with plenty of RAM and storage shows that ~6GB of system memory is used up in the “Extracting *.dmg (*.dmg)…” stage. It’s actually had me on edge on more than one occasion, hoping that the restore succeeds.

skortela, 2022-08-08:
I have plenty of RAM, but my idevicerestore process was 32 bit. So it cannot allocate more that 2GB of memory.
I compiled 64 bit version, and it is working. In extracting phase process reserved about 2.5GB of memory.

nikias, 2022-10-10:
The Cryptex1,SystemOS memory issue has been fixed in the meantime.

closed 2022-10-10 by nikias